dsmrreader/dsmr-reader #569: Could not read block error

marcfon, 2018-12-26:
Er gaat iets mis met het opstarten van m'n backend... Hieronder de foutmelding uit de logfile.

Tot nu toe heeft alles maanden lang als een zonnetje gedraaid. 

Twee dingen die er wellicht iets mee te maken hebben... 1) ik heb kort hiervoor het filesystem moeten fixen met fsck 2) ik heb een apt-get upgrade gedraaid maar weet niet zeker of daar ook python packages tussen zaten.

Alle aanwijzingen in de juiste richting om dit op te lossen zijn welkom!

```
- Creating day & hour statistics for: 2018-12-24 00:00:00+01:00.
 >>> Uncaught exception :: could not read block 18 in file "base/16385/16985": read only 0 of 8192 bytes
 >>> <bound method AppConfig._on_backend_called_signal of <AppConfig: dsmr_stats>> ::   File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/dispatch/dispatcher.py", line 210, in send_robust
    response = receiver(signal=self, sender=sender, **named)

  File "/home/dsmr/dsmr-reader/dsmr_stats/apps.py", line 20, in _on_backend_called_signal
    dsmr_stats.services.analyze()

  File "/home/dsmr/dsmr-reader/dsmr_stats/services.py", line 95, in analyze
    create_hourly_statistics(hour_start=hour_start)

  File "/home/dsmr/dsmr-reader/dsmr_stats/services.py", line 139, in create_hourly_statistics
    if HourStatistics.objects.filter(**creation_kwargs).exists():

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/db/models/query.py", line 715, in exists
    return self.query.has_results(using=self.db)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/db/models/sql/query.py", line 509, in has_results
    return compiler.has_results()

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/db/models/sql/compiler.py", line 1037, in has_results
    return bool(self.execute_sql(SINGLE))

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/db/models/sql/compiler.py", line 1068, in execute_sql
    cursor.execute(sql, params)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/db/backends/utils.py", line 68, in execute
    return self._execute_with_wrappers(sql, params, many=False, executor=self._execute)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/db/backends/utils.py", line 77, in _execute_with_wrappers
    return executor(sql, params, many, context)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/db/backends/utils.py", line 85, in _execute
    return self.cursor.execute(sql, params)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/db/utils.py", line 89, in __exit__
    raise dj_exc_value.with_traceback(traceback) from exc_value

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/db/backends/utils.py", line 85, in _execute
    return self.cursor.execute(sql, params)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/dispatch/dispatcher.py", line 210, in send_robust
    response = receiver(signal=self, sender=sender, **named)

  File "/home/dsmr/dsmr-reader/dsmr_stats/apps.py", line 20, in _on_backend_called_signal
    dsmr_stats.services.analyze()

  File "/home/dsmr/dsmr-reader/dsmr_stats/services.py", line 95, in analyze
    create_hourly_statistics(hour_start=hour_start)

  File "/home/dsmr/dsmr-reader/dsmr_stats/services.py", line 139, in create_hourly_statistics
    if HourStatistics.objects.filter(**creation_kwargs).exists():

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/db/models/query.py", line 715, in exists
    return self.query.has_results(using=self.db)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/db/models/sql/query.py", line 509, in has_results
    return compiler.has_results()

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/db/models/sql/compiler.py", line 1037, in has_results
    return bool(self.execute_sql(SINGLE))

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/db/models/sql/compiler.py", line 1068, in execute_sql
    cursor.execute(sql, params)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/db/backends/utils.py", line 68, in execute
    return self._execute_with_wrappers(sql, params, many=False, executor=self._execute)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/db/backends/utils.py", line 77, in _execute_with_wrappers
    return executor(sql, params, many, context)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/db/backends/utils.py", line 85, in _execute
    return self.cursor.execute(sql, params)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/db/utils.py", line 89, in __exit__
    raise dj_exc_value.with_traceback(traceback) from exc_value

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/db/backends/utils.py", line 85, in _execute
    return self.cursor.execute(sql, params)
Command completed. Sleeping for 1 second(s)...
```

marcfon, 2018-12-26:
> Het lijkt op filesystem corruptie. Weet je zeker dat je SD-kaart nog goed is? De levensduur varieert van enkele maanden tot jaren, heel willekeurig. Ik neem aan dat je al een reboot en alles hebt geprobeerd?

Het zou zo maar kunnen dat de SD-kaart het langzaam aan het begeven is. Ik zie meer file errors voorbij komen de laatste tijd. Ik clone iedere nacht de SD-kaart naar een ander kaartje als bootable backup dus die zou ik morgen eens kunnen testen.

> Als je nog een goede database back-up hebt, stel die in ieder geval veilig. Anders zul je denk ik de database moeten restoren en hopen dat de SD-kaart nog goed is (bewaar de back-up sowieso).

Database backup zit in dropbox. Dus in het slechtste geval verlies ik 1 dag aan data?

> Je zou eventueel eerst nog kunnen proberen om een extra back-up te maken van de huidige staat:
> 
> ```
> sudo su - postgres
> psql pg_dump dsmrreader > dsmrreader-backup.sql
> ```

Hmm er is schijnbaar nog een wachtwoord nodig...
`psql: FATAL:  Peer authentication failed for user "dsmrreader`

Wat zou de manier zijn om de database van gisteren terug te zetten?

Thanks voor de snelle reactie! :)

dennissiemensma, 2018-12-26:
Ik zie dat ik een typo heb gemaakt, `psql` hoeft niet in het command, het is gewoon:

```
pg_dump dsmrreader > dsmrreader-backup.sql
```

Om een back-up terug te zetten zul je dsmrreader moeten stoppen, de database verwijderen, een lege moeten aanmaken en de oude dan weer importeren.

```
sudo supervisorctl stop all

sudo sudo -u postgres dropdb dsmrreader
sudo sudo -u postgres createdb -O dsmrreader dsmrreader

# Restore een .sql file
sudo sudo -u postgres psql dsmrreader -f <PATH-TO-POSTGRESQL-BACKUP.sql>

# Of wanneer het een .sql.gz is:
zcat <PATH-TO-POSTGRESQL-BACKUP.sql.gz> | sudo sudo -u postgres psql dsmrreader

sudo supervisorctl start all
```

Zie ook deels https://dsmr-reader.readthedocs.io/nl/latest/installation/explained.html#database-backend-postgresql bij `Optioneel: Wil je ook nog een PostgreSQL database back-up herstellen?`

marcfon, 2018-12-27:
Kans lijkt mij groot dat het interdaad aan een corrupt filesystem ligt. Heb nu via mijn backup sd kaart geboot en de database teruggezet. Bij het terugzetten krijg ik 1 foutmelding maar het systeem draait in ieder geval weer. Deze sd kaart gaat de prullenbak in.

```psql:./dsmrreader-backup.sql:2647676: ERROR:  could not create unique index "dsmr_datalogger_dsmrreading_pkey"
DETAIL:  Key (id)=(2088713) is duplicated.```

marcfon, 2018-12-27:
Het kan perceptie zijn maar ik heb de indruk dat een UPDATE op de DB nu veel meer cpu load kost. Kan het zo zijn dat ik de database moet optimize of opschonen op de een of andere manier? Of dat er een index niet goed gezet is?

dennissiemensma, 2018-12-27:
In principe zou die evensnel moeten zijn, mits de I/O evenveel aan kan. Welke back-up heb je nu teruggezet, de handmatig gemaakte of een oudere uit Dropbox?

marcfon, 2018-12-27:
Ik heb de backup teruggezet die ik met de hand gemaakt heb. Het duurt nu +/- 20 seconden om 6 readings te verwerken. Voorheen was dat amper merkbaar volgens mij. Vreemd.

dennissiemensma, 2018-12-27:
Zolang hoort dat inderdaad niet te duren. Je kunt kijken of er indexes op staan met:
```
sudo su - postgres
psql dsmrreader
\dS dsmr_datalogger_dsmrreading;
```

Je krijgt als het goed is dan iets te zien als:
```
                                              Table "public.dsmr_datalogger_dsmrreading"
             Column              |           Type           |                                Modifiers                                 
---------------------------------+--------------------------+--------------------------------------------------------------------------
 id                              | integer                  | not null default nextval('dsmr_datalogger_dsmrreading_id_seq'::regclass)
 timestamp                       | timestamp with time zone | not null
 electricity_delivered_1         | numeric(9,3)             | not null
 electricity_returned_1          | numeric(9,3)             | not null
 electricity_delivered_2         | numeric(9,3)             | not null
 electricity_returned_2          | numeric(9,3)             | not null
 electricity_currently_delivered | numeric(9,3)             | not null
 electricity_currently_returned  | numeric(9,3)             | not null
 extra_device_timestamp          | timestamp with time zone | 
 extra_device_delivered          | numeric(9,3)             | 
 processed                       | boolean                  | not null
 phase_currently_delivered_l1    | numeric(9,3)             | 
 phase_currently_delivered_l2    | numeric(9,3)             | 
 phase_currently_delivered_l3    | numeric(9,3)             | 
 phase_currently_returned_l1     | numeric(9,3)             | 
 phase_currently_returned_l2     | numeric(9,3)             | 
 phase_currently_returned_l3     | numeric(9,3)             | 
Indexes:
    "dsmr_datalogger_dsmrreading_pkey" PRIMARY KEY, btree (id)
    "dsmr_datalogger_dsmrreading_9d090a40" btree (processed)
    "dsmr_datalogger_dsmrreading_timestamp_1f7020d1_uniq" btree ("timestamp")
```
Onderin staan dan de indexes. Je zou het daarnaast ook nog even kunnen checken bij de tabellen `dsmr_consumption_electricityconsumption` en `dsmr_consumption_gasconsumption`.

marcfon, 2018-12-27:
Hmmm, toch geen indexes op die fields zo te zien. Toch nog iets mis gegaan bij het terugzetten van de DB. 

```
                                              Table "public.dsmr_datalogger_dsmrreading"
             Column              |           Type           |                                Modifiers                                 
---------------------------------+--------------------------+--------------------------------------------------------------------------
 id                              | integer                  | not null default nextval('dsmr_datalogger_dsmrreading_id_seq'::regclass)
 timestamp                       | timestamp with time zone | not null
 electricity_delivered_1         | numeric(9,3)             | not null
 electricity_returned_1          | numeric(9,3)             | not null
 electricity_delivered_2         | numeric(9,3)             | not null
 electricity_returned_2          | numeric(9,3)             | not null
 electricity_currently_delivered | numeric(9,3)             | not null
 electricity_currently_returned  | numeric(9,3)             | not null
 extra_device_timestamp          | timestamp with time zone | 
 extra_device_delivered          | numeric(9,3)             | 
 processed                       | boolean                  | not null
 phase_currently_delivered_l1    | numeric(9,3)             | 
 phase_currently_delivered_l2    | numeric(9,3)             | 
 phase_currently_delivered_l3    | numeric(9,3)             | 
 phase_currently_returned_l1     | numeric(9,3)             | 
 phase_currently_returned_l2     | numeric(9,3)             | 
 phase_currently_returned_l3     | numeric(9,3)             | 
Indexes:
    "dsmr_datalogger_dsmrreading_processed_add8a852" btree (processed)
    "dsmr_datalogger_dsmrreading_timestamp_1f7020d1" btree ("timestamp")
```

dennissiemensma, 2018-12-27:
Je zou kunnen kijken of je de missende index met de hand kan toevoegen. Ik ben er echter niet zo bekend mee als MySQL, dus het is een gok of het in 1x goed gaat:

```
sudo su - postgres
psql dsmrreader

ALTER TABLE ONLY public.dsmr_datalogger_dsmrreading
    ADD CONSTRAINT dsmr_datalogger_dsmrreading_pkey PRIMARY KEY (id);
```

dennissiemensma, 2018-12-27:
Ja dat lijkt me het beste, ga je daar uitkomen? Wie weet kom je nog wel meerdere duplicate records tegen. Wel vreemd overigens, gezien je DSMR-reader uit had staan tijdens de restore (hoop ik).

marcfon, 2018-12-27:
Hoppa! Fixed :)

Achteraf gezien had ik toch iets minder snel over de eerste foutmelding heen moeten lezen.

Geen idee hoe die dubbele record erin is gekomen. Misschien stond de logger toch nog aan bij de restore.

```
SELECT * FROM public.dsmr_datalogger_dsmrreading WHERE id = 2088713;
``

Levert inderdaad 2 records op.

```
DELETE FROM public.dsmr_datalogger_dsmrreading WHERE id = 2088713;
```

```
ALTER TABLE ONLY public.dsmr_datalogger_dsmrreading ADD CONSTRAINT dsmr_datalogger_dsmrreading_pkey PRIMARY KEY (id);
```

En na het aanmaken van de index loopt het weer als vanouds.

Thanks Dennis!