dsmrreader/dsmr-reader #570: ImportError: cannot import name 'Signal'

rodeho, 2018-12-27:
### Jouw omgeving
*(Indien van toepassing)*
* DSMR-reader versie *(rechtsbovenin applicatie zichtbaar)*: `v1.2.7`
* Hardware *(RaspberryPi 2, 3 of iets anders)*: `RaspberryPi 3...`
* DSMR-protocol versie van slimme meter *(v2, v4 of v5)*: `v4`

###
Ha Dennis na wat apt-update updates is de omgeving ermee gestopt. Nu lijkt het een relatie met Django te hebben.

Ben jij hiervan op de hoogten en heb je eventueel een oplossing? Django bijwerken met PIP gaf vooralsnog nog geen oplossing. 

Requirement already satisfied: MarkupSafe>=0.23 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages (from jinja2->coreschema->coreapi==2.3.3->-r dsmrreader/provisioning/requirements/base.txt (line 2)) (1.0)

 --- Checking & synchronizing database changes/migrations.
Traceback (most recent call last):
  File "./manage.py", line 8, in <module>
    from django.core.management import execute_from_command_line
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/core/management/__init__.py", line 12, in <module>
    from django.core.management.base import (
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/core/management/base.py", line 14, in <module>
    from django.db import DEFAULT_DB_ALIAS, connections
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/__init__.py", line 1, in <module>
    from django.core import signals
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/core/signals.py", line 1, in <module>
    from django.dispatch import Signal
ImportError: cannot import name 'Signal'

 --- Checking & synchronizing static file changes.
Traceback (most recent call last):
  File "./manage.py", line 8, in <module>
    from django.core.management import execute_from_command_line
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/core/management/__init__.py", line 12, in <module>
    from django.core.management.base import (
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/core/management/base.py", line 14, in <module>
    from django.db import DEFAULT_DB_ALIAS, connections
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/__init__.py", line 1, in <module>
    from django.core import signals
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/core/signals.py", line 1, in <module>
    from django.dispatch import Signal
ImportError: cannot import name 'Signal'

 --- Reloading app code...

 * Reloading process: dsmr_webinterface (Gunicorn)   [FAILED] PID file does not exist! (yet?)
 * Reloading process: dsmr_backend                   [FAILED] PID file does not exist! (yet?)
 * Reloading process: dsmr_datalogger                [FAILED] PID file does not exist! (yet?)
 * Reloading process: dsmr_mqtt                      [FAILED] PID file does not exist! (yet?)

 --- Clearing cache...
Traceback (most recent call last):
  File "./manage.py", line 8, in <module>
    from django.core.management import execute_from_command_line
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/core/management/__init__.py", line 12, in <module>
    from django.core.management.base import (
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/core/management/base.py", line 14, in <module>
    from django.db import DEFAULT_DB_ALIAS, connections
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/__init__.py", line 1, in <module>
    from django.core import signals
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/core/signals.py", line 1, in <module>
    from django.dispatch import Signal
ImportError: cannot import name 'Signal'

 --- Deployed version:
1.27

dennissiemensma, 2018-12-28:
Bedankt voor je vraag, vervelend dat het niet meer werkt. Ik denk dat je het beste kunt beginnen met het herinstalleren van de virtualenv.

```
sudo supervisorctl stop all

sudo rm -rf /home/dsmr/.virtualenvs/dsmrreader

sudo sudo -u dsmr virtualenv /home/dsmr/.virtualenvs/dsmrreader --no-site-packages --python python3

sudo sudo -u dsmr /home/dsmr/.virtualenvs/dsmrreader/bin/pip3 install -r /home/dsmr/dsmr-reader/dsmrreader/provisioning/requirements/base.txt -r /home/dsmr/dsmr-reader/dsmrreader/provisioning/requirements/postgresql.txt

sudo supervisorctl start all
```

Ik hoor graag of dat het oplost. Het in in principe een herinstallatie van alle requirements.

dennissiemensma, 2018-12-28:
Fijn om te horen, dat lijkt me een goed idee. Ik heb de titel aangepast zodat ik het later kan toevoegen in de docs.

dennissiemensma, 2018-12-29:
Ik heb het toegevoegd voor de volgende versie.

closed 2018-12-29 by dennissiemensma