dsmrreader/dsmr-reader #572: Stil on the old version after updating

Blackpander, 2018-12-29:
Ik maak  gebruik van versie 1.26.1 van DSMR-Reader op een Raspberry Pi 3B.

Ik heb volgens de handleiding de volgende handelingen gedaan:

sudo su - dsmr
./deploy.sh

Dit verloopt allemaal goed alleen aan het eind staat er deployed version 1.26.1. Dus niet de laatste versie 1.27.

dennissiemensma, 2018-12-30:
Bedankt voor je melding. Ik neem aan dat je niet op een aparte branch hebt gedraaid eerder?

Je kunt dit nog proberen:
```
sudo su - dsmr
git fetch
git checkout master
git pull
./deploy.sh
```

dennissiemensma, 2019-01-02:
@Blackpander ben je hier nog uit gekomen?

Blackpander, 2019-01-02:
Hi @dennissiemensma Ja, ik kreeg de melding dat ik moest comitten, omdat ik een lokale aanpassing had gemaakt. Wat uiteraard zo was aangezien ik een aanpassing heb gemaakt om het DSMR-Reader Dashboard in een iFrame te kunnen gebruiken.

Ik heb dit opgelost met:

`git reset --hard origin/master
./deploy.sh

`

dennissiemensma, 2019-01-02:
Top. Je zou de tool overigens prima in een iFrame moeten kunnen gebruiken, zie ook dit oude ticket met daarin de oplossing. Dit raakt geen bestanden die onder versiebeheer staan:

https://github.com/dennissiemensma/dsmr-reader/issues/344#issuecomment-315523564

closed 2019-01-02 by dennissiemensma

Blackpander, 2019-01-02:
@dennissiemensma Thanks! base.py hersteld en dit toegevogd aan de settings.py :)