dsmrreader/dsmr-reader #573: Toevoegen vaste kosten levering

BlackDex, 2019-01-01:
Om de kosten per dag wat beter te maken zou het een leuke toevoeging zijn om de "Vaste kosten levering per dag" toe te kunnen voegen per onderdeel, gas en elektriciteit.

Ondanks dat het vaak een klein bedrag is wat per dag er boven op komt, zou dit het prijsoverzicht wat meer kloppend maken.

dennissiemensma, 2019-01-01:
Bedankt voor je suggestie. Ik begrijp de meerwaarde helemaal, maar ik heb in het verleden echter besloten om geen verdere prijsinformatie toe te voegen, gezien het geen basis-functionaliteit is m.b.t. het uitlezen van de slimme meter (en om het project behapbaar te houden). 
De huidige prijzen zitten er dan ook puur in voor een grove indicatie van het ruwe verbruik en dan vooral dan ook ten opzichte van elkaar. Daarnaast zorgt het toevoegen van vaste dagprijzen voor aardig wat complexiteit, die dan op meerdere plekken meegenomen moeten worden (archief, dashboard, notificaties, dagstatistieken). Dus voor nu laat ik het voor wat het is.

BlackDex, 2019-01-01:
Ah, duidelijk. Vanuit dat standpunt gezien is het kleine extra meer moeite. Thx voor de uitleg

closed 2019-01-01 by BlackDex

poudenes, 2020-04-24:
Is het standpunt nog steeds van kracht? Ik had eigenlijk exact dezelfde vraag en zag deze post. 
Een post van vorig jaar januari. Het zou zomaar kunnen dat de gedachtegang nu anders is geworden? :D

dennissiemensma, 2020-04-24:
Bedankt voor het opzoeken. Ik heb hier vrij recent naar gekeken en daaruit bleek dat het ontzettend veel werk was om het overal door te doen, dus in dat opzich laat ik het nog steeds voor wat het is.

poudenes, 2020-04-25:
Thanks voor je terug koppeling. Ik ben net begonnen wellicht na een jaar weet ik een gemiddelde kostprijs per dag aan verbruik en kan ik de dagkosten splitsen en verwerken in de kostprijs voor gas en energie. Dan wordt de uitkomst nog ietsjes beter .... creativiteit loslaten heet dat haha