dsmrreader/dsmr-reader #576: DSMR-reader roadmap tot medio april 2019

dennissiemensma, 2019-01-04:
Release [v1.28.0](https://github.com/dennissiemensma/dsmr-reader/releases/tag/v1.28.0) is komende tijd de laatste release, tot medio april 2019. Wegens andere prioriteiten ben ik minder actief met DSMR-reader tot die tijd. 

Later in het voorjaar vervolg ik weer de werkzaamheden qua ontwikkelingen en updates. Evenals het uitfaseren van ondersteuning voor ``Python 3.4`` (die dan End of Life is) en het upgraden naar ``Django 2.1``.

Webserve, 2019-01-12:
Top. In ieder geval bedankt voor al je inzet. Super app..

dennissiemensma, 2019-04-15:
Ik ben inmiddels sinds gisteravond weer thuis van drie maanden reizen in het buitenland. Daar had ik alleen de mogelijkheid om te lezen en reageren per e-mail. Dus op korte termijn wil ik wat tijd steken in het door- en teruglezen van alle issues, vragen en suggesties die jullie gedaan hebben.

Bedankt allemaal voor jullie geduld.

closed 2019-04-15 by dennissiemensma