dsmrreader/dsmr-reader #577: Wens: Vergelijk van meerdere dagen alleen op totaalniveau in een tabel met evt staafdiagrammen.

joepv, 2019-01-05:
**Omgeving:**
* DSMR-reader versie: **v1.27**
* Hardware: **Intel NUC met Debian 9**
* DSMR-protocol versie van slimme meter: **onbekend.**

**Wens:**
Vergelijk van meerdere dagen alleen op totaalniveau in een tabel met evt staafdiagrammen.

**Uitleg:**
In de huidige versie kun je bij archief maar 1 dag in het verleden kijken naar het verbruik. Bij vergelijk kun je maar 2 dagen selecteren. Ik zou graag een overzicht van bijvoorbeeld een week, maand of manuele selectie willen zien van het verbruik/teruglevering per dag. Dit mogen totalen zijn, details zijn hier niet zo belangrijk. Zo kan ik zien wat het verbruik per dag in een bepaalde periode.

dennissiemensma, 2019-01-05:
Bedankt voor je verzoek, dat lijkt me ook zeker een nuttige aanvulling! De komende tijd ben ik minder actief met DSMR-reader, dus het zal even duren voordat ik hier concreet wat mee kan doen.

dennissiemensma, 2019-10-03:
Ik combineer je wens met #556. Dat gaat over hetzelfde, maar dan op jaarniveau.

closed 2019-10-03 by dennissiemensma