dsmrreader/dsmr-reader #579: Email/Notification on power limit exceeded

hjkoster400d, 2019-01-06:
It would be nice to be able setting a power limit. In my case we have a single phase 40A fuse, i would love to know if the usage get's above the 9000 watt. Because when this happens a lot, the main fuse of single 40A will be replaced with 3x 25A.

Hope that this is an welcome feature ;)

dennissiemensma, 2019-01-06:
Bedankt voor je suggestie. Dat is opzich een prima idee. Het zal wel enige tijd duren voordat ik er weer aan toekom zoiets voor je in te bouwen.

dennissiemensma, 2019-01-06:
Je kunt dit trouwens tot die tijd zelf ook in de gaten houden door af en toe aan de database handmatig te vragen naar metingen waar dit gebeurde:

```
sudo su - postgres
psql dsmrreader

select read_at, currently_delivered from dsmr_consumption_electricityconsumption where currently_delivered > 9 order by read_at desc;

# Waarbij 9 hierboven de waarde in kW is. Je kunt ook decimalen gebruiken, zoals 8.5 (8500 Watt).
```

Let wel dat metingen standaard gegroepeerd worden per minuut, dus je zult hier geen pieken mee vinden, alleen hoge gemiddelden. Wil je wel pieken, dan kun je het groeperen van de metingen per minuut uitzetten.

dennissiemensma, 2019-04-28:
@hjkoster400d heeft bovenstaande query je al afdoende geholpen in het maken van je keuze destijds? Of wellicht nu, na enkele maanden. 

Opzich kan ik een feature als deze op de lange termijn maken, maar liever alleen als er echt veel vraag naar is. En ik heb ook het idee dat je vraagstuk eigenlijk eenmalig is ("moet je de groep vervangen door 3x25A of niet?").

closed 2019-05-13 by dennissiemensma