dsmrreader/dsmr-reader #581: Voltages via MQTT

cranberryNL, 2019-01-10:
@dennissiemensma In een eerdere vraag over het loggen van voltages heb je aangegeven dit in de toekomst wellicht in te gaan bouwen. Tevens gaf je toen de suggestie om gebruik te maken van MQTT om zo complete diagrammen (of delen daarvan) te loggen. Gisteren ben ik hier verder in gedoken en het ik krijg netjes de rauwe data vanuit de slimme door DSMR reader doorgestuurd!

Graag zou ik gebruik maken van de 'split topic' functie. Ik mis daar echter nog de mapping fields voor de spanning. Zou je dit t.z.t. bij één van de volgende updates wellicht een keer mee kunnen nemen? Of zie ik een mogelijkheid over het hoofd om dit zelf te realiseren?
Als het handig is hiervoor een DSMR5 telegram te ontvangen, laat het maar even weten.

dennissiemensma, 2019-01-10:
Bedankt, dat lijkt me inderdaad een goed aanvulling! Ik ga pas medio april weer verder met ontwikkelen, dus het zal wel even duren.

Giel538, 2019-04-03:
@dennissiemensma 

Hoi,
Ik heb nog een paar vraagjes/toevoegingen over MQTT.
De voltages/stromen zou ik er ook wel bij willen zien. 
Is het ook mogelijk om het actuele gasverbruik over MQTT te sturen? Misschien heb ik er over heen gekeken maar ik kon hem niet vinden. Voor een DSMR5 meter krijg je elke 5 minuten een update. Je zou dus de (nieuwe waarde) - (vorige meting) moeten uitrekenen en deze op MQTT beschikbaar maken. Je krijgt dan elke 5 minuten een update. 

Over updates gesproken. Nu worden alle MQTT berichten elke seconde eruit gezonden ook al is er geen update. Bijvoorbeeld de spannings dips en power failures komen amper voor maar per dag krijg ik wel 60x60x24= heel veel berichten.

dennissiemensma, 2019-04-03:
Bedankt voor je suggesties. Dat zijn goede aanvullingen. Ik zal het meenemen indien mogelijk en anders splits ik ze als een aparte feature. Lijkt me nuttig.

Giel538, 2019-04-05:
Ik las dat je af zelf geen dsmr5 meter had. Mocht ik iets moeten testen moet je het maar laten weten.

dennissiemensma, 2019-04-15:
@Giel538 mbt je vraag over MQTT, is dat hetzelfde als in #611? Want anders stel ik voor om dat issue te volgen en eventueel aanvullingen te doen. Het is handiger om een apart issue te hebben per onderwerp.

Giel538, 2019-04-15:
@dennissiemensma Ja hoor prima, komt op hetzelfde neer

dennissiemensma, 2019-04-28:
Dit wordt een vervolg op #518, die waarschijnlijk over twee feature releases opgepakt wordt.

dennissiemensma, 2019-09-29:
In aankomende release komen de voltages in het dashboard de database. Daarna kunnen MQTT en API als followup

Giel538, 2019-09-29:
Top. ik ga er naar kijken!

dennissiemensma, 2019-09-30:
Ik heb dit ook werkend, bleek zeer eenvoudig uit te breiden. Gaat mee in v2.6

closed 2019-09-30 by dennissiemensma

dennissiemensma, 2019-10-01:
Bij upgraden naar v2.5 is dit ook al mogelijk. De mapping is pas gedocumenteerd in v2.6, maar werkt ook al in v2.5.

JSON:
```
phase_voltage_l1 = phase_voltage_l1
phase_voltage_l2 = phase_voltage_l2
phase_voltage_l3 = phase_voltage_l3
```

Split-topic:
```
phase_voltage_l1 = dsmr/reading/phase_voltage_l1
phase_voltage_l2 = dsmr/reading/phase_voltage_l2
phase_voltage_l3 = dsmr/reading/phase_voltage_l3
```

dennissiemensma, 2019-10-07:
Uitgebracht in [v2.6](https://github.com/dennissiemensma/dsmr-reader/releases/tag/v2.6.0).