dsmrreader/dsmr-reader #582: https and iframe include

remb0, 2019-01-23:
hoe kan ik ervoor zorgen dat dsmr-reader website via https loopt en iframe gebruik toestaat?
zodat ik hem in mijn home-assistant kan includen?

iemand hier ervaring mee?

dennissiemensma, 2019-01-23:
Voor iframe kun je het beste even zoeken in gesloten issues. Daar staat wel een howto.Voor SSL is het vooral een kwestie van de nginx vhost bijwerken. Het zou met elk certificaat moeten werken, zoals letsencrypt. Ik ben nu helaas niet in de gelegenheid om een exacte config voor je samen te stellen.

dennissiemensma, 2019-04-15:
@remb0 ben je hier inmiddels achtergekomen?

dennissiemensma, 2019-04-19:
Zie issue #344 voor hulp. Mocht je er desondanks nog steeds niet uitkomen, dan hoor ik het uiteraard graag.

closed 2019-04-19 by dennissiemensma