dsmrreader/dsmr-reader #583: installatie dsmr

hugothedog, 2019-01-23:
LET OP: Wegens verminderde beschikbaarheid tot aan medio april 2019,
        kan het zijn dat de auteur minder snel of helemaal niet in de
        gelegenheid is om te reageren. Zie ook issue #576.
----

### Jouw omgeving
*(Indien van toepassing)*
* DSMR-reader versie *(rechtsbovenin applicatie zichtbaar)*: `1.28
* Hardware *(RaspberryPi 2, 3 of iets anders)*: `RaspberryPi 3..`
* DSMR-protocol versie van slimme meter *(v2, v4 of v5)*: `v...`

### Bug, wens of iets anders
*Dank zij de zeer goede instructie was de installatie van de dsmr-applicatie een fluitje van een cent. Ik heb echter een probleem. Bij het aanklikken van de tab Configuratie verschijnt een venster Djangobeheer met de vraag naar gebruikersnaam en wachtwoord. Ik kan in de handleiding niet vinden wat ik daar dien in te vullen.
Ik  hoop niet dat ik in de instructie ergens over heen kijk.
Wellicht kunt u mij uit de droom helpen.
Alvast bedankt,
Servaas Brouns

dennissiemensma, 2019-01-24:
Bedankt voor je vraag. Het kan zijn dat je het gemist hebt.  Onderin stap 8 staan instructies hiervoor: https://dsmr-reader.readthedocs.io/nl/latest/installation/explained.html#bootstrappingVergeet niet eerst "sudo su - dsmr" te doen.

dennissiemensma, 2019-04-15:
@hugothedog is de installatie inmiddels gelukt?

closed 2019-04-28 by dennissiemensma