dsmrreader/dsmr-reader #584: Weather not always working

ramondeklein, 2019-01-23:
LET OP: Wegens verminderde beschikbaarheid tot aan medio april 2019,
        kan het zijn dat de auteur minder snel of helemaal niet in de
        gelegenheid is om te reageren. Zie ook issue #576.
----

### Jouw omgeving
* DSMR-reader versie v1.28
* Hardware: Raspberry Pi 3B+
* DSMR-protocol versie van slimme meter *(v2, v4 of v5)*: v5

### Bug
Als er iets fout gaat met weer-synchronisatie of Dropbox-synchronisatie, dan is dit nergens zichtbaar in de applicatie. Ik had het probleem dat de DNS niet goed stond, waardoor de weergegevens niet opgehaald konden worden. Je ziet dan gewoon geen weer-records (en dus geen grafiek), maar nergens is te zien dat de hostname niet geresolved kon worden. In de status zie je alleen het tijdstip van de volgende syncrhonisatie.

Hetzelfde geldt eigenlijk voor Dropbox. Ook die synchronisatie kan fout gaan en daar zie je helemaal niets van terug.

Het zou prettig zijn dat er wat meer informatie getoond kan worden bij deze problemen. Ik heb nu ook problemen dat het weer niet opgehaald wordt en de Dropbox synchronisatie werkt niet (alleen het initiële script is ooit gesynchroniseerd). Maar ik kan nergens ook maar iets vinden wat er mis kan zijn.

dennissiemensma, 2019-01-24:
Bedankt voor je opmerkingen. De indicatie bij fouten kan inderdaad stukken beter. Wellicht was de fout zichtbaar in de logs maar dat is niet heel praktisch merk ik. Dit wil ik na het voorjaar weer doen.

ramondeklein, 2019-01-24:
Het blijkt dat Twente sinds 23-01-2019 vanaf 21:00 uit de API van Buienradar is gehaald. Het is nu bijna 12 uur verder en die staat er nog steeds niet in:
```
<buienradarnl>
  <weergegevens>
    <titel>www.buienradar.nl - actuele weersinformatie</titel>
    <link>http://www.buienradar.nl</link>
    <omschrijving>Publieke weersinformatie feed</omschrijving>
    <language>nl-nl</language>
    <copyright>(C)opyright 2005 - 2019 Buienradar B.V. Alle rechten voorbehouden</copyright>
    <gebruik>Deze feed mag vrij worden gebruikt onder voorwaarde van bronvermelding buienradar.nl inclusief een hyperlink naar https://www.buienradar.nl. Aan de feed kunnen door gebruikers of andere personen geen rechten worden ontleend.</gebruik>
    <image>
      <titel>buienradar.nl</titel>
      <link>http://www.buienradar.nl/</link>
      <url>http://www.buienradar.nl/resources/images/logor.svg</url>
      <width>950</width>
      <height>78</height>
    </image>
    <actueel_weer>
      <weerstations>
        <weerstation id="6391">
          <stationcode>6391</stationcode>
          <stationnaam regio="Venlo">Meetstation Arcen</stationnaam>
          <lat>51.50</lat>
          <lon>6.20</lon>
          <datum>01/24/2019 07:50:00</datum>
          <luchtvochtigheid>79</luchtvochtigheid>
          <temperatuurGC>-4.0</temperatuurGC>
          <windsnelheidMS>1.5</windsnelheidMS>
          <windsnelheidBF>1</windsnelheidBF>
          <windrichtingGR>17</windrichtingGR>
          <windrichting>NNO</windrichting>
          <luchtdruk>-</luchtdruk>
          <zichtmeters>-</zichtmeters>
          <windstotenMS>2.9</windstotenMS>
          <regenMMPU>-</regenMMPU>
          <zonintensiteitWM2>-</zonintensiteitWM2>
          <icoonactueel ID="cc" zin="Zwaar bewolkt">https://www.buienradar.nl/resources/images/icons/weather/30x30/cc.png</icoonactueel>
          <temperatuur10cm>-4.0</temperatuur10cm>
          <url>http://www.buienradar.nl/nederland/weerbericht/weergrafieken/6391</url>
          <latGraden>51.83</latGraden>
          <lonGraden>6.33</lonGraden>
        </weerstation>
        <weerstation id="6275">
          <stationcode>6275</stationcode>
          <stationnaam regio="Arnhem">Meetstation Arnhem</stationnaam>
          <lat>52.07</lat>
          <lon>5.88</lon>
          <datum>01/24/2019 07:50:00</datum>
          <luchtvochtigheid>84</luchtvochtigheid>
          <temperatuurGC>-4.5</temperatuurGC>
          <windsnelheidMS>1.23</windsnelheidMS>
          <windsnelheidBF>1</windsnelheidBF>
          <windrichtingGR>78</windrichtingGR>
          <windrichting>ONO</windrichting>
          <luchtdruk>1010.44</luchtdruk>
          <zichtmeters>5220</zichtmeters>
          <windstotenMS>1.9</windstotenMS>
          <regenMMPU>-</regenMMPU>
          <zonintensiteitWM2>-</zonintensiteitWM2>
          <icoonactueel ID="cc" zin="Zwaar bewolkt">https://www.buienradar.nl/resources/images/icons/weather/30x30/cc.png</icoonactueel>
          <temperatuur10cm>-4.7</temperatuur10cm>
          <url>http://www.buienradar.nl/nederland/weerbericht/weergrafieken/6275</url>
          <latGraden>52.11</latGraden>
          <lonGraden>6.47</lonGraden>
        </weerstation>
        <weerstation id="6249">
          <stationcode>6249</stationcode>
          <stationnaam regio="Berkhout">Meetstation Berkhout</stationnaam>
          <lat>52.65</lat>
          <lon>4.98</lon>
[... 1222 lines cut ...]
    </verwachting_meerdaags>
    <verwachting_vandaag>
      <url>http://www.buienradar.nl/nederland/weerbericht</url>
      <urlbackup>http://www.buienradar.nl/nederland/weerbericht</urlbackup>
      <titel>In veel regio's tamelijk bewolkt weer</titel>
      <tijdweerbericht>Opgesteld op donderdag 24 jan 2019 om 05:00</tijdweerbericht>
      <samenvatting>Voornamelijk in het noordoosten heeft het gisteravond en tijdens de afgelopen nacht, dankzij enkele opklaringen, matig tot streng gevroren. Elders werd het warmteverlies getemperd door hardnekkige bewolking. Ook vandaag overdag is het vaak bewolkt met later aan zee kans op wat winterse neerslag.</samenvatting>
      <tekst>Voornamelijk in het noordoosten heeft het gisteravond en tijdens de afgelopen nacht, dankzij enkele opklaringen, matig tot streng gevroren. Elders werd het warmteverlies getemperd door hardnekkige bewolking. Ook vandaag overdag is het vaak bewolkt met later aan zee kans op wat winterse neerslag. Vanochtend maakt het oosten kans op enkele opklaringen. In de rest van het land overheersen de wolken en lokaal kan er nog wat motsneeuw vallen. De wind is overwegend zwak en veranderlijk.Vanmiddag zijn er nog steeds vrij veel wolkenvelden en later kan er met name in het westelijk kustgebied plaatselijk wat (winterse) neerslag vallen. Daardoor is gladheid mogelijk. De wind blijft zwak en komt uit uiteenlopende richtingen. De maximumtemperaturen schommelen tussen -3 graden in oostelijk Groningen en Drenthe tot +3 in het Zeeuwse kustgebied.Vanavond en de komende nacht kan er in de kustgebieden een enkele (winterse) bui vallen. In het binnenland zijn opklaringen mogelijk. Daar waar het langer opklaart, daalt de temperatuur naar waarden tussen -5 en -8: in streken met een gesloten sneeuwdek koelt het wellicht nog iets verder af. In Zeeland komt de temperatuur echter maar net onder nul. De wind is landinwaarts zwak en veranderlijk. Aan de kust waait een zwakke tot matige zuidenwind.Morgen worden de wolken van het westen uit dikker. Later op de ochtend valt er in het westen en noordw [... cut ...]
2019-01-24 07:12:25,641 INFO exited: dsmr_mqtt (exit status 1; not expected)
2
```
Dit loopt eigenlijk continu door, dus als *dsmr_mqtt* stopt, dan wordt die elke keer weer opgestart. Ik heb niets geconfigureerd aan MQTT, dus wellicht dat het daar vandaarn komt. Hier heb ik ook geen last van.

Over de dropbox synchronisatie (die alleen de eerste keer wat heeft gedaan) zie ik niets terug. Kan ik ergens het log-level verhogen zodat ik wat meer inzicht kan krijgen?

Mooi project en erg netjes opgezet!

dennissiemensma, 2019-01-24:
Bedankt voor je vragen. –Wat betreft de weer-api, ik kan pas medio april inhoudelijk voor je ernaar kijken.–Van mqtt klopt het. In een toekomstige versie zet ik mqtt standaard uit, ipv standaard aan. Dat moet het verhelpen. –En qua logging staan hier instructies voor verhogen log level: https://dsmr-reader.readthedocs.io/nl/latest/troubleshooting.html#logging

dennissiemensma, 2019-04-15:
@ramondeklein speelt het issue met buienradar nog steeds?

ramondeklein, 2019-04-16:
Nee, het lijkt er op dat Buienradar de laatste tijd dit weer opneemt. Ik vermoed dat het weerstation er gewoon tijdelijk uit heeft gelegen.

dennissiemensma, 2019-04-16:
Dat vermoed ik ook. Bedankt voor je reactie in ieder geval. :]

dennissiemensma, 2019-10-01:
Afgelopen tijd is er al aardig wat verbeterd. Ik heb nu nog wat extra foutmeldingen toegevoegd voor de volgende versie.

closed 2019-10-01 by dennissiemensma