dsmrreader/dsmr-reader #585: [1.28] Dashboard laat na herinstallatie geen actueel wattage zien

voipmeister, 2019-01-24:
LET OP: Wegens verminderde beschikbaarheid tot aan medio april 2019,
        kan het zijn dat de auteur minder snel of helemaal niet in de
        gelegenheid is om te reageren. Zie ook issue #576.
----

### Jouw omgeving
*(Indien van toepassing)*
* DSMR-reader versie *(rechtsbovenin applicatie zichtbaar)*: `v1.28`
* Hardware *(RaspberryPi 2, 3 of iets anders)*: `Gigabyte Brix`
* DSMR-protocol versie van slimme meter *(v2, v4 of v5)*: `v...`

### Bug, wens of iets anders
TL;DR: Dashboard laat geen actueel wattage zien na herinstallatie (op CentOS 7)

Gisteren attendeerde mijn vriendin mij erop dat DSMR-reader niets meer deed op de Pi. SSH werkte niet meer en de Pi bootte ook niet meer. Weer een SD-kaart kapot. Toen maar besloten om de boel op mijn Gigabyte Brix vpshost met CentOS 7 te installeren. Dat was een interessante ervaring, die ik gedocumenteerd heb (ik hoor graag even van je of je die uitgewerkt zou willen hebben voor je documentatie, er zitten nogal wat onverwachte dingen in namelijk :-) ).

Installatie is uiteindelijk gelukt, helaas was de meeste recente back-up stuk. In de tabel met metingen werd midden in een record de boel afgekapt. Ik heb een back-up gepakt van 2 weken terug (lang leve de dropbox push) die netjes afgesloten werd met `\.`.

Alles draait weer. Na de import was DSMR-reader nog wel even druk (postgres %cpu) met 'bijkomen' had ik het idee. Het enige dat nu niet lijkt te werken is het tonen van het actuele wattage rechtsbovenin de dashboard-pagina. De metingen worden wel goed gedaan, want de grafiek is wel up-to-date en lijkt goede waardes te bevatten.

Mocht je tijd en zin hebben dan hoor ik graag even je suggestie(s) om verder naar te kijken. De metingen lopen, dus ik ben al blij :-)

dennissiemensma, 2019-01-24:
Bedankt voor je melding. Helaas ben ik niet eerder dan in april in de gelegenheid om hier inhoudelijk voor je naar te kijken. Je zou eens op F12 kunnen drukken in je browser om te zien of daar fouten in staan.

dennissiemensma, 2019-04-15:
@voipmeister heb je hier inmiddels nog aanvullingen voor? Heb je het issue kunnen reproduceren of oplossen of speelt het nog steeds?

voipmeister, 2019-04-18:
Hoi @dennissiemensma , welkom terug :-)
Ik heb hier verder nog niet naar gekeken, de metingen door de dag heen worden wel goed opgeslagen en dat vond ik het belangrijkste in eerste instantie. Met F12 zie ik geen fouten. Wordt de laatste waarde uit een tabel gelezen (als nieuwste waarde), of wordt er een specifiek veld bijgewerkt met de laatste waarde na een meting?

dennissiemensma, 2019-04-18:
Voor elke meting wordt een nieuw record gemaakt. En die wordt samengevoegd in records voor per minuut (voor stroom). Na afloop van een dag worden deze records samengevoegd tot één per uur/dag, voor het archief. Dus er zijn meerdere records en de meesten blijven ook achter. Tenzij je geen retentie gebruikt, dan blijft alles bewaard.

voipmeister, 2019-04-18:
Ik zie dat ik geen retentie geconfigureerd heb. Postgres is ook erg druk, als ik de logging aanzet loopt deze behoorlijk vol met queries die continu worden uitgevoerd.

dennissiemensma, 2019-04-18:
Je zou de datalogger wat rustiger aan kunnen laten doen. Als je nu elke seconde een meting hebt, kun je die terugschroeven naar eens per 5 of 10 seconden:

```
sudo su - dsmr
vi dsmrreader/settings.py

# Voeg onderaan een regel toe (of pas hem aan als die er al is):
DSMRREADER_DATALOGGER_SLEEP = 5

./post-deploy.sh
```

Als het goed is wordt het dan wat rustiger, zonder dat je echt wat nauwkeurigheid verliest. Je kunt hem eventueel zelfs op `10` zetten.

voipmeister, 2019-04-18:
Dank voor de suggestie, ik heb die toegepast. Helaas helpt het niet echt, Postgres belast eigenlijk continu 2 cores. Ik heb het idee dat er een compactor actie o.i.d. blijft hangen (en dat dat al een maand of 3 zo is). Is dat voor jou makkelijk te zien op basis van de SQL logging van Postgres eventueel?

dennissiemensma, 2019-04-19:
Ik denk dat je het beste even logging kunt aanzetten op `DEBUG`: https://dsmr-reader.readthedocs.io/en/latest/troubleshooting.html#logging

En dan vervolgens kijken in de Supervisor log van de backend wat die exact aan het doen is.

voipmeister, 2019-04-19:
Done, dat lijkt normaal.
`[2019-04-19 16:19:12,443] INFO     Calling backend services
[2019-04-19 16:19:17,559] DEBUG    Sleeping 1 sec(s)
[2019-04-19 16:19:18,564] INFO     Calling backend services
[2019-04-19 16:19:22,014] INFO     Detected signal #1
[2019-04-19 16:19:22,014] INFO     Exiting on next run...
[2019-04-19 16:19:25,094] DEBUG    Sleeping 1 sec(s)
[2019-04-19 16:19:26,100] INFO     Exited
[2019-04-19 16:19:28,457] DEBUG    Starting infinite command loop...
[2019-04-19 16:19:28,457] INFO     Calling backend services
[2019-04-19 16:19:31,199] DEBUG     - Processed reading: 740599 @ 2019-04-19 16:18:04+02:00 (0.240 kW)
[2019-04-19 16:19:31,969] DEBUG     - Processed reading: 740600 @ 2019-04-19 16:18:14+02:00 (0.242 kW)
[2019-04-19 16:19:32,730] DEBUG     - Processed reading: 740601 @ 2019-04-19 16:18:24+02:00 (0.243 kW)
[2019-04-19 16:19:33,490] DEBUG     - Processed reading: 740602 @ 2019-04-19 16:18:34+02:00 (0.243 kW)
[2019-04-19 16:19:34,252] DEBUG     - Processed reading: 740603 @ 2019-04-19 16:18:44+02:00 (0.243 kW)
[2019-04-19 16:19:35,013] DEBUG     - Processed reading: 740604 @ 2019-04-19 16:18:55+02:00 (0.243 kW)
[2019-04-19 16:19:36,119] DEBUG    Sleeping 1 sec(s)
[2019-04-19 16:19:37,125] INFO     Calling backend services
[2019-04-19 16:19:39,953] DEBUG    Sleeping 1 sec(s)
[2019-04-19 16:19:40,959] INFO     Calling backend services
[2019-04-19 16:19:43,735] DEBUG    Sleeping 1 sec(s)
[2019-04-19 16:19:44,740] INFO     Calling backend services
[2019-04-19 16:19:47,514] DEBUG    Sleeping 1 sec(s)
[2019-04-19 16:19:48,520] INFO     Calling backend services
[2019-04-19 16:19:52,077] DEBUG    Sleeping 1 sec(s)
[2019-04-19 16:19:53,083] INFO     Calling backend services
[2019-04-19 16:19:56,660] DEBUG    Sleeping 1 sec(s)
[2019-04-19 16:19:57,666] INFO     Calling backend services
[2019-04-19 16:20:02,019] DEBUG    Sleeping 1 sec(s)
[2019-04-19 16:20:03,025] INFO     Calling backend services
[2019-04-19 16:20:07,371] DEBUG    Sleeping 1 sec(s)
[2019-04-19 16:20:08,376] INFO     Calling backend services
[2019-04-19 16:20:13,484] DEBUG    Sleeping 1 sec(s)
[2019-04-19 16:20:14,490] INFO     Calling backend services
[2019-04-19 16:20:19,586] DEBUG    Sleeping 1 sec(s)
[2019-04-19 16:20:20,591] INFO     Calling backend services
[2019-04-19 16:20:27,238] DEBUG     - Processed reading: 740610 @ 2019-04-19 16:19:54+02:00 (0.235 kW)
[2019-04-19 16:20:27,400] DEBUG    Sleeping 1 sec(s)
[2019-04-19 16:20:28,406] INFO     Calling backend services
[2019-04-19 16:20:30,231] DEBUG     - Processed reading: 740605 @ 2019-04-19 16:19:05+02:00 (0.244 kW)`

Als ik die Postgres logging ernaast open, loopt die eigenlijk continu door met dsmrreader queries, met vrij hoge belasting voor het systeem. Kun je iets roepen over hoe druk Postgres mag zijn op een goed werkende installatie?

dennissiemensma, 2019-04-19:
In principe zou die niet heel druk moeten zijn, afgezien van een paar momenten per dag.

Je zou even `dsmr_backend` kunnen stoppen in supervisor voor een minuutje, om te kijken of dat de boosdoener is. En anders hetzelfde proberen met `dsmr_datalogger`.

voipmeister, 2019-04-19:
Goede tip. Het zit 'm in de dsmr_backend. Als ik die stop, is de CPU-belasting minimaal. Start ik 'm weer, dan is de belasting ook weer terug. Ik heb de Postgres logging er even bijgepakt van een korte periode waarin de dsmr_backend aanstond en de dsmr_datalogger uit. Voor mij lijken d'r veel dubbele dingen in te zitten, maar misschien wil je ook blik werpen als je de tijd en zin ervoor kunt vinden.

[dsmr_backend-logging.txt](https://github.com/dennissiemensma/dsmr-reader/files/3099388/dsmr_backend-logging.txt)

dennissiemensma, 2019-04-20:
Ik heb even gekeken, maar het is vrij lastig zonder dat er timestamps bijzitten. Het klopt namelijk dat er wat dubbele queries gedaan worden, want elke seconde kijkt de backend of er wat gedaan moet worden.

Je zou de backend ook nog een sleep kunnen geven, van bijvoorbeeld 5 seconden. Zie https://dsmr-reader.readthedocs.io/en/latest/settings.html#dsmrreader-backend-sleep. Het werkt hetzelfde als bovengenoemde instelling voor de datalogger.

dennissiemensma, 2019-04-28:
@voipmeister heeft het verhogen van een sleep op de backend geholpen? Ik doel dus op de andere optie: `DSMRREADER_BACKEND_SLEEP`.

voipmeister, 2019-05-04:
Ik heb die suggestie nog niet toe kunnen passen, ik ga ervan uit dat ik dit weekend terugkoppeling kan geven, thanks!

dennissiemensma, 2019-05-04:
Ik zou dan helaas geen andere oorzaken meer weten. Bij mij is op een Pi 3 de load rond de `0.3`. Ik heb dan wel een meter die slechts elke 10 seconden een meting doorgeeft, maar dan nog.

dennissiemensma, 2019-05-13:
Zoals eerder besproken laatst, weet ik geen verdere oplossingen meer voor je lokale issues. Dan zou ik een keertje moeten meekijken in je database. Dat zal dan buiten Github om gaan.

closed 2019-05-13 by dennissiemensma

voipmeister, 2019-08-08:
Uiteindelijk gefikst door een nieuwe installatie op te tuigen en een restore te doen. Deze keer gebruik ik weer een pi i.c.m. een remote postgres database. Alles lijkt weer goed te werken.

dennissiemensma, 2019-08-09:
Top, fijn om te horen!