dsmrreader/dsmr-reader #586: Feature Request: Grafiektype temperatuurweergave

sjaakvanwijk, 2019-01-24:
### Jouw omgeving
* DSMR-reader versie 1.27
* Hardware: Raspberry Pi 3 Model B Rev 1.2
* Datalogger: ESP8266

### feature, 'nice to have': Ander grafiektype voor temperatuurweergave
De elektriciteitsgrafieken maken gebruik van een lijn met daaronder een gearceerd/gekleurd oppervlak. Dat is prima omdat het oppervlak tevens een maat is voor de hoeveelheid verbruikte/geproduceerde energie.
Bij de temperatuurgrafiek heeft het oppervlak onder de lijn echter geen betekenis. Daarom heeft het geen zin deze gearceerd of ingekleurd weer te geven. Dit viel mij met name op nu de laatste dagen de temperatuur steeds onder nul is. Wellicht is het een goed idee om voor de temperatuur een ander grafiek type te kiezen.
Verder niets dan lof voor dit product. Draait lekker stabiel.
m.vr.gr.
Sjaak van Wijk

dennissiemensma, 2019-01-24:
Bedankt voor je suggestie. Ik zal hier medio april inhoudelijk naar kijken om exact te zien waar je op doelt. :)

sjaakvanwijk, 2019-04-15:
@dennissiemensma Van alle grafieken is de (buiten)temperatuur de enige grafiek die waarden onder nul kan aannemen. Jammer dat afgelopen nacht de temperatuur niet onder nul is geweest. Anders had ik het kunnen laten zien. Het dagprofiel van een paar dagen geleden kan ik niet uit het archief toveren. Onderstaande grafiek is degene die ik bedoel:
[image]
Ik doel op het gearceerde oppervlak tussen de lijn en de x-as. Bij temperatuurweergave heeft het oppervlak geen betekenis, in tegenstelling tot alle andere grafieken: Daar zegt het iets over de totale productie/verbruik gedurende een bepaalde tijd. De suggestie is dus: Bij temperatuurweergave geen arcering gebruiken maar alleen een lijn weergeven.
Nogmaals, het is een klein dingetje. Kijk maar wat je er mee wilt doen.
grt, Sjaak

dennissiemensma, 2019-04-15:
Aah juist, ik zie nu wat je bedoelt! In dat opzicht klopt de arcering eveneens niet bij de andere grafieken, want het is geen optelling over langere tijd, maar puur voor de zichtbaarheid/onderscheid. Wellicht dat ik dat dan later nog verander, maar voor nu laat ik het dan voor wat het is, om alle grafieken consistent te houden.

dennissiemensma, 2019-09-10:
Voor nu laat ik dit voor wat het is. Ik ben het met je eens dat je het kan aflezen als een andere betekenis, maar ik houd ze voorlopig gelijk. Wellicht dat ik het verander als ik alle grafieken ooit op de schop gooi.

closed 2019-09-10 by dennissiemensma