dsmrreader/dsmr-reader #587: Detailniveau daggrafieken in archief

henkjanwithaar, 2019-01-26:
Het detailniveau van de grafiek waarop in het dashboard de laatste 24 uur wordt geplot is per minuut. In het archief kennen de grafieken een detailniveau per uur. Ik besef met dat het wellicht buiten de scope van deze tool ligt en als gebruiker van een lucht/water-warmtepomp vind ik het interessant te zien wat de relatie tussen buitentemperatuur en aantal 'ontdooiruns' van de buitenunit is. Ik log mijn gegevens naar PVOutput met een interval van 5 minuten en dat biedt voldoende detail. Ik gebruik echter liever DSMR als monitortool.

[image]

Is het mogelijk het detailniveau in de daggrafieken van het archief te verhogen van één uur naar ~5 minuten?

dennissiemensma, 2019-01-29:
Bedankt voor je verzoek en ik begrijp je wens. Het is helaas niet mogelijk, omdat de daggrafieken niet nauwkeuriger zijn dan per uur. Dit is om de data compact en snel opvraagbaar te maken.

dennissiemensma, 2019-04-15:
Ik kijk hier op een later moment nog opnieuw naar. Op korte termijn kan ik hier sowieso geen ondersteuning voor maken, gezien het iets van de architectuur is waar ik destijds bewust voor heb gekozen. Ik zal later kijken of ik er een instelling van kan maken, mits het niet teveel dingen raakt.

dennissiemensma, 2019-10-02:
Ik kan dit helaas niet faciliteren. Het archief is echt een vaste set aan data en bewust heel grof gehoud op uur/dag. 

Wat ik nog wel eventueel zou kunnen doen in de toekomst, is een aparte pagina maken waar je zelf kan zoeken in de geschiedenis van metingen en dat plotten in een grafiek. Dus eigenlijk wat het dashboard is, maar dan met een selecteerbare dag.

dennissiemensma, 2020-08-20:
Ik zie hier uiteindelijk geen mogelijkheden voor. 

Wellicht dat het indirect kan met een variant op #1084, waarbij ik die interval (daar 15 minuten) instelbaar maak oid. Het doel is anders, maar het gaat wel om pieken.

closed 2020-08-20 by dennissiemensma