dsmrreader/dsmr-reader #589: Installatie lukt niet op Ubuntu server

eterpstra, 2019-02-06:
Hallo Dennis,

Ik probeer een versie installatie te doen op een mini pc.
Het betreft  Ubuntu server 18.04.  Ik heb de snelle stappen doorlopen, maar nu loop ik vast op het volgende commando:

sudo sudo -u dsmr /home/dsmr/.virtualenvs/dsmrreader/bin/pip3 install -r /home/dsmr/dsmr-reader/dsmrreader/provisioning/requirements/base.txt -r /home/dsmr/dsmr-reader/dsmrreader/provisioning/requirements/postgresql.txt

Als ik dit invul dan krijg ik de volgende permission denied error:

The directory '/home/eterpstra/.cache/pip/http' or its parent directory is not owned by the current user and the cache has been disabled. Please check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
The directory '/home/eterpstra/.cache/pip' or its parent directory is not owned by the current user and caching wheels has been disabled. check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
Collecting crcmod==1.7 (from -r /home/dsmr/dsmr-reader/dsmrreader/provisioning/requirements/base.txt (line 1))
Could not install packages due to an EnvironmentError: [Errno 13] Permission denied: '/home/eterpstra/.cache/pip/wheels/50/24/4d/4580ca4a299f1ad6fd63443e6e584cb21e9a07988e4aa8daac'
Consider using the `--user` option or check the permissions.

De vorige stappen heeft hij wel gewoon netjes doorlopen dus vraag mij af waar het hem in zit.

Kun jij hierbij helpen? Of wellicht iemand anders?

Gr,
Elroy

jeroenpeters1986, 2019-02-07:
Hoi @eterpstra, onder welke user werk je normaalgesproken?

Zou je ook eens de output van `ls -la /home/eterpstra/.cache` willen posten? 

Het ziet er naar uit dat de dsmr user geen rechten heeft om te werken in de pip-map, die is nodig om de libraries te installeren

dennissiemensma, 2019-04-15:
@eterpstra ben je hier inmiddels uit gekomen?

jossie67, 2019-04-26:
Ik heb het zelfde issue maar dan iop Ubunut 16.04

<code>
WARNING: The directory '/home/jos/.cache/pip/http' or its parent directory is not owned by the current user and the cache has been disabled. Please check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
WARNING: The directory '/home/jos/.cache/pip' or its parent directory is not owned by the current user and caching wheels has been disabled. check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
Collecting crcmod==1.7 (from -r /home/dsmr/dsmr-reader/dsmrreader/provisioning/r
</code>

dennissiemensma, 2019-04-26:
@jossie67 dit zijn slechts warnings, maar mogelijk zit je onder een verkeerde user. 

Ik heb wel wat meer context nodig over wat je aan het doen was, welke stap, etc. Krijg je verder nog foutmeldingen of lukt de installatie verder wel?

eterpstra, 2019-04-26:
@dennissiemensma Ik heb het inmiddels voor elkaar. Bij mij was het inderdaad een rechten kwestie die niet goed stond. Er ontbrak een alias.

dennissiemensma, 2019-04-26:
Top, bedankt voor je terugkoppeling

closed 2019-04-28 by dennissiemensma