dsmrreader/dsmr-reader #590: Handmatig metingen invoeren

wschoot, 2019-02-06:
Hoi!

Simpele vraag, maar kon 't zo 1-2-3 niet vinden; ik heb nog geen slimme meter maar vind de app er erg mooi uitzien. Ik houd braaf elke maand alle meterstanden bij en wil die graag een app als deze kunnen plotten. 

Is er ondersteuning voor (enkel) handmatige input van metingen? 

Ik zie bij de vereisten dat een slimme meter en/of kabel benodigd is en vermoed dus van niet, maar 't zou toch tof zijn als 't kon :-)

jeroenpeters1986, 2019-02-07:
Hoi @wschoot er is niet echt een manier om handmatig je metingen in te vullen via de admin of iets dergelijks. 

Mocht je het toch willen proberen én handig zijn met programmeren, dan zou je nog kunnen overwegen iets te bouwen die de metingen via de API insert. Daar staat informatie over op https://dsmr-reader.readthedocs.io/en/latest/api.html#id2
Hoe de applicatie dit weergeeft als je het slechts éénmaal per maand in de database zet weet ik niet precies.

dennissiemensma, 2019-02-08:
Bedankt voor je vraag. Het is inderdaad niet mogelijk via de interface, omdat de applicatie regelmatig meterstand nodig heeft voor goede gegevens. Ik kan echter handmatig invoeren in de toekomst wel toevoegen, al weet ik niet of alle data dan nog accuraat is.

wschoot, 2019-02-19:
Hey, dank voor je reacties beiden. Als de app er niet op is gebouwd, doe vooral niet teveel moeite. De markt voor "analogen" als ik zal niet groot zijn dus echt een breed publiek maak je er niet perse blij mee :)
 Ik zocht naar een mooie manier om van m'n excelsheet iets fanciers te maken maar heb nog niet de luxe van een slimme meter

dennissiemensma, 2019-04-15:
@wschoot bedankt voor je reactie. In dat geval laat ik dit voor wat het is. 

Want de applicatie heeft eigenlijk twee metingen per dag nodig voor accurate data. Anders krijg je hele vreemde grafieken met je complete maandverbruik op één dag.

closed 2019-04-15 by dennissiemensma