dsmrreader/dsmr-reader #591: Geen data piektarief

broekieman, 2019-02-10:
LET OP: Wegens verminderde beschikbaarheid tot aan medio april 2019,
        kan het zijn dat de auteur minder snel of helemaal niet in de
        gelegenheid is om te reageren. Zie ook issue #576.
----

### Jouw omgeving
*(Indien van toepassing)*
* DSMR-reader versie *(rechtsbovenin applicatie zichtbaar)*: `v1.X.Y`
* Hardware *(RaspberryPi 2, 3 of iets anders)*: `RaspberryPi X...`
* DSMR-protocol versie van slimme meter *(v2, v4 of v5)*: `v...`

### Bug, wens of iets anders
* Bijvoorbeeld stappen hoe een probleem te reproduceren is...
* Of simpelweg een beschijving van een feature die je graag wenst te zien...

dennissiemensma, 2019-02-10:
Bedankt voor je melding. Uit mijn hoofd zou ik niet zo 123 een oorzaak of oplossing weten. Helaas kan ik pas medio april inhoudelijk voor je kijken.

broekieman, 2019-02-11:
Ik heb de tarieven nu samenvoegd. Het is vreemd dat het telwerk voor het 
piektarief niet werkt terwijl de grafieken tijdens de piek gewoon worden 
opgebouwd en de totaalstand van electra ook klopt.

Met vriendelijke groet,

Maarten Broekman

------ Origineel bericht ------

<author@noreply.github.com>
Verzonden: 10-2-2019 14:35:26

dennissiemensma, 2019-04-15:
@broekieman werkt het inmiddels naar behoren of heb je nog hulp nodig?

broekieman, 2019-04-15:
Het loste zichzelf op, maar ik ben gestopt met de software. Dank voor je reactie. Met vriendelijke groet,Maarten Broekman-------- Oorspronkelijk bericht --------Onderwerp: Re: [dennissiemensma/dsmr-reader] Geen data piektarief (#591)Van: Dennis Siemensma Aan: dennissiemensma/dsmr-reader Cc: broekieman ,Mention @broekieman werkt het inmiddels naar behoren of heb je nog hulp nodig?

[
{
"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/dennissiemensma/dsmr-reader/issues/591#issuecomment-483282211",
"url": "https://github.com/dennissiemensma/dsmr-reader/issues/591#issuecomment-483282211",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

dennissiemensma, 2019-04-16:
Dank voor je reactie. In dat geval laat ik het voor wat het is.

closed 2019-04-16 by dennissiemensma