dsmrreader/dsmr-reader #592: feature request: reboot en shutdown opties in configuratie scherm

janwijbrand, 2019-02-11:
### Bug, wens of iets anders
Het zou handig zijn om vanuit de configuratie schermen (of in elk geval door admin credentials beschermd) een knop te hebben om de raspberry pi te herstarten of uit te zetten. Ter inspiratie: octopi heeft ook die optie in zijn UI en ik maak er vaker dan van te voren gedacht, gebruik van. Een iets minder ervaren gebruiker (ik heb verscheidende rpi's met dsmr reader aan familieleden gegeven) hoeft dan niet te SSHen om zo de shutdown te doen voordat, bijvoorbeeld, de stroom er even af moet.

p.s. Ik weet dat je op het moment geen tijd hebt, maar wie weer voor later. Ik zal ook zelf een poging doen dit feature toe te voegen.

dennissiemensma, 2019-02-11:
Bedankt voor je suggestie! Ik zal het later overwegen of het een goede plek heeft en makkelijk te realiseren is.

dennissiemensma, 2019-04-28:
Ik heb hier even naar gekeken, maar om dit te realiseren zou de `dsmr` gebruiker `sudo`-rechten moeten hebben. Dat heb ik bewust niet gedaan, zodat de gebruiker alleen bij het eigen project kan en geen andere rechten heeft. Daarmee kan ik dit helaas niet voor je realiseren.

closed 2019-04-28 by dennissiemensma