dsmrreader/dsmr-reader #593: 8n1 115200    7n1 9600

Arnaud48, 2019-02-11:
LET OP: Wegens verminderde beschikbaarheid tot aan medio april 2019,
        kan het zijn dat de auteur minder snel of helemaal niet in de
        gelegenheid is om te reageren. Zie ook issue #576.
----

### Jouw omgeving
*(Indien van toepassing)*
* DSMR-reader versie *(rechtsbovenin applicatie zichtbaar)*: `v1.X.Y`
* Hardware *(RaspberryPi 2, 3 of iets anders)*: `RaspberryPi X...`
* DSMR-protocol versie van slimme meter *(v2, v4 of v5)*: `v...`

### Bug, wens of iets anders
* Bijvoorbeeld stappen hoe een probleem te reproduceren is...
* Of simpelweg een beschijving van een feature die je graag wenst te zien...
De dsmr reader werkt op onze meter landis en gyr e350  8n1 115200 
Nu wil ook een andere meter meten dit is een ouder type een iskra mt832   7n1  9600 wat moet ik dan aanpassen , kan het niet vinden. 
Bedankt

Arnaud48, 2019-02-11:
Ik zag wel bij status dat er metingen voor verwerking klaar staan en weer verwerkt worden, echter zie geen nieuwe grafiek, alleen die van dag daarvoor van andere meter.

dennissiemensma, 2019-02-12:
Je kunt de instellingen hier wijzigen: https://dsmr-reader.readthedocs.io/nl/latest/admin/datalogger.htmlDe applicatie heeft verder (nog) geen ondersteuning voor verschillende of meerdere meters. Als je dezelfde instantie van dsmr-reader wisselt over meerdere meters, is de kans groot dat het niet goed verwerkt wordt.

Arnaud48, 2019-02-12:
Ik had getest met de landis en gyr dat ging goed.
Ook de instelling naar dsmr 2... gezet. Systeem opnieuw opgestart.
Moest er dan nog iets gedaan worden? 
Bedankt

dennissiemensma, 2019-02-13:
Je zult dan eigenlijk de database moeten leeggooien qua metingen en verwerkte data. Ik ben helaas niet in de gelegenheid om de exacte tabellen en commando's te maken voor je. Als alternatief kun je de hele database verwijderen en opnieuw aanmaken.

dennissiemensma, 2019-04-15:
@Arnaud48 ben je hier inmiddels uit gekomen of heb je nog meer assistentie nodig?

closed 2019-04-28 by dennissiemensma