dsmrreader/dsmr-reader #594: PV installatie koppeling

binijman, 2019-02-12:
LET OP: Wegens verminderde beschikbaarheid tot aan medio april 2019,
        kan het zijn dat de auteur minder snel of helemaal niet in de
        gelegenheid is om te reageren. Zie ook issue #576.
----

### Jouw omgeving
*(Indien van toepassing)*
* DSMR-reader versie *(rechtsbovenin applicatie zichtbaar)*: `v1.X.Y`
* Hardware *(RaspberryPi 2, 3 of iets anders)*: `RaspberryPi X...`
* DSMR-protocol versie van slimme meter *(v2, v4 of v5)*: `v...`

### Bug, wens of iets anders
* Bijvoorbeeld stappen hoe een probleem te reproduceren is...
* Of simpelweg een beschijving van een feature die je graag wenst te zien...

### Allereerst ... Wat een super app! Bedankt!

Hebben jullie overwogen om een verbinding te maken met Zonnepanelen KWh meters (Mbus , S0 of puls) of in mijn geval SolarEdge (fabrikant omvormer) API interface? Op deze manier kan niet alleen alle energie in huis goed in beeld gebracht worden maar belangrijker nog het daadwerkelijk verbruik van stroom (per fase) in huis gemeten worden ipv enkel de in en uitstroom naar de netbeheerder. Zou een fijne toevoeging zijn.

dennissiemensma, 2019-02-13:
Bedankt voor je suggestie. Ik heb dat zeker overwogen maar het is vrij complex om dat binnen de scope van dit project te doen. Vooral qua testen en ondersteuning van alle merken. Daarom heb ik besloten om alleen export ondersteuning te maken voor pvoutput, zodat die partij eea voor je kan berekenen.

pooh22, 2019-03-13:
Ik heb mijn eigen p1 uitlezer en de zonnepanelen lees ik uit via bluetooth. Het probleem van alleen de 
P1 uitlezen is dat je nooit precies weet hoeveel stroom je zelf verbruikt.

Een uitbreiding van deze `dsmr-reader` met een API om PV productie mee te nemen (bijvoorbeeld elke minuut de Etoday, de totale productie van de dag, ophalen of accepteren via een API) zou ervoor zorgen dat je een goed beeld krijgt van je eigen gebruik, bovenop energie afname/levering naar het net.

Voor mij is dit een essentiele feature voor een P1 uitlezer i.c.m. zonnepanelen!

dennissiemensma, 2019-03-14:
Bedankt voor je vraag, ik begrijp je wens helemaal. Ik heb nog niet nagedacht over een API, dat zou in de toekomst nog een optie zijn. Echter, het punt waar ik mee zit is dat het niet direct gerelateerd is aan de slimme meter (en dus dit project) en dat het aardig wat werk is om zoiets functioneel en correct te maken. Daarnaast is er ook al PVouput.org, waar diverse omvormers ook ondersteuning voor hebben. Is het een optie om je waarden daarheen te sturen, samen met de bestaande optie in dsmr-reader? Ik weet dat diverse gebruikers het nu op die manier doen.

pooh22, 2019-03-14:
Hoi Dennis, bedankt voor je antwoord. Ik wil vooral mijn eigen gegevens bij mij thuis houden (dus geen PVoutput.org). Het lijkt me dat als je als doel hebt een interface maken voor de slimme meter heb je het mooi voor elkaar met `dsmr-reader`, maar als je de naam van het project nou zou veranderen naar `energie-monitor` valt het uitlezen van de zonnepanelen (en powerwall??), en weergeven van je daadwerkelijke gebruik ineens wel binnen de omschrijving ;-)

Anyway, het lijkt me een mooie uitbreiding waar iedereen met zonnepanelen wat aan heeft. En die groep groeit natuurlijk gestaag ;-)

Poechi, 2019-04-02:
Ook ik zit met dezelfde vraag, echter ben ik zelf maar begonnen in een development omgeving om voor deze pracht applicatie een uitbreiding te maken voor het monitoren van zonnepanelen. Om het zo standaard mogelijk te houden is mijn voorwaarde een 'kWh dinrail meter' met een s0 output en als logger de S0PCM. Deze twee te samen zouden het gewenste resultaat moeten opleveren. Wel is het voor een niet programmeur een ware zoektocht (even Django en python leren). De admin interface is zo goed als klaar. Binnenkort de meter aansluiten en dan de output zien in te lezen :).

dennissiemensma, 2019-04-15:
@pooh22 bedankt voor je suggestie. Ik begrijp je wens helemaal en in principe is alles te maken. Alleen vraag ik me af of ik de tijd heb om zo'n groot project kwalitatief te onderhouden. Dus voor nu houd ik dat even af en focus ik me op de wat specifiekere verbeteringen van het project. Wellicht dat er op termijn iets generiekers in kan komen voor meer informatie.

dennissiemensma, 2019-04-15:
@Poechi bedankt voor je aanvulling. Ik begrijp je wens helemaal, maar het toevoegen van dergelijke functionaliteit maakt het project vrij complex. Dus op korte termijn kan ik hier geen ondersteuning voor geven.

R0yk3, 2019-08-23:
@dennissiemensma Top programma. Om wat minder afhankelijk te zijn het type en merk van de PV kun je wellicht een MQTT client bijvoegen. Ik heb een PI aan mijn PV hangen die nu de data via mqtt exporteert. 3 waarden, Watt op dit moment. KWH van vandaag en KWH overall. Ik lees dit nu in via mijn Domoticz maar zou fantastisch zijn om dat in dit dashboard te zien.
En als het via mqtt binnenkomt ben je niet afhankelijk het type omvormer wat gebruikt wordt.

braham2019, 2019-10-29:
@dennissiemensma : de SolarEdge API is volledig beschreven op https://www.solaredge.com/sites/default/files/se_monitoring_api.pdf. 

Ik gebruik hem al op een e-ink display (icm weerbericht van Darksky) om de totale opgewekte energie van de dag zelf op te vragen. Daar kan ik nu, dankzij jouw API, ook de opgevraagde en teruggegeven energie op tonen.

De gegevens van SolarEdge die interessant zouden zijn om mee op het DSMR dashboard te tonen, zijn "Huidig Vermogen", dit is hetgeen momenteel wordt opgewekt door de PV installatie. Deze API call kan om het kwartier gedaan worden. Het geeft dan wel mooi inzicht in hetgeen zelf opgewekt wordt, en verbruikt wordt.

dennissiemensma, 2019-10-29:
Bedankt voor je toevoeging. Het is op dit moment nog niet zeker of ik wel of niet externe PV-data ga opnemen in DSMR-reader. 
Mijn voorkeur heeft het exporteren van slimme meter data naar iets wat beide bronnen kan inlezen. Dit op de scope van het project beperkt te houden.

R0yk3, 2019-10-29:
Ik ben bezig om alles in te lezen in Node-red. Om er vervolgens een mooi dashboard van te maken. Soortgelijk aan pvoutput maar dan van mezelf. Als het er een beetje net uitziet kan ik de node-red blad wel delen. Zou in elke andere installatie moeten werken.

dennissiemensma, 2019-11-04:
Ik heb de afgelopen tijd nogmaals nagedacht over het oorspronkelijke verzoek en ben nog steeds van mening dat ik dit niet binnen het project kan doen. 

Dit project specialiseert zich primair in het uitlezen van de slimme meter en het weergeven van die gegevens. Daarnaast was het tweede doel om die gegevens makkelijk te exporteren (API, MQTT, CSV, PVOutput, Mindergas.nl), zodat gebruikers zelf hun eigen data kunnen gebruiken voor aggregaties, vergelijkingen en andere projecten. Daar is alle mogelijkheid voor en als er behoefte is aan een nieuwe ontsluiting/uitgaande datastroom, dan hoor ik dat uiteraard graag.

Het uitlezen, samenvoegen (met verbruik), plotten en synchroniseren van extra data uit omvormers of andere bronnen is een specialiteit an zich en daar is een heel project op zichzelf aan te wijden. Ik denk dat PVOuput daar een goed voorbeeld van is. Verder denk ik dat daar op termijn ook wel iets voor komt uit de hobby-wereld (als dat er nu al niet is). Ik lees hierboven en op Tweakers her en der ook al goede suggesties/initiatieven.

Tot slot bied ik liever een applicatie aan die een beperkt aantal dingen goed doet, dan eentje die van heel veel dingen een beetje doet. Dat is namelijk een zeer grote valkuil bij dit soort projecten. En er zijn natuurlijk nog een hoop verbeterpunten binnen dit project zelf.

Daarbij begrijp ik dat dit besluit voor een aantal gebruikers teleurstellend of jammer is, maar ik ben hier liever realistisch in.

closed 2019-11-04 by dennissiemensma

Erik050572, 2019-11-12:
Heren,
Misschien kunnen jullie hier wat mee:
Er is al een programma om de API van Solaredge uit te lezen en dit lokaal in een database op te slaan.
[https://github.com/debrigand/Solaredge-monitoring](url)

AlbertHakvoort, 2021-02-16:
Het lijkt mij inderdaad te veel werk om al die verschillende inverters te gaan ondersteunen, maar als er een (rest) api zou zijn waar je de data vanuit je S0 meter of SolarEdge of welk apparaat kan doorsturen is het toch een "simpele" oplossing.

dennissiemensma, 2021-02-16:
@AlbertHakvoort bedankt voor je suggestie. Wat er dan gebeurt is dat alle leveranciers hun omvormersoftware compatible moeten maken met zo'n API. Om nog maar te zwijgen over alle dataopslag, logica en aggregaties die daarmee gedaan moeten worden binnen DSMR-reader. 
Daardoor wordt het deels een kloon van wat bijvoorbeeld PVOutput al aanbiedt. Het lijkt me dan praktischer om PVOutput zich te laten specialiseren op het eigen gebied en DSMR-reader op de slimme meter te houden.