dsmrreader/dsmr-reader #595: 1-fase VS 3-fase Graph differences

jorkzijlstra, 2019-02-15:
LET OP: Wegens verminderde beschikbaarheid tot aan medio april 2019,
        kan het zijn dat de auteur minder snel of helemaal niet in de
        gelegenheid is om te reageren. Zie ook issue #576.
----

### Jouw omgeving
*(Indien van toepassing)*
* DSMR-reader versie *(rechtsbovenin applicatie zichtbaar)*: `v1.2.8`
* Hardware *(RaspberryPi 2, 3 of iets anders)*: `RaspberryPi 1?`
* DSMR-protocol versie van slimme meter *(v2, v4 of v5)*: `v4.2`

### Bug, wens of iets anders
* Bijvoorbeeld stappen hoe een probleem te reproduceren is...
* Of simpelweg een beschijving van een feature die je graag wenst te zien...
I just turned on the 3-fase view since I have that and the overview graphs show different data.

[image]

[image]

I was expecting the first graph to be the aggregated version of the last one. Mainly the L+ metrics are missing from the first graph. To me it look like the top graph only show the L+ when for all 3 fases there is a positive number. When there is 0 for one of the metric is disappears

Regards, Jork Zijlstra

dennissiemensma, 2019-02-15:
Bedankt voor je melding! Ik zal hier medio april inhoudelijk voor je naar kijken.

jorkzijlstra, 2019-02-15:
Hoi Dennis, geen probleem. 

Ik ga ergens  deze week switchen naar een rock pi 4 en weet niet of de data mee kan komen. 
Heb je nog meer data nodig? Iets van bijvoorbeeld de data calls?

Grt, Jork

dennissiemensma, 2019-02-15:
Ik denk dat ik daar op dit moment helaas weinig mee kan doen, maar ik laat dat tzt dan weten. Erg bedankt voor je aanbod!

jorkzijlstra, 2019-03-08:
Ik dacht dat heel misschien de raspberry pi 1 te langzaam zou zijn en metingen zou missen (niet genoeg cpu cycles hebben om alles op te vangen). Nu had ik deze te leen om deze software te testen en heb nu zelf een Rock Pi B 4b aangeschaft en daar een nvme disk aan gehangen. Performance wise is er dus nu geen probleem meer, maar het probleem lijkt nog steeds aanwezig.

Ik heb de data ook kunnen overzetten naar mijn nieuwe setup. Laat het maar weten of je nog wat van mij nodig hebt als je er naar gaat kijken. 

Hierbij nog wat screenshot:
[image]

[image]

dennissiemensma, 2019-03-08:
Bedankt voor je toevoeging. Ik zal het in april verder bekijken.

dennissiemensma, 2019-04-15:
Zou je nog eens kunnen proberen uit te leggen wat je exact mist of niet goed is? Ik lees er waarschijnlijk overheen, maar ik wil zeker weten dat we over hetzelfde praten. :]

jorkzijlstra, 2019-04-15:
Hoi Dennis,

Ten eerste maar even bevestigen dat ik de grafieken goed lees. Als ik beide grafieken namelijk lees op het zelfde tijdstip verwacht ik bij de bovenste grafiek hetzelfde totaal als in de onderste grafiek waarbij ik alle fasen samen op tel.

In de bovenstaande screenshots kan je zien dat bij de geaggregeerde versie 305W staat terwijl de som van de 3fase grafiek kom ik op 349W uit. Ook lever ik bij de 3 fase grafiek stroom terug terwijl in de andere grafiek ik niks terug lever.

Klopt mijn verwachting dus dat de geaggregeerde versie het totaal van de 3 fase versie zou moeten bevatten?

MvG, Jork

dennissiemensma, 2019-04-15:
Aah juist, ik snap nu wat je bedoelt en ik zie hetzelfde bij mij thuis (ook zonder teruglevering). Kan het zijn dat je het "groeperen van metingen" aan hebt staan (standaard = aan)? Zo ja, dan worden gemiddeldes berekend over alle metingen binnen de minuut en dat is dan onnauwkeurig. 

De data van de twee grafieken komen namelijk uit verschillende waardes in het telegram uit je meter, dus het is niet iets wat de applicatie checkt of narekent. Ik ga er vanuit dat de slimme meter exact sluitende waardes geeft. Je kunt:

A) Het groeperen van metingen uitzetten en kijken of het dan wel klopt.

B) Of eerst even een telegram uit je slimme meter bekijken en hier posten, dan check ik even of de optelling daarvan inderdaad klopt (ik ga uit van wel). Als je het hier post, haal wel even de (twee) serienummers uit het telegram. Deze staat bijna bovenaan en de tweede staat in dezelfde regel als je gasstand.

jorkzijlstra, 2019-04-16:
Hoi Dennis,

Bedankt voor je snelle reactie. Ik heb inmiddels de groeperen van metingen naar "per meting" gezet waardoor de grafiek idd gedetailleerde werd.

De grafiek is idd meer op elkaar gaan lijken, maar bevatten nog niet dezelfde waardes.

[image]

[image]

[image]
 
Een meting: 
´´´
1-3:0.2.8(50)
0-0:1.0.0(190416210449S)
0-0:96.1.1(.......)
1-0:1.8.1(000104.265*kWh)
1-0:1.8.2(000096.684*kWh)
1-0:2.8.1(000090.434*kWh)
1-0:2.8.2(000219.845*kWh)
0-0:96.14.0(0002)
1-0:1.7.0(00.127*kW)
1-0:2.7.0(00.000*kW)
0-0:96.7.21(00010)
0-0:96.7.9(00002)
1-0:99.97.0(1)(0-0:96.7.19)(000101010000W)(0000000506*s)
1-0:32.32.0(00007)
1-0:52.32.0(00010)
1-0:72.32.0(00010)
1-0:32.36.0(00000)
1-0:52.36.0(00000)
1-0:72.36.0(00001)
0-0:96.13.0()
1-0:32.7.0(233.1*V)
1-0:52.7.0(233.8*V)
1-0:72.7.0(232.9*V)
1-0:31.7.0(000*A)
1-0:51.7.0(000*A)
1-0:71.7.0(000*A)
1-0:21.7.0(00.003*kW)
1-0:41.7.0(00.000*kW)
1-0:61.7.0(00.124*kW)
1-0:22.7.0(00.000*kW)
1-0:42.7.0(00.000*kW)
1-0:62.7.0(00.000*kW)
0-1:24.1.0(003)
0-1:96.1.0(......)
0-1:24.2.1(190416210012S)(00139.128*m3)
!9A51
´´´

In welk bestand wordt deze meting geparsed? Ben zelf programmeur en kan python op zich wel lezen. Ik ben al wel even wezen spitten en vind het er logisch uit zien maar het parse gedeelte heb ik nog niet kunnen vinden.

Grt, Jork

dennissiemensma, 2019-04-17:
In principe zijn het deze regels in het telegram en dat lijkt wel goed:

```
1-0:1.7.0(00.127*kW)

1-0:2.7.0(00.000*kW)

...

1-0:21.7.0(00.003*kW)
1-0:41.7.0(00.000*kW)
1-0:61.7.0(00.124*kW)

1-0:22.7.0(00.000*kW)
1-0:42.7.0(00.000*kW)
1-0:62.7.0(00.000*kW)
```

Het parsen gebeurt in: https://github.com/dennissiemensma/dsmr-reader/blob/master/dsmr_datalogger/services.py#L170

Ik zie hier zelf trouwens wisselende gegevens. Soms komen de drie-fasen exact overeen en soms niet. Wellicht goed om nog wat extra telegrammen te bekijken om te zien of je van hetzelfde last hebt.

jorkzijlstra, 2019-04-17:
Hoi Dennis,

Het lijkt inderdaad de meter te zijn. Ik zie hier ook dat soms de drie-fasen exact overeen en soms niet.

[meter_log-1604.txt](https://github.com/dennissiemensma/dsmr-reader/files/3091144/meter_log-1604.txt)

Even een grote aanname: als programmeur kan ik mij voorstellen dat voor de totalen iets van windowing gedaan wordt, waardoor je eventueel dit soort gedrag kan krijgen.

In ieder geval bedankt voor het helpen nakijken

dennissiemensma, 2019-04-18:
Bedankt voor je terugkoppeling. Ik kan daar dus helaas niet veel aan doen. Heb je nog andere vragen m.b.t. dit issue?

jorkzijlstra, 2019-04-18:
Hoi Dennis, geen andere vragen meer en ik sluit dit issue dan ook.

Grt, Jork

closed 2019-04-18 by jorkzijlstra