dsmrreader/dsmr-reader #597: Missing dsmr-reader directory during configuration

Zilv3r, 2019-03-03:
I'm following the tutorial at https://dsmr-reader.readthedocs.io/nl/latest/installation/explained.html

System: Raspberry Pi with Hassbian

At the end of step 6 I'm encountering an error when logging out (ctrl-d) and logging in (sudo su - dsmr):

`-su: cd: /home/dsmr/dsmr-reader: No such file or directory`

The prompt says:

`(dsmrreader) dsmr@hassbian:~ $`

Zilv3r, 2019-03-03:
Found the problem: I did not log in as dsmr user before cloning the git step 5

closed 2019-03-03 by Zilv3r

dennissiemensma, 2019-03-03:
Bedankt voor je terugkoppeling, fijn dat je het zelf zo snel hebt gevonden!

Zilv3r, 2019-03-04:
Dank voor je reactie. En wat een fantastisch programma! :-)

Wat ik nog wel een beetje vreemd vond bij de installatie volgens het stappenplan 'Installatie: Toegelicht' was het commando:

`sudo sudo -u postgres psql -c "alter user dsmrreader with password 'dsmrreader';"`

Ik heb dit letterlijk ge-copy-paste, maar het was voor mij niet duidelijk of dat de bedoeling was (of dat ik tussen de quotes nog een wachtwoord moest invullen ofzo). Misschien dat anderen daar ook tegenaan lopen en dat het handig is dit nog even expliciet te melden in het stappenplan. Het draait hier in elk geval helemaal super volgens mij.

dennissiemensma, 2019-04-15:
Bedankt voor je toevoeging. Het wachtwoord staat ook in een van de configs, dus als je die zo hebt overgenomen is dat prima. Indien gewenst kan men het wachtwoord wijzigen, maar ik denk dat de user-case daarvoor vrij beperkt is.