dsmrreader/dsmr-reader #598: Data verzamelen via netwerk (ser2net)

Gerben321, 2019-03-03:
Beste ontwikkelaars,

Ik vind DSMR-Reader er wel goed uit zien, een platform met potentie. Nu wil ik het graag proberen, maar ik heb een wat andere use case. Ik heb namelij keen Raspberry 1 waarmee ik data verzamel, verder niks. Ik gebruik ser2net om het uit te lezen in Home Assistant, maar aangezien die GUI niet echt geschikt is hiervoor zou ik graag dit platform willen testen.

Dus mijn vraag is; kan ik dit ook installeren op een externe server (Liefste via Docker natuurlijk), en dan het IP en de port opgeven van mijn Raspberry en het dan alsnog uitlezen? Dat zou geweldig zijn!

dennissiemensma, 2019-03-03:
Bedankt voor je vraag. Het is sowieso mogelijk, want zo gebruik ik het zelf ook (alleen dan zonder docker).De applicatie kan geen data uit je raspberry 1 trekken, maar je raspberry 1 kan wel data doorsturen naar de applicatie (dus net andersom). Op de raspberry 1 installeer je dan het voorbeeldscript uit dit hoofdstuk: https://dsmr-reader.readthedocs.io/nl/latest/api.html#api-v1-remote-datalogger .Vervolgens installeer je de applicatie op je externe server. Dit zou ook via docker moeten kunnen, al geef ik daar zelf geen ondersteuning voor. Zie: https://dsmr-reader.readthedocs.io/nl/latest/installation/docker.htmlVervolgens stel je een API key in in de applicatie en die gebruik je dan ook in het script op je raspberry 1.Hopelijk helpt dit je wat op weg.

Gerben321, 2019-03-04:
Super! Dat heb ik net over het hoofd gezien. Zal er vandaag even induiken!

dennissiemensma, 2019-04-15:
@Gerben321 heb je hier nog een terugkoppeling voor inmiddels?

Gerben321, 2019-04-15:
@dennissiemensma Sorry helemaal vergeten! Heb het inmiddeld werkend gekregen ja. Wel wat 'issues' wat betreft het persisten van bepaalde settings in Docker, maar dat is verder geen issue voor jou. Moest wat code in de docker aanpassen, dus moet hem niet updaten of clearen want dan werkt het niet meer, haha.

Maar ziet er super uit DSMR reader!

closed 2019-04-15 by Gerben321

dennissiemensma, 2019-04-15:
Oke fijn, bedankt voor je snelle reactie.