dsmrreader/dsmr-reader #600: CPU gebruik relatief hoog

brainfrz82, 2019-03-12:
LET OP: Wegens verminderde beschikbaarheid tot aan medio april 2019,
        kan het zijn dat de auteur minder snel of helemaal niet in de
        gelegenheid is om te reageren. Zie ook issue #576.
----

### Jouw omgeving
*(Indien van toepassing)*
* DSMR-reader versie *(rechtsbovenin applicatie zichtbaar)*: `v1.28`
* Hardware *(RaspberryPi 2, 3 of iets anders)*: `RaspberryPi 3b`
* DSMR-protocol versie van slimme meter *(v2, v4 of v5)*: `v4`

### Bug, wens of iets anders
Allereerst dank voor deze mooie applicatie, eerder gebruikte ik domoticz om mijn slimme meter uit te lezen, en het gebrek aan overzicht en de vele in mijn geval ongebruikte features deden mij altijd wat zeer aan de ogen.

Nu ik een paar dagen DSMR-Reader heb draaien, valt mij op dat deze op mijn raspberry pi altijd zo'n 10% CPU pakt. Ik vroeg mij af of dit bij anderen ook het geval is. De 10% an sich is niet merkbaar, de pi staat toch vrijwel enkel te idlen, maar het is jammer dat hij daardoor niet terugschakelt naar 600mhz om energie te besparen, hij blijft dus altijd op 1200mhz draaien.

Domoticz zag ik eigenlijk nooit terug dat hij uberhaubt CPU gebruikte. Nu is het extra energieverbruik van de Pi te verwaarlozen, zeker als ik dat wegzet tegen de besparing die ik kan hebben nu ik mooi inzicht heb :) maar ik vraag mij gewoon af of dit normaal is, en of daar wat mee te doen is :)

jeroenpeters1986, 2019-03-12:
Ik kan mijn usage wel met je delen, maar daar heb je niet zo heel veel aan.
`top - 13:43:28 up 65 days,  7:54,  1 user,  load average: 0.32, 0.41, 0.41`
Mijn Raspberry Pi heeft nog een aantal andere taken en is derhalve wel meer bezig. Ik heb geen noemenswaardige verandering gezien toen ik hem op deze Pi3 installeerde. 

Je Pi verwerkt een hoop berichten per dag, dus ik zou zeggen dat dit volkomen normaal is.

brainfrz82, 2019-03-12:
@jeroenpeters1986 dankje, mijn load average ziet er ongeveer hetzelfde uit, en die van mij doet ook meer, maar met domiticz was het doorgaans tussen de 0.01 en 0.05. Ik heb op basis van https://github.com/dennissiemensma/dsmr-reader/issues/413 even gekeken of het wijzigen daarvan effect had, en dat had het, met de settings op 60 was er vrijwel geen CPU gebruik meer, maar dan haalt hij maar eens per minuut de gegevens op. Het semi-real-time inzicht is voor mij de meest belangrijke eigenschap, helaas doet mijn slimme meter geen DSMR 5 :)
Ik ga dr maar even vanuit dat ik geen configuratiefouten heb gemaakt en dat dit gedrag normaal is :)

jeroenpeters1986, 2019-03-12:
Oh dat scheelt weer, dat is een mooie wijziging inderdaad!

dennissiemensma, 2019-03-12:
Bedankt voor je bericht! Je zou kunnen spelen met "nice" in de Supervisor config. Daarmee kun je de prio van het process op de CPU aangeven dacht ik. Hoger getal is lagere prioriteit (1 tot max 19). Standard staat die volgens mij op 5, 10 en nog iets. Je zou die allemaal op 19 kunnen zetten, al weet ik niet zeker of het OS daarmee ook perse minder CPU verbruikt. Je kunt het aanpassen in /etc/supervisor/conf.d/ en dan herladen met "supervisorctl reread" en "supervisorctl update". Geen garantie dus dat het helpt. Wellicht kun je ook zoiets gebruiken voor de postgresql database, maar ik heb dat zelf nog niet geprobeerd.

brainfrz82, 2019-03-12:
Ik zal vanavond eens kijken of de nice instellingen wat veranderen, ik vermoed het niet aangezien het taken zijn die toch moeten gebeuren, en nice is enkel een indicatie van prioriteit. Maar misschien voorkomt het dat hij de CPU hoger gaat klokken.

brainfrz82, 2019-03-27:
Even een updateje, de 2 "schuldige" processen (datalogger en backend) draaien al onder een hoger nice niveau, dus dat helpt helaas niet. Ik heb nog wel even verder gespeeld met de 2 settings "DSMRREADER_BACKEND_SLEEP" en "DSMRREADER_DATALOGGER_SLEEP".

De datalogger die draait al standaard elke 10 seconden, en alles wat ik daar in verhoog is direct nadelig voor mijn uitlezing, maar de Backend lijkt standaard elke seconde te draaien. Door deze te verhogen naar 10 seconden merk ik in het gebruik van de app geen enkel verschil, maar de load average op mij pi3 gaat van +-0.35 naar +-0.10, met als voordeel dat hij nu vrijwel nooit meer naar 1200mhz gaat.
Kunnen daar nadelen aan hangen die ik wellicht niet direct zie?

dennissiemensma, 2019-03-27:
De backend is puur voor alle verwerking, het groeperen van metingen, uploaden naar externe systemen, etc. In principe zou je die prima een hogere sleep kunnen geven. Het kan wel zijn dat sommige gegevens heel iets langer duren voordat ze zichtbaar zijn. Zou verder goed moeten gaan.

brainfrz82, 2019-03-27:
Ah, top, dan laat ik het zo staan, in het ergste geval duurt de "live" weergave nu tot 9 seconden langer voordat het getoond wordt, dat is prima bruikbaar :)

dennissiemensma, 2019-04-15:
Prima! Dan laat ik dit issue nu voor wat het is. Ik heb ook geen andere oplossing voor je namelijk en als dit voor jou werkt, is dat mooi.

closed 2019-04-15 by dennissiemensma