dsmrreader/dsmr-reader #601: Clear cache

jorkzijlstra, 2019-03-12:
LET OP: Wegens verminderde beschikbaarheid tot aan medio april 2019,
        kan het zijn dat de auteur minder snel of helemaal niet in de
        gelegenheid is om te reageren. Zie ook issue #576.
----

### Jouw omgeving
* DSMR-reader versie *(rechtsbovenin applicatie zichtbaar)*: `v1.28`
* Hardware *(RaspberryPi 2, 3 of iets anders)*: `Rock pi 4`
* DSMR-protocol versie van slimme meter *(v2, v4 of v5)*: `v5`

### Bug, wens of iets anders
Er is net bij mij de elektra en gasmeter vervangen (zal de uur en dag stats nog moeten resetten vanwege negatieve getallen). Hierdoor wordt u dsmr V5 gebruikt i.p.v. V4.2. Wat is begrijp is dat de gas metingen er automagisch bij moeten komen te staan. Echter dit is tot zover niet het geval. 

In de codebase zag ik dat er op `get_capabilities` een cache ingesteld staat. Ik vond ook een `dsmr_stats_clear_cache` dat naar mijn idee deze cache zou moeten elgen zodat er de nieuwe settings worden ingelezen. Het script voor deze clear cache lijkt alleen niet meer te staan. 

```
bash-4.4# ./manage.py dsmr_stats_clear_cache
Unknown command: 'dsmr_stats_clear_cache'
```

Is er een andere manier om de `get_capabilities` in te zien / cache te clearen?

dennissiemensma, 2019-03-12:
Bedankt voor je melding. De cache staat standaard ingesteld op slechts één minuut (om queries te verminderen) , dus er is geen reden om die te legen. Het command dat je gevonden hebt is hernoemd, maar lost je problem ook niet op. Wellicht kun je even wachten tot na 01:15, om te kijken of de statistieken dan wel gegenereerd worden.

jorkzijlstra, 2019-03-13:
Hoi Dennis,

#601 bestond vanmiddag ineens niet meer en heb #602 daardoor aangemaakt.

De statistieken zijn er nog niet en zal met #602 verder gaan.

Bedankt voor de snelle reactie.

closed 2019-03-13 by jorkzijlstra