dsmrreader/dsmr-reader #602: Clear cache / get capabilities

jorkzijlstra, 2019-03-12:
LET OP: Wegens verminderde beschikbaarheid tot aan medio april 2019,
        kan het zijn dat de auteur minder snel of helemaal niet in de
        gelegenheid is om te reageren. Zie ook issue #576.
----

### Jouw omgeving
* DSMR-reader versie *(rechtsbovenin applicatie zichtbaar)*: `v1.28`
* Hardware *(RaspberryPi 2, 3 of iets anders)*: `Rock Pi 4`
* DSMR-protocol versie van slimme meter *(v2, v4 of v5)*: `v5`

### Bug, wens of iets anders
Ik heb net een nieuwe gasmeter en electra meter gekregen en ben van dsm V4.2 naar V5 gegaan. Ik had verwacht dat er inmiddels wel gas statistieken terug waren te vinden maar tot op heden nog niks (pas 3 uur geleden de meter gekregen).

Nu probeer ik te debuggen waar het probleem zit. Er komt namelijk geen data in `dsmr_consumption_gasconsumption` tabel te staan.

Nu was ik in de code aan het kijken en kwam daar de `get_capabilities` methode tegen. Ik dacht ook te zien dat hier een cache voor staat. Uitvoeren van `./manage.py dsmr_frontend_clear_cache` verhelpt dit probleem niet.

Ik probeer dus nu te kijken of mijn nieuwe meter uberaupt dus gas metingen aanleverd.

Enig idee hier ik dit kan debuggen / checken?

dennissiemensma, 2019-03-12:
Ik denk dat je hiervoor het makkelijkste zelf een meting kunt uitlezen: https://dsmr-reader.readthedocs.io/nl/latest/installation/explained.html#your-first-reading-optional . Hiervoor zul je errst kort de datalogger moeten stoppen (als root) met "supervisorctl stop dsmr_datalogger". Na het uitlezen kun je hem weer starten met "supervisorctl start dsmr_datalogger". In de meting moet je op zoek naar een regel bijna onderaan waar "m3" in staat. Dat moet je gasstand bevatten. Als dat goed is, kun je kijken in de database, maar daar kom ik later nog op terug als het nodig is.

jorkzijlstra, 2019-03-13:
Hoi Dennis,

Bedankt voor de tip. Mijn meter lijkt dus helemaal geen gas standen aan te leveren. Ik zal morgen een kijken of de meterstanden bij mijn leverancier aankomen en eventueel Stenden bellen als dat niet zo het geval is.

[meter_log.txt](https://github.com/dennissiemensma/dsmr-reader/files/2961851/meter_log.txt)

jorkzijlstra, 2019-03-16:
Hoi Dennis,

Ik heb even gebeld met Stedin en zij gaven aan dat het automagisch binnen 30 dagen zou gaan werken. En zowaar heb ik inmiddels gasmetingen die geregistreerd worden.

Voor nu hoef ik niet verder te debuggen. Hartelijk dank voor je snelle hulp.

closed 2019-03-16 by jorkzijlstra