dsmrreader/dsmr-reader #603: hoe gaat dsmr-reader om met server downtime?

RezzZ, 2019-03-13:
Ik kan het in de [FAQ](https://dsmr-reader.readthedocs.io/en/latest/) niet zo 123 vinden.

Hoe gaat dsmr-reader om met server downtime ivm het updaten van je system of eventuele crashes? Heb ik dan een gat in mijn totale verbruik of wordt dat bij de eerst volgende ontvangen telegram netjes gecorrigeerd? Ik denk het laatste maar hoor het graag!

Misschien kan deze vraag in de FAQ toegevoegd worden?

dennissiemensma, 2019-03-13:
In principe wordt het totale verbruik hersteld en heb je dan alleen een gat of verschuiving in de detailweergave. Als de downtime over de dagwisseling heen schuift, dan kan het zijn dat het verbruik op de volgende dag wordt gerekend. In de meeste gevallen is dat niet erg denk ik.

RezzZ, 2019-03-13:
duidelijk! bedankt voor de toelichting

closed 2019-03-13 by RezzZ