dsmrreader/dsmr-reader #604: 504 Time-out op status pagina

Freez, 2019-03-15:
Sinds kort heb ik problemen met DSMR, de site reageert heel traag en op de status pagina krijg ik een 504. Rond Kerst een crash gehad vanwege een volle SD kaart, ik heb toen de database op een ander systeem terug gezet en met vacuum opgeschoond. Dit ging 2 maanden goed maar sinds kort heb ik dus performance problemen. Nu is mijn aansluiting wel geupgrade van 1 x 16A naar 3 x 16A, heeft dit misschien meer telegrammen als gevolg?

Via TOP zie ik best  wat CPU gebruik voor python en postgres
PID USER      PR  NI    VIRT    RES    SHR S  %CPU %MEM     TIME+ COMMAND
434 dsmr      30  10   58372  44768   8464 S  23.3  4.7 137:55.34 python3
433 dsmr      25   5   45868  13448   7040 S  14.1  1.4 157:13.00 python3
21585 postgres  20   0  197660  62560  61024 D  10.2  6.6   0:01.29 postgres
21611 postgres  20   0  197128  18412  17100 S   8.5  1.9   0:00.26 postgres
66 root      20   0       0      0      0 S   3.6  0.0   4:23.38 mmcqd/0
21612 postgres  20   0  197088  15916  14796 D   3.3  1.7   0:00.10 postgres

* DSMR-reader versie: 1.28
* Hardware:  Raspberry Pi 2 Model B Rev 1.1
* DSMR-protocol: v5

dennissiemensma, 2019-03-15:
De raspberry pi 2 is amper snel genoeg voor alle processen, helemaal in combinatie met dsmr v5. Heb je al geprobeerd om een sleep te gebruiken voor de datalogger? Zie https://dsmr-reader.readthedocs.io/nl/latest/settings.html#dsmrreader-datalogger-sleep

Freez, 2019-03-16:
Voorheen ging het goed maar sinds mijn verhuizing is het prut met de performance. Ik heb de setting aangepast naar 1.5, even kijken wat het doet en anders een nieuwe pi.

Bedankt voor je tip!

closed 2019-03-16 by Freez

dennissiemensma, 2019-03-16:
Je zou ook nog kunnen spelen met de backend sleep. En de datalogger sleep op 5 of 10 secs zetten, mits je de nauwkeurigheid niet nodig hebt. Voor het overzicht van je uur- of dagverbruik is 1 telegram per 10 seconde namelijk meer dan zat. Ik weet dat er tenminste 1 andere gebruiker is die ook een pi 2 gebruikt met wat aangepaste instellingen.

dennissiemensma, 2019-03-16:
En als je geen MQTT gebruikt, kun je dat proces ook nog permanent uitzetten in Supervisor.

Freez, 2019-03-16:
Ik ga eens spelen met die instellingen, ik gebruik geen MQTT maar volgens mij draait dat proces bij mij ook niet.

pi@raspberrypi:/etc/supervisor/conf.d $ sudo supervisorctl status
dsmr_backend                     RUNNING   pid 9616, uptime 0:45:40
dsmr_datalogger                  RUNNING   pid 9615, uptime 0:45:40
dsmr_webinterface                RUNNING   pid 9614, uptime 0:45:40