dsmrreader/dsmr-reader #605: Geen upload naar PVOutput

BramJ85, 2019-03-20:
### Jouw omgeving
* DSMR-reader versie: `v1.28`
* Hardware: `RaspberryPi 3`
* DSMR-protocol versie van slimme meter: `v5`

PVOutput is enabled en ingesteld met API key en systemID, upload staat enabled maar er komt niets tevoorschijn op PVoutput. Het Status scherm geeft bij "Meest recente synchronisatie met PVOutput:" geen datum/tijd. 
Bij de "PVOutput: "Add Status" configuratie" in Django staat dat de volgende upload 
20 maart 2019 00:10 geweest had moeten zijn, maar toen is niets gebeurd.

In de dsmr_backend.log komen de volgende regels voorbij:

   File "/home/dsmr/dsmr-reader/dsmr_pvoutput/apps.py", line 20, in _on_backend_called_signal
       dsmr_pvoutput.services.export()

   File "/home/dsmr/dsmr-reader/dsmr_pvoutput/services.py", line 124, in export
       data=data

Vervolgens heb ik in de settings.py de logging op debug gezet en toen zag ik in de dsmr_backend.log dergelijke regels voorbij komen:

[2019-03-20 20:28:09,989] DEBUG     - PVOutput | Uploading data: {'v4': 248, 'v3': 3177, 'd': '20190320', 'n': 1, 't': '20:28'}

Er wordt dus wel daadwerkelijk een poging gedaan iets te sturen...

dennissiemensma, 2019-03-21:
De logregels die je opgeeft zijn onderdeel van een zogenaamde exception of foutmelding. Vaak staat onderaan dan de werkelijke foutmelding. Kun je eens kijken of je die ziet? En wellicht goed om je API credentials van PVouput.org te dubbelchecken.

BramJ85, 2019-03-21:
Dit is het hele stuk uit de log bij de exception:
```
[2019-03-21 18:41:56,835] CRITICAL Uncaught exception:   File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/dispatch/dispatcher.py", line 210, in send_robust
    response = receiver(signal=self, sender=sender, **named)

  File "/home/dsmr/dsmr-reader/dsmr_pvoutput/apps.py", line 20, in _on_backend_called_signal
    dsmr_pvoutput.services.export()

  File "/home/dsmr/dsmr-reader/dsmr_pvoutput/services.py", line 124, in export
    data=data

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/requests/api.py", line 116, in post
    return request('post', url, data=data, json=json, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/requests/api.py", line 60, in request
    return session.request(method=method, url=url, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/requests/sessions.py", line 533, in request
    resp = self.send(prep, **send_kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/requests/sessions.py", line 646, in send
    r = adapter.send(request, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/requests/adapters.py", line 516, in send
    raise ConnectionError(e, request=request)

[2019-03-21 18:41:56,835] DEBUG    Sleeping 1 sec(s)
[2019-03-21 18:41:57,844] INFO     Calling backend services
[2019-03-21 18:41:59,115] DEBUG     - PVOutput | Uploading data: {'v4': 473, 'v3': 2894, 'd': '20190321', 'n': 1, 't': '18:41'}
```

Heb de API key en systemID (cijfers, niet de naam) nog eens gecontroleerd, maar die kloppen toch echt. 
Vanaf een andere Pi stuur ik de zonnepaneeldata met dezelde key/systemID zonder problemen

BramJ85, 2019-03-21:
MinderGas ook geprobeerd, maar deze geeft ook een exception:

```
[2019-03-21 18:59:44,115] CRITICAL Uncaught exception:   File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/dispatch/dispatcher.py", line 210, in send_robust
    response = receiver(signal=self, sender=sender, **named)

  File "/home/dsmr/dsmr-reader/dsmr_mindergas/apps.py", line 20, in _on_backend_called_signal
    dsmr_mindergas.services.export()

  File "/home/dsmr/dsmr-reader/dsmr_mindergas/services.py", line 69, in export
    'reading': str(last_gas_reading.delivered)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/requests/api.py", line 116, in post
    return request('post', url, data=data, json=json, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/requests/api.py", line 60, in request
    return session.request(method=method, url=url, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/requests/sessions.py", line 533, in request
    resp = self.send(prep, **send_kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/requests/sessions.py", line 646, in send
    r = adapter.send(request, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/requests/adapters.py", line 516, in send
    raise ConnectionError(e, request=request)
```

BramJ85, 2019-03-21:
ok *schaam*
 
om een of andere reden wil mijn Pi geen DNS resolven via mijn modem/router's IP
paar public DNS'en toegevoegd en nu werkt het wel

Sorry...

dennissiemensma, 2019-03-21:
Fijn dat je het gevonden hebt!

dennissiemensma, 2019-04-15:
Ik sluit deze nu en je hoeft je niet te schamen hoor. Want DSMR-reader heeft momenteel niet hele makkelijke logging om uit te lezen dus dat helpt jou ook niet echt mee ;-)

closed 2019-04-15 by dennissiemensma