dsmrreader/dsmr-reader #606: authenticatie API browser

brammeleman, 2019-03-20:
```
dsmrr-reader versie: 1.28
hw: pi3+
os: raspbian
```

Het lukt me niet om de interact functionaliteit van de API documentatie te gebruiken. Wat voor instellingen ik ook gebruik bij `authentication` links-onderaan, het resultaat is altijd:
```
{"detail":"Invalid auth key"}
```
Ik heb een API token aangemaakt, deze doet het via de command line en ook in arduino code, maar niet in de API browser.
Bij authentication, kies ik `token`.  Voor Scheme `X-AUTHKEY` en natuurlijk m'n Token. Maar geen succes.
Ook alle daar gesuggereerde authentication schemes geprobeerd: Bearer, Token, JWT
Daarnaast ook `basic` authentication geprobeerd met het admin login. Dit geeft hetzelfde resultaat.

In dsmr heb ik debugging aangezet, maar dat levert geen extra logregels op, dus dat helpt me ook niet verder.

Wat zie ik over het hoofd. Heeft iemand een idee?

dennissiemensma, 2019-03-21:
Het is een tijdje geleden dat ik die docs werkend heb gekregen. Het kan zijn dat het simpelweg niet werkt, want ik had ontzettend moeite om die docs destijds überhaupt werkend te krijgen. Wellicht ondersteunt het alleen de custom header authenticatie van dit project zelf. Hier zal ik later naar moeten kijken.

dennissiemensma, 2019-04-18:
Ik denk dat het komt omdat dit project custom authenticatie gebruikt. De automatische docs, van het onderliggende API framework, herkennen dat helaas niet en kunnen er niet mee overweg.

Daarmee heb ik hier niet zo snel een oplossing voor. Ik kan me vaag herinneren dat ik er destijds wat tijd in had gestoken om het werkend te krijgen, maar dat het niet lukt.

Als alternatief zou ik eventueel op termijn nog meerdere authenticatiemethodes kunnen toestaan. En dan specifiek de standaard aanbevolen methodes.

dennissiemensma, 2019-04-18:
Hmm ik heb nu meerdere geprobeerd: basic, token en session. Maar ze werken allemaal niet. Ook niet als ik ze inschakel in mijn ontwikkelomgeving.

dennissiemensma, 2019-05-04:
Ik heb inmiddels authenticatie via `Token` werkend. Dit werkt dan ook in de browser. Ik zal dit meenemen in de volgende release.

dennissiemensma, 2019-05-04:
Gebruik wordt dan:
```
Authentication: Token
Scheme: Token
Token: <key>
```

brammeleman, 2019-05-06:
Top! Dank je!
/Bram

dennissiemensma, 2019-05-13:
Gemerged naar `development` branch.

closed 2019-05-13 by dennissiemensma

dennissiemensma, 2019-05-20:
Gereleased in [v2.1.0](https://github.com/dennissiemensma/dsmr-reader/releases/tag/v2.1.0)