dsmrreader/dsmr-reader #607: DSMR cconfiguratie

sjaakveenendaal, 2019-03-24:
LET OP: Wegens verminderde beschikbaarheid tot aan medio april 2019,
        kan het zijn dat de auteur minder snel of helemaal niet in de
        gelegenheid is om te reageren. Zie ook issue #576.
----

### Jouw omgeving
*(Indien van toepassing)*
* DSMR-reader versie *(rechtsbovenin applicatie zichtbaar)*: `v1.X.Y`
* Hardware *(RaspberryPi 2, 3 of iets anders)*: `RaspberryPi X...`
* DSMR-protocol versie van slimme meter *(v2, v4 of v5)*: `v...`

### Bug, wens of iets anders
* Bijvoorbeeld stappen hoe een probleem te reproduceren is...
* Of simpelweg een beschijving van een feature die je graag wenst te zien...

sjaakveenendaal, 2019-03-24:
Vandaag DSMR reader geïnstalleerd al heel iets voor een leek vindt ik zelf ;-) maar nu wil ik inloggen in de configuratie en heb geen idee wat de gebruikersnaam en het wachtwoord is?

is er een standaard wachtwoord?

dennissiemensma, 2019-03-24:
Wellicht heb je de stap overgeslagen. Je kunt de gebruiker aanmaken of resetten met de instructies onderaan hoofdstuk 8. https://dsmr-reader.readthedocs.io/nl/latest/installation/explained.html#bootstrapping

sjaakveenendaal, 2019-03-25:
bedankt voor de info alles werkt nu naar behoren top..

dennissiemensma, 2019-04-15:
Fijn om te horen!

closed 2019-04-15 by dennissiemensma