dsmrreader/dsmr-reader #608: krijg Dropbox  niet werkend

zinker, 2019-03-27:
Hallo, ik heb de applicatie dsmrreader werkend alleen is het nog niet 1 keer gelukt om dit werkend te krijgen met dropbox.

Ik heb naar mijn mening alles gedaan wat op de site staat vermeld maar geen data in mijn dropbox map te vinden.

vraag 1: werkt deze mogelijkheid normaal gesproken nog ?
vraag 2:  hoe kan ik testen of er een verbinding is, of hoe doe ik een test upload ?

gr. ad

dennissiemensma, 2019-03-28:
Ik denk dat je het beste kunt beginnen met de logfiles. Kijk of daar iets in staat. Zie https://dsmr-reader.readthedocs.io/nl/latest/troubleshooting.html

jeroenpeters1986, 2019-03-28:
**Vraag1**
Het werkt normaalgesproken nog. Bij mij is de backup van vannacht om 3:39 gelukt, ik heb hem zojuist bekeken.

**Vraag2**
Wat Dennis aangaf is inderdaad het handigst, check eerst de logs. 
Komt daar niets uit, wees er dan zeker van dat je alles van https://dsmr-reader.readthedocs.io/nl/latest/admin/backup_dropbox.html hebt gevolgd. 
Staat bijvoorbeeld wel het access token erin en niet per ongeluk de secret?

Komt er wel iets uit de logs, kijk dan of de melding duidelijk genoeg is om je verder te helpen, of post hem hier :-)

dennissiemensma, 2019-04-15:
@zinker ben je hier inmiddels uit gekomen?

closed 2019-05-04 by dennissiemensma