dsmrreader/dsmr-reader #609: Volgorde stroomfases niet consequent

RichieB2B, 2019-04-02:
### Mijn omgeving
* DSMR-reader versie *(rechtsbovenin applicatie zichtbaar)*: `v1.28`
* Hardware *(RaspberryPi 2, 3 of iets anders)*: `RaspberryPi 3B`
* DSMR-protocol versie van slimme meter *(v2, v4 of v5)*: `v5`

### Minor annoyance
In de 3-fasen grafiek is de volgorde van boven naar beneden: L3, L2, L1. Als ik hover over deze grafiek verschijnt er een legenda waarin van boven naar beneden staat: L1, L2, L3. Dit is niet erg intuïtief. 

[image]

dennissiemensma, 2019-04-03:
Bedankt voor je melding. Het zou kunnen dat het een volgorde dingetje is dat makkelijk op te lossen is. Ik kijk er later deze maand naar, wanneer ik weer tijd heb voor het project.

dennissiemensma, 2019-04-18:
Het lijkt erop dat ik het niet kan aanpassen.

Strikt gezien kan ik de volgorde van de labels wijzigen, maar wat er dan gebeurt is dat de kleuren door elkaar heen gaan. In jouw geval wordt L1 dan opeens blauw.

En als ik de volgorde van de labels én de data verander, dan komt het uiteindelijk weer op hetzelfde uit (schuift mee).

Ik had gehoopt dat de labels gebonden waren aan de data, maar ze zijn helaas gewoon losse opties...

closed 2019-04-18 by dennissiemensma

RichieB2B, 2019-04-18:
Ok, dus de pop-up kan niet (makkelijk) worden aangepast. Maar kan de volgende van de fases in de grafiek niet worden omgedraaid?

dennissiemensma, 2019-04-19:
Als ik de data omdraai, krijg ik hetzelfde probleem, maar dan omgekeerd. Zie de voorbeelden:
[image]

[image]

Ik had zelf ook niet verwacht dat dit zo'n issue zou zijn, het zou triviaal moeten zijn.

RichieB2B, 2019-04-19:
Suf zeg. Dus het is een bug in de library die wordt gebruikt? Welke is dat, dan meld ik het daar.

dennissiemensma, 2019-04-19:
Ik weet niet of het een bug is eigenlijk. Of meer een tekortkoming. Het heet e-charts.