dsmrreader/dsmr-reader #611: MQTT publiceert vaker dan nodig

RezzZ, 2019-04-10:
### Jouw omgeving
* DSMR-reader versie *(rechtsbovenin applicatie zichtbaar)*: `v1.28`
* Hardware *(RaspberryPi 2, 3 of iets anders)*: `RaspberryPi 3B+`
* DSMR-protocol versie van slimme meter *(v2, v4 of v5)*: `v5`

### Bug, wens of iets anders
MQTT publiceert veel frequenter dan noodzakelijk. Ik vermoed dat bij elk ontvangen telegram DSMR reader alles publiceert wat staat ingeschakeld. Ook dagstatistieken, ookal kunnen deze urenlang hetzelfde (bv bij gasverbruik). Is dit niet een beetje overbodig? of is de overhead voor de broker zo laag dat de effort om code te schrijven om alleen wijzigingen te publiceren hier niet tegen op weegt?

dennissiemensma, 2019-04-10:
Bedankt voor je aanvulling. Als ik het me goed herinner liep iemand recent tegen hetzelfde aan. Ik zal kijken of ik dit eenvoudig kan realiseren, wanneer ik weer aan de slag ga.

trizz, 2019-04-12:
Wellicht is het dan wel een idee om de [`retain`](http://www.steves-internet-guide.com/mqtt-retained-messages-example/) flag te zetten om ervoor te zorgen dat systemen die de MQTT data gebruiken niet hoeven wachten op een statuswijziging als deze herstart worden.

dennissiemensma, 2019-09-30:
Pak ik op via #723. Ik zal kijken of dat werkbaar is.

closed 2019-09-30 by dennissiemensma