dsmrreader/dsmr-reader #612: Geen metingen meer.

pascalbuijs, 2019-04-10:
Dag Dennis,

Inmiddels 3 installaties gedaan welke super werken maar helaas bij mijn vader kwam er na een paar maanden spontaan geen data meer binnen. De logging aangezet maar daar ook niets zichtbaar. Wanneer ik met putty de usb interface rechtstreeks uitlees zie ik netjes alle telegrammen voorbij komen. De USB interface word ook wel geconnect op /dev/ttyUSB0. Er is ook voldoende ruimte beschikbaar. Inmiddels heb ik het met een volledig schone installatie getest maar ik krijg het niet meer aan de gang. Kun je mij een paar handvaten geven om verder te zoeken?

dennissiemensma, 2019-04-10:
Bedankt voor je vraag. Zie je ook helemaal niets voorbijkomen in de dsmr_datalogger zodra je de logging aan zet? Want ik zou dan tenminste een foutmelding of iets verwachten.

pascalbuijs, 2019-04-11:
Nee, helemaal niets. Zojuist nog even opnieuw de het hele proces doorlopen om de logging aan te zetten maar het bestand dsmrreader.log blijft leeg. ./post-deploy is succesvol uitgevoerd. Alle 4 processen zijn gestart.

dennissiemensma, 2019-04-11:
Er zijn meerdere logbestanden. Zodra je logging aanzet, zou je meer informatie in Supervisor of de logs van Supervisor moeten zien. Heb je die ook gecheckt? Die zijn namelijk belangrijker dan de andere log.

dennissiemensma, 2019-05-04:
Ik hoor graag als je hier later alsnog hulp bij nodig hebt.

closed 2019-05-04 by dennissiemensma