dsmrreader/dsmr-reader #614: Ideeen voor het verbeteren van DSMR-Reader

remb0, 2019-04-12:
DSMR-reader is fantastisch!  alleen merk ik dat het instelling niet geheel zonder terminal mogelijk is.  Ik heb een aantal ideeen waarmee het programma gebruiksvriendelijker zou kunnen worden.

- Google drive integratie (is eenvoudiger in te regelen dan dropbox).
- loggen details/locaties en uitlezen via de website mogelijk maken, waardoor je niet de terminal in hoeft.
- Usb port keuze veranderen in een listbox die enkel devices laat zien die ook gemount zijn
- update checker
- https en port opties vanuit GUI.

het zijn kleine punten waar nu ook omheen valt te werken, maargoed mocht je ze leuk vinden kan het dsm-reader wat al super fantastisch is nog een klein beetje beter maken.

dennissiemensma, 2019-04-12:
Bedankt voor je verbeterpunten! Een aantal heb ik al naar gekeken maar kon ik niet zo 123 werkend krijgen. Mogelijk doe ik daar opnieuw een poging toe. Verder zijn sommige zaken niet mogelijk, omdat het root-toegang nodig heeft. Het veranderen van de poort en https heeft wijzigingen in Nginx config nodig namelijk en dsmr-reader heeft bewust geen onnodige permissies. Op de andere punten kom ik terug wanneer ik weer aan de slag ga.

dennissiemensma, 2019-10-02:
Voor google drive komt er in v3.x losstaande integratie. Voor automatische update-meldingen heb ik #727 gemaakt. De andere opties kunnen helaas niet of zijn wat complex om sluitend te krijgen.

closed 2019-10-02 by dennissiemensma