dsmrreader/dsmr-reader #615: Na middernacht is verbruik van gisteren niet meer zichtbaar

RichieB2B, 2019-04-13:
### Mijn omgeving
* DSMR-reader versie *(rechtsbovenin applicatie zichtbaar)*: `v1.28`
* Hardware *(RaspberryPi 2, 3 of iets anders)*: `RaspberryPi 3B`
* DSMR-protocol versie van slimme meter *(v2, v4 of v5)*: `v5`

### Minor annoyance

De verbruikscijfers bovenaan het dashboard laten het huidige verbruik zien sinds middernacht. Als er net een nieuwe dag is begonnen (op middernacht) springen de getallen op 0, maar in het archief is de vorige dag (die net is afgelopen) nog niet zichtbaar. Dat is jammer, want ik wil vaak juist op dat moment zien hoeveel ik die dag heb verbruikt.

dennissiemensma, 2019-04-13:
Bedankt voor je melding. Ik begrijp goed dat het wat vervelend kan zijn, maar dit is by design. De dagstatistieken kunnen pas berekend worden, zodra de eerste gasmeting van het uur ná (niet op) middernacht is geweest. Vaak is dat iets na 1 uur 's nachts. Ik zal later nog eens dubbelchecken of die logica nog steeds nodig/van toepassing is.

RichieB2B, 2019-04-14:
Mijn gasmeter geeft elke 5 minuten een meting af. Dus misschien kan je de logica aanpassen zodat de berekening gedaan wordt na de eerste gasmeting na middernacht?

dennissiemensma, 2019-04-15:
Ik zal daar later eens naar kijken. Het is lang geleden dat ik de logica gemaakt heb, maar er was wel een goede reden voor. Overigens was dat nog wel met DSMR 4, waarbij er slechts elk uur een nieuwe gasstand gepubliceerd werd.

dennissiemensma, 2019-10-02:
Ik heb een refactoring gedaan die niet zozeer naar de tijd kijkt, maar puur of er al een gasmeting op de nieuwe dag is gedaan. Dat zou het moeten oplossen.

 Ik probeer het eerst even hier voor een paar dagen en als het werkt gaat het mee in v2.6

closed 2019-10-02 by dennissiemensma

dennissiemensma, 2019-10-07:
Uitgebracht in [v2.6](https://github.com/dennissiemensma/dsmr-reader/releases/tag/v2.6.0).