dsmrreader/dsmr-reader #617: Vraag: jaarverbruik inzichtelijk maken

RualdOrdelman, 2019-04-17:
Ik gebruik met veel plezier de DSMR software. Echter mis ik iets wat volgens mij een mooie aanvulling zou zijn. 

Ik zou graag conform mijn energiecontract tellers kunnen zien. Je kunt nu dagtellers inzien, maar eigenlijk wil je ook weten of je maandverbruik een beetje conform je voorschot zit EN hoeveel verbruik je op jaarbasis hebt gehad.

Stel dat op 9 juli mijn jaarrekening wordt opgemaakt op basis van de dan geldende tellerstanden (afname en teruglevering). Dan zou ik graag willen dat het nieuw te tellen jaarverbruik vanaf dan ingaat. De totale tellerstand zal dan doormoeten tellen, zodat de teller van de meter ook doet. Echter de jaarteller moet dan gereset worden en de maandteller moet elke volle maand worden gereset.

Zou zoiets denkbaar zijn? Volgens mij een hele mooie aanvulling op de DSMR reader software. Ook omdat je ook al de energieprijzen van je contract kan invoeren. Als je daar dan ook de datum ik kan stellen waarop de jaarrekening wordt opgemaakt. :)

dennissiemensma, 2019-04-17:
Bedankt voor je suggestie. Ik ben het met je eens dat het een mooie aanvulling zou zijn, alleen is het wel wat lastig vergelijken met je energieleverancier. Je zit namelijk ook nog met (energie)belastingen, netwerkkosten en andere factoren die je leverancier misschien wel of niet in rekening brengt (zoals een welkomst- of verleng/trouwe-klant-korting).

Dit project is dan ook bewust niet bedoeld om de kosten van je leverancier te vergelijken. Wél om de meterstanden en je verbruik te monitoren (en dat te vergelijken met je leverancier). 
Voor je vraagstuk om dit per maand te kunnen zien, wellicht dat ik daar ooit nog iets voor kan maken, waarbij er rekening wordt gehouden met de dag van de maand waarop je contract ingaat. 

En om dit per jaar te kunnen inzien, is het denk ik het handigste om voor elke jaar een apart contract in te stellen, zelfs als je dezelfde prijzen hebt vastgelegd als het voorgaande jaar. Is misschien elk jaar wat werk, maar zou je hooguit een minuutje van je tijd moeten kosten.

RualdOrdelman, 2019-04-17:
Nice. Ik zou al enorm geholpen om het verbruik en opwekking per maand en jaar te kunnen loggen en bekijken. Voor de jaarlogging moet je dan inderdaad iets hebben als dag/maand dat het contract loopt. 

De prijs per kWh met alle extras dan in HA prima berekenen. Incl de opwekking en verschil tussen dag en nachttarief, saldering, etc.

PascalBu, 2019-11-25:
Is het ook mogelijk om dan vaste kosten toe te voegen? In mijn contract(en) vragen zowel de energieleverancier als de netbeheerder een dagelijks bedrag, wat nu niet meegenomen wordt in de kostenberekening van DSMR. Belasting is makkelijker, want dat bedrag kan evt per leveringseenheid worden bijgeteld (maar zou nog steeds mooi zijn om dit apart te kunnen registreren). De verrekening van de belasting op elektriciteit is dan weer dagelijks, dus dat zal dan weer bij de eerdere kosten terecht moeten komen. Ook een eventuele eenmalige korting van de leverancier kan hier nog bij dan?

M.a.w., kun je het mogelijk maken om de volgende kosten toe te voegen aan de tarieven om zo een betere benadering te geven van de daadwerkelijke kosten in vergelijking met het contract:
- Vaste leveringskosten electriciteit (per dag)
- Netbeheerkosten elektriciteit (per dag)
- Energiebelasting elektriciteit (per kWh)
- Vermindering energiebelasting elektriciteit (per dag)
- Vaste leveringskosten gas (per dag)
- Netbeheerkosten gas (per dag)
- Energiebelasting gas (per m3)
- Eenmalige korting leverancier (per jaar)

dennissiemensma, 2019-11-26:
@PascalBu bedankt voor je verzoek. Hoewel ik je wens zeker begrijp, zijn alle uitzonderingen ook exact de reden dat ik dit niet eerder heb ingebouwd. Er zijn teveel variabelen om dit goed te krijgen en het werkt niet voor iedereen. En elke keer als er een andere variant is, zal ik dat weer moeten inbouwen. Om nog maar te zwijgen over teruglevering en saldering.
Daarnaast is het doel van dit project puur om je een grove inschatting te geven van je verbruik, met de kosten van de eenheden als visualisatie. Er zitten teveel haken en ogen aan.

PascalBu, 2019-11-26:
Helemaal eens dat er heel veel scenario's te verzinnen zijn!

Maar toch vraag ik mij af of het simpelweg mogelijk is om één dagelijks bedrag ergens te kunnen definiëren wat bij de kosten wordt opgeteld? Dan kun je zelf voor al die verschillende scenario's berekenen wat dat dagelijks bedrag is, DSMR hoeft het alleen bij het dagtotaal op te tellen... Net zoals ik nu al doe voor de extra bedragen per eenheid.

dennissiemensma, 2019-11-27:
Ik zal het overwegen tzt. Dit issue staat in voorlopig nog niet op de planning.

dennissiemensma, 2020-05-21:
Ik heb hier nog een aantal keren opnieuw over nagedacht. Kort: Dit project is niet bedoeld om de energieleverancier exact te controleren, dus ik laat dit voor wat het is.

De gedetaileerde overweging hiervoor is dat het nu ook al mogelijk is om het verbruik voor een contract in te zien. Daarnaast kun je vanaf de volgende release ook wat makkelijker zoeken naar de exacte meterstanden van een dag. Daarmee hebben gebruikers al twee soorten gegevens om de rekening te 'controleren'. 
Verder zijn er ook nog wensen zoals #846, waarbij het nog wat makkelijker wordt om dit te doen. Ook is het nu al mogelijk om te zien wat het (jaar)totaal is van het huidige contract, zowel op contractbasis als op kalenderjaar basis. 

Verder valt de exacte facturatie van een leverancier/overheid, zoals de prijzen, belastingen, eenmalige kosten, vaste kosten, periodes, variabele prijzen en dergelijke buiten dit project. Daarnaast zijn die soms dusdanig complex, dat ik nooit kan garanderen dat het gaat kloppen voor alle situaties en leveranciers. Dat is een hele andere tak van sport.

closed 2020-05-21 by dennissiemensma