dsmrreader/dsmr-reader #618: Running DSMR and Pi-Hole on RPi3

Robert126, 2019-04-18:
Beste Dennis,

Als eerste mijn complimenten over de applicatie die je gebouwd hebt! 

### Jouw omgeving
*(Indien van toepassing)*
* DSMR-reader versie *(rechts bovenin applicatie zichtbaar)*: laatste versie (18 April 2019) 
* Hardware *(RaspberryPi 3)*: Raspbarian laatste versie, 16GB microSD
* DSMR-protocol versie van slimme meter * geen idee. Net DSMR geïnstalleerd
* Slimme meter: ISKRA AM550

### Bug, wens of iets anders
- Het probleem is een ip adres/ poort conflict
Zowel DSMR als Pi-Hole zitten op het zelfde ip adres en poort.
Ik heb geprobeerd om het poort nummer aan te passen in /etc/lighttpd/lighttpd.conf dit past kennelijk van alle bij de applicaties het poort nummer aan. Na reboot kan ik geen van beide webpagina's benaderen.

Hopelijk kun je me verder helpen om dit probleem op te lossen!
Hartelijke groet,
Robert

dennissiemensma, 2019-04-18:
Ik denk dat het het makkelijkste is als je de poort van de andere applicaties laat voor wat het is en die van dsmr-reader aanpast.

Wil je eens kijken of je dat lukt met de stappen uit dit oude issue -> #169 ?

Ik hoor graag als je er niet uitkomt.

Robert126, 2019-04-18:
Dank Dennis! Voor je zeer snelle reactie!

Dit is de oplossing zoals omschreven in #169  beschreven door Dennis. Deze werk nog steeds!

Yes you can change it to any port for dsmrreader, but it's not an application setting.

If you followed the installation guide from the docs, you are running a Nginx web server in front of the application. It's also the one your browser 'talks' with.

The web server determines which ports are used for serving applications. And you can change it quite easily:

Make sure you are logged in on your rpi as root user or a user with sudo-capabilities.
Open Nginx configuration file: sudo vi /etc/nginx/sites-enabled/dsmr-webinterface, you should see something similar as:

server {
    listen      80;
    server_name _;  # No hostname available? Just use the underscore for wildcard matching. :]

Change the directive listen 80; to anything you want, such as listen 8080;.
Save the file.
Have Nginx check you didn't make any typo's: sudo service nginx configtest.
It should say OK.
Now apply changes safely: sudo service nginx reload.
The application should now be accessible on the port you changed it to.

Sorry dat ik zelf #169 nietz zelf heb gevonden.
Fijne avond en een goed Pasen!
Robert

dennissiemensma, 2019-04-18:
Fijn om te horen dat het gelukt is en geen dank hoor. Daarvoor ben ik hier immers. Groet!

closed 2019-04-18 by dennissiemensma