dsmrreader/dsmr-reader #621: API integratie zonnepanelen (SolarEdge)

rodeho, 2019-04-20:
Ha Dennis

Je hebt geen emailadres publiek beschikbaar, vandaar even op deze manier. Er zijn ontwikkelaars welke een integratie hebben geschreven voor Zonnepanelen van SolarEdge. 

https://github.com/amrij/zonnepanelen (de electra variant heeft dan SE integratie)

https://gathering.tweakers.net/forum/list_messages/1721977

dennissiemensma, 2019-04-20:
Bedankt voor je verzoek. Ik heb echter een tijd geleden al besloten om geen integratie te doen met omvormers. Dit project richt zich namelijk puur op het uitlezen van de slimme meter en het weergeven en exporteren (bijvoorbeeld naar PVOutput.org) ervan.

De reden is o.a. dat het integreren van dergelijke functionaliteit veel tijd kost en ik het ook niet goed kan testen. Zelf heb ik immers geen zonnepanelen en omgeving om tegen te testen. Maar de belangrijkste reden is dat ik dan voor mijn gevoel iets maak wat iemand anders ook al gemaakt heeft (en met veel meer kennis), zoals PVOutput.org. Ik zou dan het wiel opnieuw proberen uit te vinden. En als ik de scope van dit project nog breder trek dan heeft dat ook grote gevolgen voor de ontwikkeling ervan.

Het kan zijn dat ik ooit een generieke ingang maak waarbij via de API in dsmr-reader gegevens mbt opwekking ingeschoten kunnen worden. Maar dat zal sowieso niet op korte termijn zijn.

rodeho, 2019-04-20:
Ha Dennis

Ik bedoelde de post omgedraaid. Er is vanuit die zonnepanelen website integratie naar jouw API. Zodat beide op één website worden afgebeeld. 

Cool toch :)

dennissiemensma, 2019-04-20:
Bedankt voor je toelichting. Het was mij niet helemaal duidelijk dat je dat bedoelde. 

Leuk om te lezen dat er een integratie gemaakt is! Mocht er nog ondersteuning nodig zijn dan hoor ik het graag. Sinds gisteren zijn er overigens twee API-calls toegevoegd die al lange tijd beschikbaar waren, maar misten in de documentatie. #619

closed 2019-04-20 by dennissiemensma