dsmrreader/dsmr-reader #623: Internal Server Error na update 2.0.2

hjkoster400d, 2019-04-21:
### Jouw omgeving
*(Indien van toepassing)*
* DSMR-reader versie *(rechtsbovenin applicatie zichtbaar)*: `v2.0.2`
* Hardware *(RaspberryPi 2, 3 of iets anders)*: `RaspberryPi 3`
* DSMR-protocol versie van slimme meter *(v2, v4 of v5)*: `v4`

### Bug, wens of iets anders
Na de upgrade van 1.x.x naar 2.0.2 krijg ik Internal Server Error. Tijdens de upgrade duurde de clearing cache stap erg lang, en gaf uiteindeljk een foutmelding, zie hieronder:

File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/requests/exceptions.py", line 9, in <module>
    from urllib3.exceptions import HTTPError as BaseHTTPError
ImportError: cannot import name 'HTTPError'

Een herstart van de raspberry heeft niet geholpen, heb je enig idee waar dit zou kunnen zitten?

Alvast bedankt!

p.s: ik ben nog steeds erg blij met de DSMR reader! Bedankt voor alle tijd die je erin stopt!

dennissiemensma, 2019-04-21:
Heb je nog een tweede keer ./deploy.sh geprobeerd?

dennissiemensma, 2019-04-21:
Je zou dit ook nog even kunnen proberen:

```
sudo su - dsmr
pip install urllib3 --force-reinstall
./deploy.sh
```

Dat herinstalleert de betreffende package `urllib3` als het goed is.

hjkoster400d, 2019-04-22:
Hi

Urllib3 opnieuw geïnstalleerd, en de betreffende error is verdwenen. Heb er wel een nieuwe voor terug😅

  File "/home/dsmr/dsmr-reader/dsmr_stats/services.py", line 7, in <module>
    from dateutil.relativedelta import relativedelta
ImportError: cannot import name 'relativedelta'

Heb ik iets bijzonders gemist tijdens de upgrade dat ik deze meldingen krijg?

In ieder geval heel erg bedankt voor je hulp 👍

Helloitsme, 2019-04-22:
Hier ook foutmeldingen. En kan de website ook niet meer bereiken.

jeroenpeters1986, 2019-04-22:
Hoi @hjkoster400d  en @Helloitsme 
Weten jullie ook welke Python versie jullie hebben? Ik heb 3.4.2 nog, en daar krijg ik de melding `SyntaxError: can use starred expression only as assignment target`.
Ik ga daarom even kijken of ik kan upgraden naar Python 3.6.x

Helloitsme, 2019-04-22:
is het ok als ik hier een dump plaats van alle meldingen tijdens de update?

Helloitsme, 2019-04-22:
Hierbij de meldingen die ik krijg tijdens de update

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Sat Jan  5 08:32:10 2019 from 192.168.2.1
pi@raspberrypi:~ $ sudo su - dsmr
(dsmrreader)dsmr@raspberrypi:~/dsmr-reader $ ./deploy.sh

--- You are currently running version:
1.28

Please make sure you've read EACH release note BEFORE deploying, since some
changes might not always be backwards compatible or depend on your database!

     >>>   https://github.com/dennissiemensma/dsmr-reader/releases   <<<

~~~ Still want to deploy? (type Y + ENTER to continue or press CTRL + C to abor                                                                                        t)
y

--- Pulling remote repository for new commits...
remote: Enumerating objects: 214, done.
remote: Counting objects: 100% (214/214), done.
remote: Compressing objects: 100% (41/41), done.
remote: Total 300 (delta 180), reused 197 (delta 163), pack-reused 86
Receiving objects: 100% (300/300), 204.39 KiB | 0 bytes/s, done.
Resolving deltas: 100% (203/203), completed with 64 local objects.
From https://github.com/dennissiemensma/dsmr-reader
   02deb78..9daf68c  master     -> origin/master
* [new branch]      562-live-gas-api -> origin/562-live-gas-api
* [new branch]      574-telegram -> origin/574-telegram
* [new branch]      588-dropbox-sync-issues -> origin/588-dropbox-sync-issues
   3f477b8..5958819  development -> origin/development
* [new tag]         v2.0.2     -> v2.0.2
* [new tag]         v2.0.0     -> v2.0.0
* [new tag]         v2.0.1     -> v2.0.1

--- Merging/updating checkout.
Updating 02deb78..9daf68c
Fast-forward
.github/ISSUE_TEMPLATE.md                          |   2 +-
.travis.yml                                        |   4 +-
docs/api.rst                                       | 108 +++
docs/changelog.rst                                 |  47 +-
docs/credits.rst                                   |   2 +
docs/locale/nl/LC_MESSAGES/api.mo                  | Bin 19590 -> 21282 bytes
docs/locale/nl/LC_MESSAGES/api.po                  | 160 ++--
docs/locale/nl/LC_MESSAGES/changelog.mo            | Bin 79598 -> 81268 bytes
docs/locale/nl/LC_MESSAGES/changelog.po            | 841 +++++++++++----------
docs/locale/nl/LC_MESSAGES/credits.mo              | Bin 11670 -> 11969 bytes
docs/locale/nl/LC_MESSAGES/credits.po              | 121 +--
dsmr_api/tests/v2/test_generic.py                  |   3 +
dsmr_consumption/services.py                       |   3 +-
dsmr_frontend/migrations/0016_v200_release.py      |  37 +
dsmr_frontend/migrations/0017_v201_release.py      |  37 +
dsmr_frontend/migrations/0018_v202_release.py      |  37 +
.../regression/test_renamed_docs_reverse_url.py    |   3 +
dsmr_frontend/tests/webinterface/test_dashboard.py |   3 +-
.../test_next_sync_setting_retroactive.py          |   3 +
dsmrreader/__init__.py                             |   3 +-
dsmrreader/locales/nl/LC_MESSAGES/django.mo        | Bin 73656 -> 74319 bytes
dsmrreader/locales/nl/LC_MESSAGES/django.po        |  13 +-
dsmrreader/provisioning/requirements/base.txt      |  12 +-
dsmrreader/provisioning/requirements/dev.txt       |   4 +-
dsmrreader/provisioning/requirements/mysql.txt     |   2 +-
.../provisioning/requirements/postgresql.txt       |   4 +-
dsmrreader/provisioning/requirements/test.txt      |   8 +-
[... 285 lines cut ...]
*** Error compiling '/tmp/pip-build-irh1zhlc/django/django/db/backends/base/sche                                                                                        ma.py'...
  File "/tmp/pip-build-irh1zhlc/django/django/db/backends/base/schema.py", line                                                                                         300
    "definition": ", ".join(constraint for constraint in (*column_sqls, *constra                                                                                        ints) if constraint),
                                                                     ^
SyntaxError: can use starred expression only as assignment target

*** Error compiling '/tmp/pip-build-irh1zhlc/django/django/db/backends/mysql/bas                                                                                        e.py'...
  File "/tmp/pip-build-irh1zhlc/django/django/db/backends/mysql/base.py", line 4                                                                                        3
    **conversions,
     ^
SyntaxError: invalid syntax

*** Error compiling '/tmp/pip-build-irh1zhlc/django/django/db/backends/mysql/int                                                                                        rospection.py'...
  File "/tmp/pip-build-irh1zhlc/django/django/db/backends/mysql/introspection.py                                                                                        ", line 98
    ))
    ^
SyntaxError: invalid syntax

*** Error compiling '/tmp/pip-build-irh1zhlc/django/django/db/backends/mysql/ope                                                                                        rations.py'...
  File "/tmp/pip-build-irh1zhlc/django/django/db/backends/mysql/operations.py",                                                                                         line 14
    **BaseDatabaseOperations.integer_field_ranges,
     ^
SyntaxError: invalid syntax

*** Error compiling '/tmp/pip-build-irh1zhlc/dj

dennissiemensma, 2019-04-22:
@Helloitsme je draait nog Python `3.4`. Als je DSMR-reader `v2.x` wilt gebruiken, zul je Python moeten upgraden. Tot die tijd kun je het beste teruggaan naar `v1.x`.

```
sudo su - dsmr
git checkout tags/v1.28.0
./post-deploy.sh
```

dennissiemensma, 2019-04-22:
@hjkoster400d wil je eens kijken of het herinstalleren van alle dependencies het oplost?

```
sudo su - dsmr
pip3 install -r dsmrreader/provisioning/requirements/base.txt --force-reinstall
./post-deploy.sh
```

jeroenpeters1986, 2019-04-22:
Ah, je hebt ook Python 3.4, we zuleln moeten upgraden :-)

Helloitsme, 2019-04-22:
Dank je.

Enige probleem dat ik nu heb is dat sudo su – dsmr ineens om een wachtwoord vraagt (deed het nooit en ik weet het eerlijk gezegd ook niet)

dennissiemensma, 2019-04-22:
@Helloitsme zit je wel onder de juiste root/pi-user? Wellicht zit je al onder de `dsmr` user?

Helloitsme, 2019-04-22:
> 
> 
> @Helloitsme zit je wel onder de juiste root/pi-user? Wellicht zit je al onder de `dsmr` user?

waarschijnlijk :-)
post deploy loopt nu

Helloitsme, 2019-04-22:
post deploy heeft gedraait. echter website nog niet toegangkelijk.
zie op het eind dit staan.
 Reloading process: dsmr_webinterface (Gunicorn)   [FAILED] PID file does not exist! (yet?)

dennissiemensma, 2019-04-22:
Ik denk dat de processen gecrasht zijn en dus niet herstarten. Je kunt ze geforceerd (her)starten.

Je moet dan even terug naar de root/pi-user (CTRL + D) en dan:
```
sudo supervisorctl restart all
```

Helloitsme, 2019-04-22:
> sudo supervisorctl restart all

:-) 👍

Helloitsme, 2019-04-22:
ben blij dat hij weer draait. 
Later maar even nadenken over wat er dan moet gebeuren om naar versie 2 te gaan.

dennissiemensma, 2019-04-22:
Ik kan je aanraden om te upgraden naar Python 3.5 of hoger. Python 3.4 is End of Life en ontvangt geen (security)updates meer van de ontwikkelaars. Het onderliggende framework dat ik gebruik (Django) ondersteunt die Python-versie nu ook niet meer. Dus dit project is nu backwards-incompatible met Python 3.4 (vandaar de versie-bump naar 2.0).

Afhankelijk van je OS kan Python upgraden veel of weinig werk zijn. Soms moet je het OS upgraden, soms kun je het er simpelweg naast installeren. 
En dan kun je zelf ook de keuze maken of dat het waard is. Het is ook een optie om het te laten voor wat het is, afhankelijk van hoeverre de huidige versie van dsmr-reader afdoende voor je is of niet.

Helloitsme, 2019-04-22:
> Ik kan je aanraden om te upgraden naar Python 3.5 of hoger. Python 3.4 is End of Life en ontvangt geen (security)updates meer van de ontwikkelaars. Het onderliggende framework dat ik gebruik [...]

Ik draai op een Pi, met daarop alleen dsmr. Staan er ergens instructies over hoe ik python kan upgraden?

dennissiemensma, 2019-04-22:
Ik splits dit topic even af, want uiteindelijk is een oude Python-versie een ander issue dan wat @hjkoster400d heeft.

@Helloitsme je kunt wellicht beter even naar #624 gaan.

hjkoster400d, 2019-04-22:
Heb raspbian geupgrade naar 9  #624  en de geforceerde herinstallatie gedaan van de dependencies. Alles werkt weer als vanouds! 

Super bedankt weer voor je hulp!

dennissiemensma, 2019-04-22:
Aah top, de upgrade gaat je sowieso wat werk schelen wanneer de volgende Python-versie EOL is. Bedankt voor je terugkoppeling!

closed 2019-04-22 by dennissiemensma