dsmrreader/dsmr-reader #624: Hulp bij upgraden naar Python 3.5+

dennissiemensma, 2019-04-22:
Splitsing van #623, over upgraden van Python 3.4.

dennissiemensma, 2019-04-22:
@Helloitsme welk OS draai je? Je kunt dit zien als root/pi met `sudo lsb_release -a`

Helloitsme, 2019-04-22:
> sudo lsb_release -a

raspbian 8.0

dennissiemensma, 2019-04-22:
Ik zie dat het mogelijk is om zelf Python 3.5 erbij te installeren. Het nadeel is alleen wel dat je Python dan zelf zult moeten compileren. En dat kan in een keer goed gaan, maar als je tegen issues aanloopt, kan ik je daar niet makkelijk mee helpen.

Het alternatief is om ofwel je huidige OS proberen te upgraden of om simpelweg je Pi opnieuw te installeren met de laatste Raspbian (9, Stretch). 

Dus deze opties heb je denk ik:
* Op Raspbian 8 blijven en nieuwe Python proberen te installeren. (http://www.knight-of-pi.org/installing-python3-6-on-a-raspberry-pi/)

* Upgraden naar Raspbian 9, met je huidige OS. (https://www.raspberrypi.org/documentation/raspbian/updating.md)

* Pi opnieuw installeren met Raspbian 9. (https://www.raspberrypi.org/documentation/installation/installing-images/README.md)

Zelf draai ik overigens ook Raspbian 9, maar ik moest destijds toch al noodgedwongen herinstalleren wegens een defecte SD-kaart.

jeroenpeters1986, 2019-04-22:
Ik zit zelf ook nog op Raspbian 8. Ik had de Knights-of-pi link inmiddels ook gevonden en ben bezig daarmee te installeren.

Als ik mijn setup weer werkend heb (want ik wil wel graag DSMR Reader 2.0), dan post ik mijn commandoset hier.

jeroenpeters1986, 2019-04-22:
Het is mij zojuist gelukt met deze set:

Ik ga er hierbij van uit dat je ingelogd bent op je pi als de `pi` user.

`
sudo apt-get install python3-dev libffi-dev libssl-dev -y
wget https://www.python.org/ftp/python/3.6.8/Python-3.6.8.tar.xz
tar xJf Python-3.6.8.tar.xz
cd Python-3.6.8
./configure
make
sudo make install
sudo pip3 install --upgrade pip
su - dsmr
deactivate
cd ~
mv .virtualenvs/dsmrreader .virtualenvs/old-dsmrreader
python3.6 -m venv ~/.virtualenvs/dsmrreader
workon dsmrreader
pip3 install -r dsmrreader/provisioning/requirements/base.txt --force-reinstall
pip3 install -r dsmrreader/provisioning/requirements/mysql.txt
./post-deploy.sh
`

Ik ben wel één van de weinigen die nog MySQL gebruikt, dus mocht iemand mijn instructies opvolgen, let er dan op dat je ook nog `pip3 install -r dsmrreader/provisioning/requirements/postgresql.txt` uitvoert voor de `./post-deploy.sh`

Als het goed werkt kun je de oude virtualenv verwijderen. Ik heb deze eerst verplaatst naar een andere naam omdat het post-deploy script er vanuit gaat dat je virtualenv `dsmrreader` heet, met die naam maak je direct de Python3.6 env aan.

Helloitsme, 2019-04-22:
Is de postgresql.txt  regel in plaats van of naast die met betrekking tot mysql?

jeroenpeters1986, 2019-04-22:
@Helloitsme, In het kort: Ja

Hoewel het installeren van de mysql lib niet erg is, kan het wel voor weer wat meldingen zorgen als mysql niet echt op je machine aanwezig is. Het is daarom inderdaad aan te bevelen `mysql.txt` te vervangen door `postgresql.txt` :-)

dennissiemensma, 2019-04-24:
@Helloitsme heb je het inmiddels kunnen oplossen?

Helloitsme, 2019-04-24:
Ik kwam helaas niet uit. Eerst geprobeerd om nieuwe versie van Python te installeren, maar liep weer vast toen de software weer begon te vragen om het dsmr password (en sudo su dsmr niet werkte).

En pip3 install -r dsmrreader/provisioning/requirements/base.txt --force-reinstall
Aangaf dat het bestand niet gevonden kon worden

Uiteindelijk besloten dat het waarschijnlijk beter was om vanaf nul weer te beginnen: ik heb raspbian versie 9 geinstalleerd en daarna dsmr weer opnieuw en de backup database gebruikt (mis nu de data van 1 dag, maar dat is niet zo’n drama)

Bedankt voor de support!

dennissiemensma, 2019-04-24:
Dat is (deels) spijtig om te horen, maar ook weer fijn dat je het nu weer werkend hebt. 

Ik verwacht dat je met Rasbian 9 weer een aardig tijdje vooruit kan. Mocht iets toch niet lekker werken dan hoor ik het uiteraard graag.

closed 2019-04-24 by dennissiemensma

Helloitsme, 2019-04-24:
In ieder geval een “wake up call”  dat alleen de dsmr software updaten (wat ik regelmatig deed) niet genoeg is….