dsmrreader/dsmr-reader #625: Meter statistics  blijft leeg

vincentweeda, 2019-04-23:
### Jouw omgeving
* DSMR-reader versie : v2.0.2
* Hardware: RaspberryPi 3
* DSMR-protocol versie van slimme meter:  v4
* Python versie (sudo su - dsmr):  3.5.3

### Bug
Wanneer ik de statistieken opvraag van mijn slimmemeter in het admin menu via de optie "DSMR Meter statistics (readonly)" krijg ik  sinds de nieuwe versie data terug.

Als ik kijk op het dashboard onder "Statistics" is het blok van de meter statisieken leeg. 

Zie ook de screenshots ter (hopelijk) verduidelijking.

[image]
[image]

dennissiemensma, 2019-04-23:
Dat is vreemd. Kun je eens in de database kijken of je gegevens ziet?

```
sudo su - postgres
psql dsmrreader
select * from dsmr_datalogger_meterstatistics;
```

vincentweeda, 2019-04-23:
De volgende info is zichtbaar in de database; 

`
 id |       timestamp        | electricity_tariff | power_failure_count | long_power_failure_count | voltage_sag_count_l1 | voltage_sag_count_l2 | voltage_sag_count_l3 | voltage_swell_count_l1 | voltage_swell_count_l2 | voltage_swell_count_l3 | rejected_telegrams | dsmr_version
----+------------------------+--------------------+---------------------+--------------------------+----------------------+----------------------+----------------------+------------------------+------------------------+------------------------+--------------------+--------------
  1 | 2019-04-23 16:20:11+00 |                  2 |                   0 |                        0 |                    0 |                      |                      |                      0 |                        |                        |                  0 | 42
(1 row)
`

dennissiemensma, 2019-04-23:
Het lijkt erop dat de gegevens wel bijgewerkt worden, maar dat ze leeg zijn. Alsin dat de meter zegt dat er geen storingen enzo geweest zijn. Als er geen gegevens zijn omte tonen, worden ze verborgen.

Dus in dat opzicht klopt het, ware het niet dat ik dan eigenlijk ook nog het kopje moet verbergen in dit geval.

vincentweeda, 2019-04-23:
Dat zou het wel duidelijker maken denk ik. Of eventueel de waardes ook weergeven indien ze nul zijn?

dennissiemensma, 2019-04-24:
Ik zal er over nadenken wat handig is.

dennissiemensma, 2019-09-10:
Ik heb het visueel aangepast voor de volgende versie. Ze worden altijd getoond. Als ze leeg zijn, staat er straks een `-`.

closed 2019-09-10 by dennissiemensma