dsmrreader/dsmr-reader #626: DSMR reader kapot -> Waar staat de database?

eterpstra, 2019-04-23:
### Jouw omgeving
*(Indien van toepassing)*
* DSMR-reader versie *(rechtsbovenin applicatie zichtbaar)*: `v1..28
* Hardware *(RaspberryPi 2, 3 of iets anders)*: `Ubuntu server 
* DSMR-protocol versie van slimme meter  v5

Hallo Dennis,

Om 1 of andere reden is mijn DSMR reader ermee opgehouden.ALs ik nu de webpagina open, krijg ik het volgende scherm te zien:

https://www.mupload.nl/img/0pbdzlfjqr27w.png

Als ik probeer de applicatie een upgrade te geven, dan krijg ik ook allemaal foutmeldingen:

[code]
 --- You are currently running version:
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "dsmrreader/__init__.py", line 17, in <module>
    from django.utils.version import get_version
ImportError: No module named django.utils.version

     [!] FAILED to call Django (did you activate the 'dsmrreader' VirtualEnv before running?)

[!] Halted in pre-deployment: Either user ABORTED or script FAILED
[/code]

Hoe kan ik dit in orde maken weer? Waar kan ik de database vinden zodat ik alles opnieuw kan installeren als het nodig is?

Alvast bedankt.

jeroenpeters1986, 2019-04-23:
Hallo @eterpstra, het ziet er naar uit dat er geen bestanden worden ingeladen die normaalgesproken 'static files' worden genoemd. 
Zit je onder de `dsmr`-user ingelogd, als je de `deploy.sh` probeert uit te voeren?

eterpstra, 2019-04-23:
Hallo Jeroen,

Dat klopt. Ik zit inderdaad onder dsmr user. Overigens lijkt de applicatie wel gewoon te draaien. Op mijn telefoon is de weergave nog wel goed (als ik incognito ga niet uiteraard)

eterpstra, 2019-04-23:
Als ik overigens nu naar configuratie ga in het menu , krijg ik dit te zien:

Server Error
Sorry, something unexpected happened.
Exception:
ProgrammingError: column dsmr_backup_backupsettings.folder does not exist
LINE 1: ...ess", "dsmr_backup_backupsettings"."backup_time", "dsmr_back...
                                                             ^
Traceback:
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/core/handlers/base.py", line 126, in _get_response
    response = wrapped_callback(request, *callback_args, **callback_kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/utils/decorators.py", line 142, in _wrapped_view
    response = view_func(request, *args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/views/decorators/cache.py", line 44, in _wrapped_view_func
    response = view_func(request, *args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/contrib/admin/sites.py", line 223, in inner
    return view(request, *args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/solo/admin.py", line 63, in change_view
    self.model.objects.get_or_create(pk=self.singleton_instance_id)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/db/models/manager.py", line 82, in manager_method
    return getattr(self.get_queryset(), name)(*args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/db/models/query.py", line 487, in get_or_create
    return self.get(**lookup), False

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/db/models/query.py", line 397, in get
    num = len(clone)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/db/models/query.py", line 254, in __len__
    self._fetch_all()

dennissiemensma, 2019-04-24:
Kan het zijn dat je `.bashrc` niet meer de virtualenv activeert? Probeer het eens handmatig en herinstalleer dan de packages:

```
sudo su - dsmr

source ~/.virtualenvs/dsmrreader/bin/activate
cd ~/dsmr-reader

pip3 install -r dsmrreader/provisioning/requirements/base.txt --force-reinstall

./deploy.sh
```

dennissiemensma, 2019-04-24:
En als antwoord op je vraag, De databasegegevens staan in `/var/lib/postgresql/`. Je kunt handmatig een backup maken met: 
`sudo -u postgres pg_dump dsmrreader | gzip --fast > /var/lib/postgresql/dsmr-backup.sql.gz`

De backup staat dan in `/var/lib/postgresql/dsmr-backup.sql.gz`.

eterpstra, 2019-04-24:
@dennissiemensma Bedankt voor je snelle reactie. Met betrekking tot het deployen: Dit is nu goed gegaan. Ik draai nu op v2.0.2. Echter blijft de gui er vreemd uit zien. het lijkt dus wel of er dingen ontbreken.

Overigens kan ik met sudo -u niet en backup maken. Hij geeft aan permission denied:

eterpstra@eterpstra:/var/lib/postgresql/10$ sudo -u postgres pg_dump dsmrreader | gzip --fast > /var/lib/postgresql/dsmr-backup.sql.gz
-bash: /var/lib/postgresql/dsmr-backup.sql.gz: Permission denied

dennissiemensma, 2019-04-24:
* Je zult het backup-command moeten uitvoeren als sudo-user. Dus ofwel als `root` ofwel als `pi`m als die user bestaat. Of je moet je eigen user sudo-rechten geven.

* Wat betreft de GUI. Ik vermoed dat het iets kleins is met de staticfiles, zoals Jeroen hierboven al aangaf. Kun je eens kijken met F12 in je browser wat de melding is? Ik vermoed dat er wat 404's staan.

eterpstra, 2019-04-24:
Gaat het om dit? https://www.mupload.nl/img/oi360p28j.png

Ik weet niet of je zoiets opnieuw kunt inladen? Voor het backuppen zal ik even proberen om als root in te loggen

eterpstra, 2019-04-24:
Met Root zie ik idd de bestanden maar hij kan postgres niet vinden. Bedoel je postgrey ?

dennissiemensma, 2019-04-24:
* Mbt de staticfiles, waarschijnlijk werkt dat niet goed, Kun je eens kijken of dit iets oplost: 
```
sudo su - dsmr
./manage.py collectstatic --noinput
```

Komen daar fouten uit? En zijn ze dan terug te vinden op: `/var/www/dsmrreader/static/`? 

* Wat voor OS draai je? Wellicht heet de database systeem-user anders.

eterpstra, 2019-04-24:
@dennissiemensma Ik krijg dan het volgende te zien:

dsmr@eterpstra:~/dsmr-reader$ ./manage.py collectstatic --noinput
Traceback (most recent call last):
  File "./manage.py", line 8, in <module>
    from django.core.management import execute_from_command_line
ModuleNotFoundError: No module named 'django'

Ik draai Ubuntu op een X86 machine

dennissiemensma, 2019-04-24:
Je zult eerst de bashrc van de `dsmr` user moeten fixen, want nu loop je constant tegen een niet-geactiveerde virtualenv aan. Of je moet hem bij elke login handmatig activeren.

Dit staat in de installatieinstructies: https://dsmr-reader.readthedocs.io/en/latest/installation/explained.html#virtualenv

Of:
```
sudo sh -c 'echo "source ~/.virtualenvs/dsmrreader/bin/activate" >> /home/dsmr/.bashrc'
sudo sh -c 'echo "cd ~/dsmr-reader" >> /home/dsmr/.bashrc'
```

Test even of dat goed werkt, staat ook in bovenstaande link.

Zo ja, probeer het nogmaals met collectstatic.

jeroenpeters1986, 2019-04-24:
hoi @eterpstra het lijkt er op dat je toch niet in een virtualenv zit, daarom kan hij Django niet vinden

eterpstra, 2019-04-24:
@dennissiemensma Oke bedankt! dat lijkt gelukt te zijn:

```
(dsmrreader) dsmr@eterpstra:~/dsmr-reader$ ./manage.py collectstatic --noinput
0 static files copied to '/var/www/dsmrreader/static', 440 unmodified.

```

dennissiemensma, 2019-04-24:
Ik weet niet of een herinstallatie zin heeft als zoiets als dit speelt. Het meest voorkomende issue in dit geval is simpelweg een rechten-issue. Je bent uiteraard vrij om dat te proberen. Je kunt de database dan gewoon laten staan en het aanmaken van de database + user in de handleiding overslaan.

Alles werkt echter, alleen kan de webserver (nginx) niet de statische bestanden serveren. Het lijkt erop dat `dsmr` wel bij de bestanden kan, maar wellicht de webserver niet. Kun je eens kijken in de logfiles van Nginx? Vermoedelijk is het pad: `/var/log/nginx/error.log`.

Ik verwacht dat daar fouten in staan in de trend van `permission denied`. Zodra je de GUI opent, zouden er nieuwe fouten bij moeten komen. Als dat zo is, is het even zoeken naar wat er exact veranderd is op je server qua rechten.

Je kunt dit ook makkelijk testen met: `sudo -u www-data ls -l /var/www/dsmrreader/static/`.
Als `www-data` geen toegang heeft tot die map, dan is het een rechten-issue.

dennissiemensma, 2019-05-04:
@eterpstra is dit uiteindelijk nog gelukt?

dennissiemensma, 2019-06-12:
Mocht ik je alsnog nog verder kunnen helpen dan hoor ik het graag.

closed 2019-06-12 by dennissiemensma