dsmrreader/dsmr-reader #627: wachtwoord dsmr user

newtech2015, 2019-04-25:
### Jouw omgeving
*(Indien van toepassing)*
* DSMR-reader versie *(rechtsbovenin applicatie zichtbaar)*: `v1.X.Y`
* Hardware *(RaspberryPi 2, 3 of iets anders)*: `RaspberryPi X...`
* DSMR-protocol versie van slimme meter *(v2, v4 of v5)*: `v...`

### Bug, wens of iets anders
* Bijvoorbeeld stappen hoe een probleem te reproduceren is...
* Of simpelweg een beschijving van een feature die je graag wenst te zien...

dennissiemensma, 2019-04-25:
De `dsmr` gebruiker heeft geen wachtwoord. Je kunt gewoon switchen naar deze gebruiker op de commandline, door `sudo su - dsmr` te doen. 

Let op dat je daarvoor wel eerst `root` of een gebruiker met `sudo`-rechten moet zijn, anders krijg je de vraag om het wachtwoord.

newtech2015, 2019-04-25:
ja, zo staat t in de gebruiksaanwijzing, maar dan vraagt hij het wachtwoord... dat weet ik niet, en is op geen enkele manier te wijzingen, kan zelfs user niet verwijderen

dennissiemensma, 2019-04-25:
Kun je een screenshot maken van je terminal? De meest voorkomende oorzaak is namelijk dat je dit met een user probeert te doen die geen root is of sudo-rechten heeft.

newtech2015, 2019-04-25:
ben nu het image aan het terug zetten, daar dsmr reader nu niet meer werkt.kom er op terug

newtech2015, 2019-04-25:
in de backup van vannacht zit een sql file.. maar, waar moet ik deze in kopiëren?

overigens, voor de 2e keer, is de hele installatie onbruikbaar na de update.. ik laat de update maar voor wat t is, en zet t image weer terug.. zou  fijn zijn, als ik de locatie van de sql file mag weten, zodat ik de database weer compleet heb zover

newtech2015, 2019-04-25:
dit is de output van de update:

root@bananapi_m2:~# su - dsmr
(dsmrreader)dsmr@bananapi_m2:~/dsmr-reader $ ./deploy.sh

 --- You are currently running version:
1.28

 Please make sure you've read EACH release note BEFORE deploying, since some
 changes might not always be backwards compatible or depend on your database!

     >>>   https://github.com/dennissiemensma/dsmr-reader/releases   <<<

 ~~~ Still want to deploy? (type Y + ENTER to continue or press CTRL + C to abort)
Y

 --- Pulling remote repository for new commits...
remote: Enumerating objects: 233, done.
remote: Counting objects: 100% (233/233), done.
remote: Compressing objects: 100% (60/60), done.
remote: Total 317 (delta 189), reused 205 (delta 161), pack-reused 84
Receiving objects: 100% (317/317), 216.48 KiB | 0 bytes/s, done.
Resolving deltas: 100% (212/212), completed with 64 local objects.
From https://github.com/dennissiemensma/dsmr-reader
   c5c038a..9daf68c  master     -> origin/master
 * [new branch]      518-voltages -> origin/518-voltages
 * [new branch]      562-live-gas-api -> origin/562-live-gas-api
 * [new branch]      574-telegram -> origin/574-telegram
 * [new branch]      588-dropbox-sync-issues -> origin/588-dropbox-sync-issues
   3f477b8..a468bd7  development -> origin/development
 * [new tag]         v2.0.2     -> v2.0.2
 * [new tag]         v2.0.0     -> v2.0.0
 * [new tag]         v2.0.1     -> v2.0.1

 --- Merging/updating checkout.
Updating c5c038a..9daf68c
Fast-forward
 .github/ISSUE_TEMPLATE.md                          |   5 -
 .travis.yml                                        |   4 +-
 docs/api.rst                                       | 108 +++
 docs/changelog.rst                                 |  47 +-
 docs/credits.rst                                   |   2 +
 docs/locale/nl/LC_MESSAGES/api.mo                  | Bin 19590 -> 21282 bytes
 docs/locale/nl/LC_MESSAGES/api.po                  | 160 ++--
 docs/locale/nl/LC_MESSAGES/changelog.mo            | Bin 79598 -> 81268 bytes
 docs/locale/nl/LC_MESSAGES/changelog.po            | 841 +++++++++++----------
 docs/locale/nl/LC_MESSAGES/credits.mo              | Bin 11670 -> 11969 bytes
 docs/locale/nl/LC_MESSAGES/credits.po              | 121 +--
 dsmr_api/tests/v2/test_generic.py                  |   3 +
 dsmr_consumption/services.py                       |   3 +-
 dsmr_frontend/migrations/0016_v200_release.py      |  37 +
 dsmr_frontend/migrations/0017_v201_release.py      |  37 +
 dsmr_frontend/migrations/0018_v202_release.py      |  37 +
 .../regression/test_renamed_docs_reverse_url.py    |   3 +
 dsmr_frontend/tests/webinterface/test_dashboard.py |   3 +-
 .../test_next_sync_setting_retroactive.py          |   3 +
 dsmrreader/__init__.py                             |   3 +-
 dsmrreader/locales/nl/LC_MESSAGES/django.mo        | Bin 73656 -> 74319 bytes
 dsmrreader/locales/nl/LC_MESSAGES/django.po        |  13 +-
 dsmrreader/provisioning/requirements/base.txt      |  12 +-
 dsmrreader/provisioning/requirements/dev.txt       |   4 +-
 dsmrreader/provisioning/requirements/mysql.txt     |   2 +-
 .../provisioning/requirements/postgresql.txt       |   4 +-
 dsmrreader/provisioning/requirements/test.txt      |   8 +-
 dsmrreader/provisioning/requirements/travis.txt    |   2 +-
 28 files changed, 937 insertions(+), 525 deletions(-)
 create mode 100644 dsmr_frontend/migrations/0016_v200_release.py
 create mode 100644 dsmr_frontend/migrations/0017_v201_release.py
 create mode 100644 dsmr_frontend/migrations/0018_v202_release.py

[... 590 lines cut ...]

 --- Reloading app code...

 * Reloading process: dsmr_webinterface (Gunicorn)   [FAILED] PID file does not exist! (yet?)
 * Reloading process: dsmr_backend                   [OK]
 * Reloading process: dsmr_datalogger                [FAILED] PID file does not exist! (yet?)
 * Reloading process: dsmr_mqtt                      [OK]

 --- Clearing cache...
Traceback (most recent call last):
  File "./manage.py", line 8, in <module>
    from django.core.management import execute_from_command_line
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/core/management/__init__.py", line 260
    subcommands = [*get_commands(), 'help']
                          ^
SyntaxError: can use starred expression only as assignment target

 --- Deployed version:
2.0.2

 >>> Deployment complete. <<<
(dsmrreader)dsmr@bananapi_m2:~/dsmr-reader $
(dsmrreader)dsmr@bananapi_m2:~/dsmr-reader $ supervisorctl start all
error: <class 'socket.error'>, [Errno 13] Permission denied: file: /usr/lib/python2.7/socket.py line: 224
(dsmrreader)dsmr@bananapi_m2:~/dsmr-reader $

newtech2015, 2019-04-25:
hoi,
de update geef ik op, daarna werkt t programma niet meer.nu heb ik een backup file van afgelopen nacht, die wil ik graag met winscp op de juiste locatie op de pi koppieeren, echter kom ik niet achter de locatie van de sql file. Zou je deze willen prijsgeven?
groeten, ron

dennissiemensma, 2019-04-25:
Als het goed is kun je gewoon terug naar de vorige versie met: `sudo su - dsmr` en dan `git checkout tags/v1.28.0` en tot slot `./post-deploy.sh`. Kun je dat eens proberen? Ik zit nu niet achter mijn PC dus ik kan niet alles makkelijk voor je uitwerken. Het (her)starten van Supervisor kun je trouwens alleen als root doen.

newtech2015, 2019-04-25:
ik kom niet verder met het terug zetten van de backup, dit is wat ik doe:
eerst de drop-box backup van afgelopen nacht 2.00 naar locatie /home/dsmr/dsmr-reader/backups/ gekopieerd

supervisorctl stop all

cd /home/dsmr/dsmr-reader/backups/
zcat /home/dsmr/dsmr-reader/backups/dsmrreader-postgresql-backup-Thursday.sql.gz | sudo sudo -u postgres psql dsmrreader

supervisorctl start all

hierna moet ik een reboot doen, om het process van het meter-uitlezen weer aan de gang te krijgen, ga ik dan bij archief, 24-april kijken, dan staat de data er tot gisteren 12.00 in, dat is datum zoals t in t image van kaartje zit... maar de backup is van afgelopen nacht 2.00.
er wordt dus niks terug gezet!
gebruik dit programma nu ruim 2 weken, en nu.. door een update-poging, die niet werkte.. kom ik erachter dat terughalen van data niet gaat.. dit heeft bij mij meer prio dan de update.. want, als dit niet goed werkt heeft hele logging geen zin.
wat doe ik fout?

jeroenpeters1986, 2019-04-25:
In alle eerlijkheid, bij de update wordt gevraagd of je alle changelogs doorgenomen hebt. Heb je dat gedaan? 
Ik niet en daarom kwam ik er achter dat mijn Python versie niet meer de juiste was. Sinds versie 2.0 van DSMR-reader moet je Python 3.5 of hoger gebruiken, of blijven op de vorige (die overigens uitstekend werkt). Ik denk dat dat ook bij jou het geval is. Ik kon upgraden op Raspbian (zie ook issue #624), maar ik weet niet of dat bij jou ook zo is omdat je bananapi gebruikt.

Wat er denk ik in jouw geval aan de hand is, is dat je database niet (meer) leeg is. 
Wat je nog zou kunnen doen VOOR je restore commando is.

`
sudo sudo -u postgres -c "drop database dsmrreader"
sudo sudo -u postgres createdb -O dsmrreader dsmrreader
`

Daarmee gooi je de database weg en maak je hem weer aan. Hopelijk lukt het daarmee. Ik hoor het graag

newtech2015, 2019-04-25:
misschien is dit een goed oplossing, maar hij wil dit commando niet:
root@bananapi_m2:/home/dsmr/dsmr-reader/backups# sudo sudo -u postgres -c "drop database dsmrreader" -u postgres createdb -O dsmrreader dsmrreader
sudo: kan computernaam bananapi_m2 niet herleiden
usage: sudo -h | -K | -k | -V
usage: sudo -v [-AknS] [-g group] [-h host] [-p prompt] [-u user]
usage: sudo -l [-AknS] [-g group] [-h host] [-p prompt] [-U user] [-u user]
            [command]
usage: sudo [-AbEHknPS] [-r role] [-t type] [-C num] [-g group] [-h host] [-p
            prompt] [-u user] [VAR=value] [-i|-s] [<command>]
usage: sudo -e [-AknS] [-r role] [-t type] [-C num] [-g group] [-h host] [-p
            prompt] [-u user] file ...
iemand een idee?

jeroenpeters1986, 2019-04-25:
Ik denk dat ik een “sudo” te veel heb, sorry!

-- 
Groeten,

Jeroen Peters

newtech2015, 2019-04-25:
sudo sudo -u postgres createdb -O dsmrreader dsmrreader

is precies wat in gebruiksaanwijzing staat, maar werkt helaas niet, 1 sudo of 2 minder werkt ook niet.
Weet niemand waar die database precies staat, kan er met winscp gewoon bij namelijk

jeroenpeters1986, 2019-04-25:
@newtech2015  de database is niet gewoon een bestandje op je harde schijf, het is een bepaald formaat je niet zo even kunt editen. Er staat wel wat in de docs (https://dsmr-reader.readthedocs.io/en/latest/faq.html#how-can-i-move-the-database-location) maar daar heb je niets aan.

Wil je, als je aangeeft dat iets niet werkt, ook aangeven wat je op je scherm krijgt?

Probeer het nogmaals met:
`
sudo -u postgres psql -c "drop database dsmrreader"
sudo -u postgres psql -c "create database dsmrreader"
sudo -u postgres psql -c "grant all privileges on database dbname to dbuser"
`

dennissiemensma, 2019-04-25:
@newtech2015 ik ben inmiddels weer thuis en het is spijtig om te horen dat je inmiddels een herinstallatie hebt gedaan. Heb je toevallig ook al de database verwijderd? Zo nee, doe dat vooral niet.

Nu ik alles teruglees, zie ik dat alle problemen zijn ontstaan omdat je een oudere versie van Python gebruikt:
```
 File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.4/site-packages/django/core/management/__init__.py", line 260
   subcommands = [*get_commands(), 'help']
                         ^
```

Nou was dat niet heel erg. En kon je zo weer terug naar de vorige DSMR-reader versie met behoud van al je gegevens. Alleen hangt dat nu even af van of je database nog bestaat.

jeroenpeters1986, 2019-04-25:
Als dat zo was (dat @newtech2015 nog terugmoet, ik dacht dat al gebeurd was), dan is het verwijderen van de database inderdaad niet nodig.

dennissiemensma, 2019-04-25:
@newtech2015 als je de database nog niet verwijderd hebt, kun je dit proberen:

```
sudo su - dsmr
git checkout tags/v1.28.0
./post-deploy.sh
```

Daarna weer even terug naar de `root`-user en dan:
```
supervisorctl restart all
```

newtech2015, 2019-04-25:
mijn dank aan iedereen die probeert mee te denken!

ik heb, toevallig, gisteren een kaart-image gemaakt.. waardoor ik god dank.. niet al teveel data kwijt ben.momenteel, ben ik weer dit image aan het terugzetten.
echter, geen idee waarom... maar, het restoren van de backup die NA dit image gemaakt is, lijkt zinloos... de data die erin moet zitten, wordt niet zichtbaar in de historie.
dus, wat doe ik nu fout?
zo heb ik de data gerestored:eerst drop-box backup van vannacht gekopieerd naar: /home/dsmr/dsmr-reader/backups
supervisorctl stop all

cd /home/dsmr/dsmr-reader/backups/
zcat /home/dsmr/dsmr-reader/backups/dsmrreader-postgresql-backup-Thursday.sql.gz | sudo sudo -u postgres psql dsmrreader

supervisorctl start all
moet ik mss daarna nog iets doen, zodat hij database her-bouwt of herlaadt?
dat restoren goed werkt, is voor mij wel echt heel belangrijk, anders heeft het hele loggen weinig zin vind ik.

dennissiemensma, 2019-04-25:
@newtech2015 dus ik begrijp dat je de oude database overschreven hebt? Het is belangrijk dat je bij ons aangeeft wat je wel en niet gedaan hebt, want dat is mij niet helemaal duidelijk.

* Heb je de code nu uitgecheckt op de vorige versie (v1.28.0)? Als je een herinstallatie doet, krijg je standaard de laatste versie (2.0.2) en die werkt niet met jouw Python-versie.

* Het restoren werkt perfect, maar het is mij niet duidelijk welke stappen je exact doet. Ik zie commands, maar ik mis de output en je mist ook nog andere commands daarvoor.

Ik ben zo weer afwezig, dus ik kan je alleen wat tips geven.

* **Als je zeker weet dat je geen poging wilt doen om je bestaande data te herstellen**, of je het al verwijderd hebt, dan kun je de database verwijderen, opnieuw aanmaken en vullen met je Dropbox backup:
```
# Als ROOT user
sudo sudo -u postgres dropdb dsmrreader
sudo sudo -u postgres createdb -O dsmrreader dsmrreader
zcat /home/dsmr/dsmr-reader/backups/dsmrreader-postgresql-backup-Thursday.sql.gz | sudo sudo -u postgres psql dsmrreader
```

* Voordat je dingen start, zorg ervoor dat je NIET op de laatste versie zit:
```
# Als ROOT user
sudo su - dsmr
git checkout tags/v1.28.0
./post-deploy.sh
```

* Tot slot kun je alles weer starten:
```
# Als ROOT user
supervisorctl start all
```

newtech2015, 2019-04-25:
Hoi,
nou.. nu werkt het weer met de gegevens uit de backup erbij, heel erg bedankt!!het zat 'm in het dropdb commando
super!
Ron

closed 2019-04-25 by newtech2015