dsmrreader/dsmr-reader #628: Vragen en suggestie

bkorsten, 2019-04-25:
### Jouw omgeving
*(Indien van toepassing)*
* DSMR-reader versie *(rechtsbovenin applicatie zichtbaar)*: 2.0.2
* Hardware *(RaspberryPi 2, 3 of iets anders)*: `RaspberryPi 3`
* DSMR-protocol versie van slimme meter *(v2, v4 of v5)*: `v5`

### Bug, wens of iets anders
Vraag. Ik gebruik de software nu zo’n drie maanden maar zie dat er ondertussen 6.228.462 telegrammen met een foute CRC zijn detecteerde. Dit lijkt me extreem hoog. Ik heb al eens een ander kabel op m’n P1 poort geprobeerd, maar lijkt niets uit te maken. De data komt redelijk overeen met dat wat mijn energie maatschappij rapporteerde (op een paar euro per maand na). Ik heb een ISKRA AM550 uit 2017 (onlangs geïnstalleerd). Is dit een bekend issue ? Of mogelijk een probleem met de P1 poort ?

Wens. Ik heb een PI touch lcd scherm waarop ik het dashboard wil koppelen, alleen is nu alles erg ini-mini en niet heel erg touch vriendelijk (via de raspberry browser). Is het een idee om een voor dit doel specifiek dashboard te creëren ? Met bijvoorbeeld het huidig gebruik versus het gemiddelde gebruik van die week oid ? Het idee dat ik er bij heb is om dit display in de woonkamer te zetten en op die manier heel snel te kunnen zien wat de status van het verbruik is.

Verder bijzonder tevreden over de tool ! Zeer doordacht en erg volwassen product :)

dennissiemensma, 2019-04-26:
Bedankt voor je vragen. 

* Bedoel je dat 1/3 telegrammen foutief is of lees ik de aantallen verkeerd? Je zou nog even kunnen kijken in de logs-folder van het project. In `logs/django.log` of `logs/dsmrreader.log` zou dan ook terug moeten komen dat de CRC niet klopt. Ik zou er eentje voor je kunnen narekenen/checken.

* Wat betreft het Pi-touch screen, ik kan dat zelf niet makkelijk ontwikkelen/testen en iedereen heeft weer andere voorkeuren. Wellicht kun je zelf iets simpels in elkaar draaien qua HTML? Via de API van je dsmr-reader kun je dan de gegevens ophalen die je zoekt. En vervolgens weergeven zoals je ze wilt.

bkorsten, 2019-05-05:
Hey Dennis; mbt tot het aantal foute meldingen; de verhouding is omgekeerd :( dus 2/3 van de telegrammen hebben een CRC melding. Ik neem dit nog wel even op met mijn leverancier; ik krijg de CRC meldingen ook in andere paketten (Domiticz).

De view heb ik uiteindelijk opgelost via een mqtt broker (openhab2.4) die data opslaat in influxdb, welke ik dan visualiseer in grafana. Via [deze ](https://community.openhab.org/t/influxdb-grafana-persistence-and-graphing/13761) website kun je vinden hoe.

closed 2019-05-05 by bkorsten

dennissiemensma, 2019-05-05:
Bedankt voor je terugkoppeling. Mooi dat het uiteindelijk gelukt is.