dsmrreader/dsmr-reader #629: update naar 2.02 lukt niet.. probleem python...

newtech2015, 2019-04-26:
hoi,

ik heb de moeite genomen, om python 3.7 te compilen, en middels "alternatives", in te stellen, dat ik versie 3.7 als standaard ga gebruiken. Dit is allemaal gelukt, en met commando python -V krijg ik netjes te zien: Python 3.7.0

maar: doe ik het op de "dsmr reader manier": sudo su - dsmr en dan python --version, dan krijg ik tot mijn verbazing toch nog 3.4.1 te zien.  heb eens geprobeert gewoon te updaten, maar helaas, dsmr weer helemaal over de zeik, en kon ook niet meer terug naar 1.28.
 
nu weer image van kaartje terug gezet en backup erin gekopieert, en t werkt weer! 

maar.. ik snap niet.. waarom dsmr de oude versie blijft aangeven, terwijl de rest van t OS 3.7 gebruikt.

iemand die hier iets over weet te melden?

jeroenpeters1986, 2019-04-26:
Waarschijnlijk is je alternative alleen ingesteld voor je andere user waar je normaal mee inlogt ;-)
Heb je ook een alternative ingesteld voor je `dsmr` user?

dennissiemensma, 2019-04-26:
@newtech2015 waarschijnlijk moet je de virtualenv opnieuw aanmaken, met de nieuwe Python-versie. Bij de installatie wordt de versie vastgelegd in de virtualenv, zodat je meerdere versies naar elkaar kan draaien.

Ik zal even voor je kijken hoe je dit kunt doen. Ik verwacht dat het vrij weinig werk is.

jeroenpeters1986, 2019-04-26:
In https://github.com/dennissiemensma/dsmr-reader/issues/624#issuecomment-485377652 beschrijf ik hoe ik dat voor Python 3.6 heb gedaan, voor Python 3.7 geldt hetzelfde.

```
sudo su - dsmr
deactivate
cd ~
mv .virtualenvs/dsmrreader .virtualenvs/old-dsmrreader
python3.7 -m venv ~/.virtualenvs/dsmrreader
workon dsmrreader
pip3 install -r dsmrreader/provisioning/requirements/base.txt --force-reinstall
pip3 install -r dsmrreader/provisioning/requirements/mysql.txt
./post-deploy.sh
```

Python 3.7 moet dus wel aanroepbaar zijn voor de dsmr user

dennissiemensma, 2019-04-26:
Hier wat stappen om eerst parallel aan de bestaande virtualenv eentje aan te maken en in te richten. Als dat goed gaat, dan pas de applicatie stoppen en wisselen tussen virtualenvs. 

Het voordeel is ook dat je altijd terug kan naar de oude.

```
# als ROOT
apt-get install python3.7-dev

sudo su - dsmr
deactivate

virtualenv ~/.virtualenvs/dsmrreader-NEW/ --python python3.7
source ~/.virtualenvs/dsmrreader-NEW/bin/activate
pip3 install -r dsmrreader/provisioning/requirements/base.txt -r dsmrreader/provisioning/requirements/postgresql.txt
```
* (!) Let op, als je op dit punt fouten krijgt, **ga dan NIET verder hieronder**. De applicatie loopt dan gewoon door op de oude versie.
```
# Nu terug naar root
CTRL + D
supervisorctl stop all
```

```
# Virtualenv wisselen
sudo su - dsmr
deactivate
mv ~/.virtualenvs/dsmrreader/ ~/.virtualenvs/dsmrreader-OLD/
mv ~/.virtualenvs/dsmrreader-NEW/ ~/.virtualenvs/dsmrreader/
source ~/.virtualenvs/dsmrreader/bin/activate

# Controleer nu of je de juiste Python-versie hebt met:
python -V
```

Alles goed? Dan de applicatie weer starten. Voortaan zou die dan onder Python 3.7 moeten draaien.
```
# Als ROOT
supervisorctl start all
```

newtech2015, 2019-04-26:
hoi,
zojuist python 3.7 op gerestored imaged weer opnieuw gecompiled, en nu verder met de "tips" (zie tussen oplossingen in)maar weer gestopt met het "even updaten".. 

 
 

dit geeft resultaat:
root@bananapi_m2:~# apt-get install python3.7-dev                               Pakketlijsten worden ingelezen... Klaar
Boom van vereisten wordt opgebouwd
De statusinformatie wordt gelezen... Klaar
E: Kan pakket python3.7-dev niet vinden
E: Kon geen enkel pakket vinden via regex 'python3.7-dev'

resultaat:
dsmr@bananapi_m2:~/dsmr-reader $ deactivate
-su: deactivate: opdracht niet gevonden

resultaat:
dsmr@bananapi_m2:~/dsmr-reader $ virtualenv ~/.virtualenvs/dsmrreader-NEW/ --python python3.7
Running virtualenv with interpreter /usr/bin/python3.7
Using base prefix '/usr/local/opt/python-3.7.0'
/usr/lib/python3/dist-packages/virtualenv.py:1126: DeprecationWarning: the imp module is deprecated in favour of importlib; see the module's documentation for alternative uses
  import imp
New python executable in /home/dsmr/.virtualenvs/dsmrreader-NEW/bin/python3.7
Not overwriting existing python script /home/dsmr/.virtualenvs/dsmrreader-NEW/bin/python (you must use /home/dsmr/.virtualenvs/dsmrreader-NEW/bin/python3.7)
Installing setuptools, pip...
  Complete output from command /home/dsmr/.virtuale...er-NEW/bin/python3.7 -c "import sys, pip; sys...d\"] + sys.argv[1:]))" setuptools pip:
  Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 668, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 638, in _load_backward_compatible
  File "/usr/share/python-wheels/pip-1.5.6-py2.py3-none-any.whl/pip/__init__.py", line 72, in <module>
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 668, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 638, in _load_backward_compatible
  File "/usr/share/python-wheels/pip-1.5.6-py2.py3-none-any.whl/pip/log.py", line 9, in <module>
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 668, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 638, in _load_backward_compatible
  File "/home/dsmr/.virtualenvs/dsmrreader-NEW/lib/python-wheels/setuptools-5.5.1-py2.py3-none-any.whl/pkg_resources.py", line 1520, in <module>
AttributeError: module 'importlib._bootstrap' has no attribute 'SourceFileLoader'
----------------------------------------
...Installing setuptools, pip...done.
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/virtualenv.py", line 2378, in <module>
    main()
  File "/usr/lib/python3/dist-packages/virtualenv.py", line 830, in main
    symlink=options.symlink)
  File "/usr/lib/python3/dist-packages/virtualenv.py", line 1032, in create_environment
    install_wheel(to_install, py_executable, search_dirs)
  File "/usr/lib/python3/dist-packages/virtualenv.py", line 975, in install_wheel
    'PIP_NO_INDEX': '1'
  File "/usr/lib/python3/dist-packages/virtualenv.py", line 917, in call_subprocess
    % (cmd_desc, proc.returncode))
OSError: Command /home/dsmr/.virtuale...er-NEW/bin/python3.7 -c "import sys, pip; sys...d\"] + sys.argv[1:]))" setuptools pip failed with error code 1

dennissiemensma, 2019-04-26:
**.bashrc**
Wat je sowieso nog lijkt te missen is een aangepaste `.bashrc` van de `dsmr` gebruiker. Deze activeert telkens de virtualenv, waardoor je dat niet handmatig hoeft te doen en minder fouten krijgt.

Het staat in deze stap van de installatie: https://dsmr-reader.readthedocs.io/en/latest/installation/explained.html#virtualenv

Of je kunt dit doen:
```
sudo sh -c 'echo "source ~/.virtualenvs/dsmrreader/bin/activate" >> /home/dsmr/.bashrc'
sudo sh -c 'echo "cd ~/dsmr-reader" >> /home/dsmr/.bashrc'
```

Als je namelijk `sudo su - dsmr` doet, dan zou je namelijk dit moeten zien:
```
(dsmrreader) dsmr@bananapi_m2:~/dsmr-reader $
````
Ipv:
```
dsmr@bananapi_m2:~/dsmr-reader $
```

----

**Updaten**
Je zou de manier van Jeroen hierboven nog kunnen proberen. Daarvoor kun je de map van de eerdere poging weggooien: ` /home/dsmr/.virtualenvs/dsmrreader-NEW/`

Let op dat je niet weer in paniek raakt als er iets mis gaat. Een van de eerste stappen is namelijk het verplaatsen van de oude virtualenv, zodat je die altijd kan terugzetten:
```
mv .virtualenvs/dsmrreader .virtualenvs/old-dsmrreader
```

Terugzetten kan dan met:
```
rm -rf .virtualenvs/dsmrreader
mv .virtualenvs/old-dsmrreader .virtualenvs/dsmrreader
```

newtech2015, 2019-04-26:
Hoi,
ondanks alle moeite.. en 2 complete dagen werk.. blijft de dsmr user vinden dat hij lekker de oude versie moet blijven gebruiken.nog gevonden, dat: alias python='/usr/bin/python3.7'
. ~/.bashrc
het even op 3.7 zet.. echter na een reboot staat er weer 3.4.2

ook als hij 3.7 aangeeft, kan ik niet updaten, en legt hij het hele programma lam.

ik geef dit dan ook maar op.. t is welletjes geweest

bedankt voor alle hulp

closed 2019-04-26 by dennissiemensma

jeroenpeters1986, 2019-04-26:
Jammer. Het blijft vreemd dat het niet lukt, zo’n alias zou ook niet nodig
zijn overigens.
Wat raar is is dat het wel werkt voor de oude versie waar je een zelfde
setup voor nodig hebt.
Heb je wel de python3.7 virtualenv aan kunnen maken met mijn eerdere
commando’s? Dat zie ik namelijk niet terug en daar ben ik wel benieuwd naar
nog.
-- 
Groeten,

Jeroen Peters

newtech2015, 2019-04-26:
Hoi, 
Nee, elk commando foutmeldingen. 
Groetjes 

Verzonden via Yahoo Mail op Android

mjanssens, 2019-07-11:
Blij dat ik dit issue gevonden heb, dank aan de melder en commenters.
Ik heb dsmrreader in een lxc debian container draaien.
Die heb ik geupgrade van Stretch naar Buster. Maar daarna werkte vanalles niet meer, nginx en supervisor processen wilden niet meer starten.
Met de hulp van de eerdere comments heb ik mijn bestaande virtualenv veiliggesteld en een nieuwe aangemaakt en alles draait nu als een zonnetje op Buster.
Mijn backend is overigens nog niet over naar Postgresql 11, de container heeft dsmrreader en is enkel db client (wel 11) die connect naar een 9.6 database server (Debian Stretch).
Zodra die db server overgaat naar 11 moet de database herindexeerd worden:
https://wiki.postgresql.org/wiki/Locale_data_changes
https://www.debian.org/releases/buster/amd64/release-notes/ch-information.en.html#postgresql-reindex

Ik weet niet of raspbian buster al standaard op postgresql 11 zit, wellicht dat herindexeren dan ook voor rpi geldt.