dsmrreader/dsmr-reader #630: backup niet te herstellen

rodeho, 2019-04-27:
Hoi Dennis

ik wil vanaf de Pi migreren naar een VM met Ubuntu 18. Die VM die draait prima, en ook DSMR staat klaar. Echter gaat het mis bij het importeren van de backup file, ik heb vrijdag/zaterdag getest, zipped en unzipped. Zowel via Dropbox als lokaal, er lijkt wel een gasmeting binnen te komen, Dropbox API settings etc.

Maar de 2.5miljoen metingen die komen niet goed in de DB...

Zou je eens mee willen denken?

`COPY 1
 setval
--------
      1
(1 row)

ERROR:  null value in column "id" violates not-null constraint
DETAIL:  Failing row contains (null, null, null, null, null, null, null, null, null, null, null, null, null, null, null, null, null).
CONTEXT:  COPY dsmr_datalogger_dsmrreading, line 1554873: "\N	\N	\N	\N	\N	\N	\N	\N	\N	\N	\N	\N	\N	\N	\N	\N	\N"
root@ubuntu:/home/dsmr#`

Versie 2.0.2

rodeho, 2019-04-27:
Deze data zou ik graag over zetten ;)
[image]

dennissiemensma, 2019-04-27:
Bedankt voor je vraag. Om je te helpen heb ik nog wat meer informatie nodig.

* Heb je op de VM eerst de applicatie helemaal gestopt? Je moet eerst alles stoppen, de DB verwijderen (op de VM), leeg aanmaken en dan de import doen.

* Hoe krijg je op de VM de nieuwe metingen binnen? Via de API?

rodeho, 2019-04-27:
Ha Dennis

jeps de 3 taken worden gestopt met het "sudo supervisorctl stop all" commando. Voorafgaande aan de import.

- Bij een ruwe dump krijg ik deze output op de bron Pi.
```
pi@raspberrypi:~ $ sudo sudo -u postgres pg_dump dsmrreader > backup_0427.pgsql
pg_dump: Dumping the contents of table "dsmr_datalogger_dsmrreading" failed: PQgetResult() failed.
pg_dump: Error message from server: ERROR:  timestamp out of range
pg_dump: The command was: COPY public.dsmr_datalogger_dsmrreading (id, "timestamp", electricity_delivered_1, electricity_returned_1, electricity_delivered_2, electricity_returned_2, electricity_currently_delivered, electricity_currently_returned, extra_device_timestamp, extra_device_delivered, processed, phase_currently_delivered_l1, phase_currently_delivered_l2, phase_currently_delivered_l3, phase_currently_returned_l1, phase_currently_returned_l2, phase_currently_returned_l3) TO stdout;
pi@raspberrypi:~ $
```

- Ik was nu van plan om de VM een USB koppeling te geven (via de USB NUC port) zodat hij op een normale manier de USB P1 lezer ziet. Dit heb ik echter nog niet gekoppeld omdat ik 1:1 over wil zonder teveel dataverlies en de DB eerst up wilde hebben.

- Op zich zou ik het ook wel willen proberen om de data via de API te laten transporteren, echter is dit niet helemaal helder beschreven. Er staan op diverse plaatsen posts welke volgens mij _bijna_ hetzelfde zijn, herinner ik mij.

Is het voor jou mogelijk om de stappen van een aparte datalogger -> API -> VM met DB en WEBGUI nog een maal hieronder te kunnen zetten waarbij dan ook de DB en WEBGUI processen uitgeschakeld zijn om de Pi rust te geven. Of de post aan te halen welke het meest recent en kloppend is? Het is vooral ook mbt upgrades etc.

dennissiemensma, 2019-04-28:
De foutmelding zelf zegt mij niet zoveel, als in dat die ik die zelf niet eerder heb gehad. Als ik wat rondzoek lijkt het erop dat er data is met ongeldige timestamps. 

Hoe dat exact kan gebeuren is mij onbekend, maar hoe dan ook is het zaak om te kijken of je in deze tabel records kan vinden met absurde timestamps:

```
sudo su - postgres
psql dsmrreader

# Probeer deze twee queries:
select min(timestamp) from dsmr_datalogger_dsmrreading ;
select max(timestamp) from dsmr_datalogger_dsmrreading ;
```

Komen daar resultaten uit met absurde timestamps? Die ofwel heel ver in de toekomst ofwel het verleden liggen?

----

Wat betreft de API. Je kunt daar ofwel de client-datalogger voor gebruiken, ofwel een plugin die telegrammen doorstuurt van je Pi naar de VM. In dat laatste geval draai je eigenlijk twee keer DSMR-reader en dat gebruik ik zelf voor testen, maar dat is dus een iets andere reden.

rodeho, 2019-04-28:
Ha @dennissiemensma 

Dank voor het meedenken, nee die twee queries geven hele normale waardes terug. De opstart datum en de huidige datum.

Op de "corrupte" Pi DSMR gui zie ik dat er 2.5mlj readings zijn. Het lijkt dat deze in `dsmr_datalogger_dsmrreading` opgeslagen zijn? Hoe relevant zijn deze losse metingen, want op de corrupte dag van 08-01 kan ik wel gewoon in de webGUI stats opvragen welke worden weergegeven? Is er een andere tabel met de dagtotalen voor het "archief" in de WebGUI ?

Stel dat ik deze tabel `dsmr_datalogger_dsmrreading` zou droppen en dan de db exporteren, wat is dan de impact op de nieuwe server?

rodeho, 2019-04-28:
Die twee queries geven goede output terug, de start datum en de huidige datum. De tabel `dsmr_datalogger_dsmrreading` bevat volgens mij de 2.5miljoen readings welke de Pi weergeeft? Als deze tabel verloren gaat, omdat ik hem drop en dan pas exporteer. Is dat heel erg, mis ik dan functionaliteit? Als ik nu naar de corrupte dag blader in de WebGui op de Pi dan geeft hij nl wel de data weer, dat geeft mij de indruk dat er een andere tabel is met gearchiveerde dagtotalen?

dennissiemensma, 2019-04-28:
Dat klopt, alle data wordt geaggregeerd in meerdere tabellen. Uiteindelijk is er slechts 1 tabel die 'belangrijk' is en dat is de tabel `dsmr_stats_daystatistics`. De metingen hebben verder geen toegevoegde waarde, mits je archief goed gevuld is.

Je hoeft trouwens geen tabellen te droppen om te exporteren. Je kunt hem beter dan excluden in `pg_dump`:
```
  -T, --exclude-table=TABLE    do NOT dump the named table(s)
```

rodeho, 2019-04-28:
Ik denk dat het werkt, morgen nog even kijken wanneer de teruglevering start of deze dan ook (groene velden) in de GUI zichtbaar worden.

Ter naslag voor anderen, bij mij was de tabel "dsmr_datalogger_dsmrreading" corrupt.

**Op de corrupte machine**
```
sudo supervisorctl stop all

sudo sudo -u postgres pg_dump dsmrreader   -T, --exclude-table=dsmr_datalogger_dsmrreading > backup_190428.pgsql

```

**Op de nieuwe machine**
```
sudo supervisorctl stop all
sudo sudo -u postgres dropdb dsmrreader
sudo sudo -u postgres createdb -O dsmrreader dsmrreader
sudo sudo -u postgres psql dsmrreader < backup_190428.pgsql

Omdat er dan nog een tabel mist in de database heb ik uit een reguliere backup file van DSMR reader regel 0-1930 geplakt, dan zijn alle tabellen weer aangemaakt met de juiste rechten en relaties etc.

```

dennissiemensma, 2019-04-28:
Je kunt overigens nog overwegen om retentie in te schakelen in dsmr-reader. Dan verwijdert die na X tijd oude metingen. 

Zelf heb ik hem op een maand staan en heb ik na 3,5 jaar zo'n 300.000 metingen over. De dagstatistieken zijn immers toch het belangrijkste en die worden nooit opgeschoond. Het voordeel is tevens dat de backup's sneller gemaakt en gestored worden. Kijk maar of je dat ook wil.

rodeho, 2019-04-28:
Jep heb ik op de Pi vanmiddag ingesteld, op een week. Hopelijk ruimt hij dan ook de corrupte entries op en kan ik nog een full db export draaien. Voordat de Pi alleen de serial datalogger wordt richting -> VM API.

dennissiemensma, 2019-04-28:
De retentie wordt alleen 's nachts uitgevoerd dus het kan even duren voordat die alles bij je heeft gedaan, afhankelijk van hoeveel dagen aan data je hebt.

Het opruimen schoont in jouw geval alles ouder dan een week op. Maar laat wel per uur twee metingen staan (de eerste en laatste per uur). In mijn voorbeeld zijn 300k alle metingen die over zijn na ruim 3 jaar, dus lekker compact.

closed 2019-05-13 by dennissiemensma