dsmrreader/dsmr-reader #632: levering en teruglevering en suggesties

jpschouten, 2019-05-02:
Allereerst, dank je wel voor deze prachtige software!

Levering en teruglevering zie ik nu in één grafiek. Wanneer er teruggeleverd word zie ik de overcapaciteit in de grafiek terug. Ik zou het graag los van elkaar zien. (wellicht niet mogelijk door de meter, het is een Gyr E350 DSMR 4 meter).

Verder wat suggesties:
Zonuren erbij als plugin ? om dit als vergelijk te gebruiken met opbrengst van de panelen?
Meterstanden van de afzonderlijke meters?

dennissiemensma, 2019-05-02:
Bedankt voor je suggesties. Helaas sturen alle meters de netto opbrengt terug, omdat een deel al berrekend wordt in het huishouden, vóór de slimme meter.

Wat betreft je andere suggesties. Dit project houd ik nu bewust beperkt tot alleen gegevens van de slimme meter. Ik weet niet exact wat je bedoelt met afzonderlijke meterstanden, maar die zijn in te zien in de Statistieken-pagina denk ik. Je zou kunnen kijken of je een deel van je behoeftes via PVOutput.org kan doen, hoewel ik niet weet of ze zonuren ondersteunen.

dennissiemensma, 2019-11-08:
Gesloten ivm dezlefde reden als #594

closed 2019-11-08 by dennissiemensma