dsmrreader/dsmr-reader #633: Remote datalogger vraag

WatskeBart, 2019-05-02:
Ik kan eindelijk weer gebruik maken van deze prachtige applicatie nu ik in mijn nieuwe huis ook een slimme meter heb. Alleen wil ik nu dsmr-reader in Docker draaien (dit werkt al prima) en een RPi2 gebruiken als remote datalogger.

Moet ik de RPi2 volledig installeren met dsmr-reader om hem enkel te gebruiken als remote datalogger?

dennissiemensma, 2019-05-02:
Nee in principe heb je alleen Supervisor, Python3 en een virtualenv nodig.

```
sudo apt-get install -y supervisor python3 python3-pip python3-virtualenv virtualenvwrapper

sudo useradd dsmr --home-dir /home/dsmr --create-home --shell /bin/bash
sudo usermod -a -G dialout dsmr

sudo sudo -u dsmr mkdir /home/dsmr/.virtualenvs
sudo sudo -u dsmr virtualenv /home/dsmr/.virtualenvs/dsmrreader --no-site-packages --python python3
sudo sh -c 'echo "source ~/.virtualenvs/dsmrreader/bin/activate" >> /home/dsmr/.bashrc'
sudo sh -c 'echo "cd ~/dsmr-reader" >> /home/dsmr/.bashrc'

sudo su - dsmr

pip install pyserial==3.2.1
pip install requests==2.12.4
```

En dan verder de guide volgen: https://dsmr-reader.readthedocs.io/en/latest/api.html#script

Wellicht mis ik nog iets, maar dat zal dan iets kleins zijn.

WatskeBart, 2019-05-02:
Bedankt voor de snelle reactie, ik ga er mee aan de slag.

WatskeBart, 2019-05-03:
Bedankt voor alle info, ik heb de remote datalogger aan de praat.
Het enige wat ontbrak was:

- De gebruiker dsmr toevoegen aan sudoers
- Bij het wisselen naar gebruiker **dsmr** met het commando **sudo su - dsmr** kreeg ik de melding dat de directory **dsmr-reader** niet bestond in **/home/dsmr/**. Directory aangemaakt met **mkdir dsmr-reader** en de error was weg.

closed 2019-05-03 by WatskeBart